mkschreder/juci #1: Error running

richb-hanover, 2015-05-20:


mkschreder, 2015-05-20:
You are using old version. Get latest one from inteno sdk:

git clone http://ihgsp.inteno.se/git/iopsysAA.git iopsys
cd iopsys
git checkout BB

Then in folder package/inteno/luciexpress

richb-hanover, 2015-05-20:
OK. Got a little bit farther. I did clone the inteno repository (all 254 MBytes(!)) I checked out BB, I cd'd to package/inteno/luciexpress and ran `npm install` and `node server.js`. 

On localhost:3000, I saw a web page asking for router address, user, and password.

I didn't know how to fill these in. It also seems that I need a router (simulated or real) to start using it. 

Will you be updating Github https://github.com/mkschreder/luci-express/ to reflect this info? Thanks.
```
Local server listening on http://:::3000
JSON_CALL (-> http://192.168.1.4/ubus): {"jsonrpc":"2.0","method":"list","id":1,"params":["00000000000000000000000000000000","*","",{}]}
JSON_RESP: {"jsonrpc":"2.0","result":[1,"ETIMEOUT"]}
JSON_CALL (-> http://192.168.1.4/ubus): {"jsonrpc":"2.0","method":"call","id":1,"params":["00000000000000000000000000000000","session","access",{"keys":""}]}
ERROR: Error: connect EHOSTUNREACH
JSON_RESP: {"jsonrpc":"2.0","result":[1,"Error: connect EHOSTUNREACH"]}
JSON_LOCAL: {"jsonrpc":"2.0","method":"call","id":1,"params":["00000000000000000000000000000000","local","features",{}]}
JSON_RESP: {"jsonrpc":"2.0","result":[0,{"list":["rpcforward"]}]}
JSON_CALL (-> http://192.168.1.4/ubus): {"jsonrpc":"2.0","method":"call","id":1,"params":["00000000000000000000000000000000","session","login",{"username":"","password":""}]}
ERROR: Error: connect EHOSTDOWN
JSON_RESP: {"jsonrpc":"2.0","result":[1,"Error: connect EHOSTDOWN"]}
JSON_CALL (-> http://192.168.1.4/ubus): {"jsonrpc":"2.0","method":"call","id":1,"params":["00000000000000000000000000000000","session","login",{"username":"admin","password":"admin"}]}
JSON_CALL (-> http://192.168.1.4/ubus): {"jsonrpc":"2.0","method":"call","id":1,"params":["00000000000000000000000000000000","session","login",{"username":"admin","password":"admin"}]}
JSON_RESP: {"jsonrpc":"2.0","result":[1,"ETIMEOUT"]}
JSON_RESP: {"jsonrpc":"2.0","result":[1,"ETIMEOUT"]}
ERROR: Error: connect ETIMEDOUT
ERROR: Error: connect ETIMEDOUT
ERROR: Error: connect ETIMEDOUT
^C
bash-3.2$
```

mkschreder, 2015-05-20:
Well it is a router gui after all, so it is of course quite useless without a device. But what you can do of course is build a virtualbox image and run it in virtualbox but you will not have access to a lot of the functions because it is a simulated device of course. 

The way it works is that the gui interacts with the device mainly by means of rpc. so it depends on rpc functions to be implemented on the device. Using the local server you can connect to a device that already has all the necessary rpc functions installed. Normally we just build a full image and just flash it on the real thing. 

But what also would work is compiling rpcd, ubus, libubus, uci, ubox and uhttpd with mod-ubus locally. They will build just fine. But then you will need to add uci configs to /etc/config/ manually. It is trickier unless you either are really into linux hacking or you know how everything works and what dependencies are needed. 

If you are really interested then you can get yourself an inteno router and flash it with latest official build and then everything will work out of the box. If you are interested in a real box let me know and I can give you a quote :-)

richb-hanover, 2015-05-20:
Ahah! Now I understand better. I am working with the OpenWrt team to make that software more approachable by newcomers, and was intrigued by news of your new/improved GUI. I would say that getting all that infrastructure set up is more work than my curiosity warrants, but good luck on your project.

BTW: if you are based on OpenWrt, I would strongly recommend that you build in and enable the SQM/fq_codel facilities to minimize bufferbloat. Anyone who can tout having SQM and fq_codel will have a competitive advantage for the the next 12 months. You can read more about SQM on OpenWrt at http://wiki.openwrt.org/doc/howto/sqm, check out the bufferbloat on your current connection at http://dslreports.com/speedtest or get a layman's guide to bufferbloat at http://richb-hanover.com/bufferbloat-and-the-ski-shop/

closed 2015-05-20 by richb-hanover

mkschreder, 2015-05-20:
Well newcomers do not come and start downloading and compiling software. They buy the final product that has allready been configred and compiled and quality tested. :-)

richb-hanover, 2015-05-20:
Yes, of course. And all the better for newcomers if the vendor has de-bloated the router. Thanks!

mkschreder, 2015-05-20:
while you focus on giving users a good user experience, openwrt should focus more on fixing all the memory leaks in openwrt services and utilities. Otherwise all you do is wasted time when the base software leaks like swiss cheese ;-) 

And btw, I do not represent the vendor in any way. I write my own software. If you don't like it, fix it. It's open source. That's what I do every day.