mkschreder/juci #2: Build failed.

mullex, 2015-06-12:
I git cloned the project to local and use make to build it. I then got the error message below. What is the problem?

make[1]: ../juci-build-tpl-cache: Command not found
make[1]: *** [htdocs/js/01-juci-tpl.js] Error 127
make[1]: Leaving directory `/home/alex/Projects/luci-express/juci'
make: *** [juci] Error 2

mkschreder, 2015-06-12:
It should be there. Works for me.

git clone https://github.com/mkschreder/luci-express.git
cd luci-express
make

You must have probably deleted the file or something.

mkschreder, 2015-06-12:
If you are integrating it to openwrt build process then make sure you copy
the whole juci folder to builddir and then execute make in the correct
directory.

mullex, 2015-06-13:
Many thanks. I placed the whole folder inside builddir and then It compiles fine. But i didn't find it in make menuconfig.How can i install it to openwrt.

mkschreder, 2015-06-13:
There is an openwrt package for juci in iopsys sdk under
package/inteno/webgui/juci. In juci readme are instructions on how to
checkout iopsys sdk through git. So actually you should let the wrt
makefile checkout juci and build it. Then it will also build all
dependencies and create an ipk.

mullex, 2015-06-14:
I followed the instructions. But when i run "make" inside the iop folder, I always got the error message below. 

Checking out files from the git repository...
Cloning into 'mtd-utils-1.4.5'...
ssh_exchange_identification: read: Connection reset by peer
fatal: Could not read from remote repository.

mullex, 2015-06-15:
I used your mirror of iop on github. The issue above was gone but i got the following errors. Does this work for u?

echo "Checking out files from the git repository..."; mkdir -p /home/mullex/Projects/iop/tmp/dl && cd /home/mullex/Projects/iop/tmp/dl && rm -rf ice-client-3.1.0 && [ \! -d ice-client-3.1.0 ] && git clone ssh://git@iopsys.inteno.se/ice-client.git ice-client-3.1.0 && (cd ice-client-3.1.0 && git checkout 249d6033274f93da343b7fb5879dae0edc5f9682) && echo "Packing checkout..." && 	/usr/bin/gtar cfz /home/mullex/Projects/iop/tmp/dl/ice-client-3.1.0-RC4.tar.gz ice-client-3.1.0 && mv /home/mullex/Projects/iop/tmp/dl/ice-client-3.1.0-RC4.tar.gz /home/mullex/Projects/iop/dl/ && rm -rf ice-client-3.1.0; 
Checking out files from the git repository...
Cloning into 'ice-client-3.1.0'...
ssh_exchange_identification: read: Connection reset by peer

mkschreder, 2015-06-15:
Thats ok. ice client is inteno specific utility so you will not be able to
build it anyway. The same goes for the broadcom wifi stuff that is closed
source from broadcom. But you will not need that one either unless you
build for an inteno target. Iopsys public sdk will be updated in due time
to include working alternatives for these things. But for now you can just
disable the packages that you can't compile.

Since you have gotten this far, what you can do is actually just build juci
and it's dependencies by doing "make package/inteno/webgui/juci/compile".

Also note that the github mirror you checked out was not the
latest-latest.. so you may wanna do a pull on that one.

mullex, 2015-06-15:
Sorry for keep asking questions as this is the first time i build iop though i have a couple years of experience with openwrt.

I executed the following steps:
1.git clone http://ihgsp.inteno.se/git/iopsysAA.git iop
2.cd iop
3.git checkout BB
4./iop_get_feeds.sh
5.make package/inteno/webgui/juci/compile
I got the following error:
make[1]: Entering directory `/home/alex/Projects/iop'
make[1]: *** No rule to make target `package/inteno/webgui/juci/compile'.  Stop.
make[1]: Leaving directory `/home/alex/Projects/iop'
make: *** [package/inteno/webgui/juci/compile] Error 2

When i execute the make package/inteno/webgui/juci/juci-compile, i got the following error
make: `package/inteno/webgui/juci/juci-compile' is up to date.

mkschreder, 2015-06-15:
Try checkout BB-martin

mkschreder, 2015-06-15:
A huge merge is coming soon on master. Then hopefully things will work much
better.

mullex, 2015-06-15:
I hope that can come soon. I really like your project:)

1.I tried to execute git checkout BB-martin
2.I still got errors.

make package/inteno/webgui/juci/compile V=s
Collecting package info: done
package/Makefile:105: warning: overriding recipe for target `package/cg300/clean'
package/Makefile:105: warning: ignoring old recipe for target `package/cg300/clean'
make[1]: *** No rule to make target `package/inteno/webgui/juci/compile'.  Stop.
make[1]: Leaving directory `/home/alex/Projects/iop'
make: *** [package/inteno/webgui/juci/compile] Error 2

mkschreder, 2015-06-15:
I just realized that you will not be able to build it either because it
needs to pull configs from a repo you don't have access to. I will need to
add default configs. It is actually on my todo list..

There is a script called genconfig. It takes hardware and profile as first
and second argument. It then sets top level targe package and then the rest
is handled by defconfig to select all other packages based on dependencies.
It needs to be run first before make. This actually allows one to have many
different customer profiles with different packages for each target.

What hardware are you using?

mullex, 2015-06-15:
I am using CentOS(x86_64) 7 with VMWARE as the build machine. It has 4-Core CPU,4G RAM,100G SCSI HDD.

mkschreder, 2015-06-15:
But what is your build target?

mullex, 2015-06-15:
my build target is for X86_64.

mullex, 2015-06-16:
Would you let me know when it is ready for me to build and make a try?

mullex, 2015-06-18:
Is there any progress on this?

mkschreder, 2015-06-18:
There is always progress. :-) but we focus mainly on developing for inteno
hardware. Everything else is secondary.

mullex, 2015-06-19:
Do you mean that juci will support Inteno devices for the time being? Does it support any other openwrt devices as i really like juci and want to use it in our project.

mkschreder, 2015-06-20:
It is mostly the backend that is the issue. Juci will work on any device
and if you write your own modules for your device then you're fine. It's
just that currently the backend is actually scattered around several
daemons with functions that are tied into broadcom specifics.

It would be good to break up these daemons into clean ubus modules that can
be tied with corresponding juci app, but the daemons have been around for
way longer than juci so they are used in other places as well. The daemons
have been originally created only with broadcom in mind. And then I have
just built juci on top of the existing ubus api that was already in place.

For an idea of what is needed, here is a repo that will build a juci setup
with uhttpd and all the daemons locally:

git clone https://github.com/mkschreder/juci-feed.git

Do a make then make install then sudo bash server start and go to port 8080
locally. Of course this will be missing a lot of things if you just run it
locally on a pc rather than on a real device, but it is still a fully
standalone setup that is not even dependent on openwrt to run.

So as long as the juci modules can be tailored to work on a desktop pc then
this setup will work just as well as on an inteno router. This repo is just
a proof of concept of how to make it work. I'm not responsible for any
crashes or bugs you're going to get with this setup ;-) use at your own
risk.

Do you mean that juci will support Inteno devices for the time being? Does
it support any other openwrt devices as i really like juci and want to use
it in our project.

mullex, 2015-07-08:
git clone https://github.com/mkschreder/juci-feed.git
make

I got the following error so i still couldn't get it compiled and running

..............
Compiling CSS bin/www/css/01-juci.css..
YUICompressor Version: 2.4.8

Usage: java -jar yuicompressor-2.4.8.jar [options] [input file]

Global Options
  -V, --version             Print version information
  -h, --help                Displays this information
  --type <js|css>           Specifies the type of the input file
  --charset <charset>       Read the input file using <charset>
  --line-break <column>     Insert a line break after the specified column number
  -v, --verbose             Display informational messages and warnings
  -o <file>                 Place the output into <file>. Defaults to stdout.
                            Multiple files can be processed using the following syntax:
                            java -jar yuicompressor.jar -o '.css$:-min.css' *.css
                            java -jar yuicompressor.jar -o '.js$:-min.js' *.js

JavaScript Options
  --nomunge                 Minify only, do not obfuscate
  --preserve-semi           Preserve all semicolons
  --disable-optimizations   Disable all micro optimizations

If no input file is specified, it defaults to stdin. In this case, the 'type'
option is required. Otherwise, the 'type' option is required only if the input
file extension is neither 'js' nor 'css'.
make: *** [all] Error 1

mkschreder, 2015-07-09:
Can you debug a bit more? Try a clean build from clean checkout. Make sure
you are using bash as your shell.

Check that you can run the yui compressor command manually:

yui-compressor <path to failing css file>

If nothing works then you can just comment out the line using yui
compressor. It just makes the css file smaller but it is not strictly
mandatory.

git clone https://github.com/mkschreder/juci-feed.git
make

I got the following error so i still couldn't get it compiled and running

..............
Compiling CSS bin/www/css/01-juci.css..
YUICompressor Version: 2.4.8

Usage: java -jar yuicompressor-2.4.8.jar [options] [input file]

Global Options
-V, --version Print version information
-h, --help Displays this information
--type Specifies the type of the input file
--charset Read the input file using
--line-break Insert a line break after the specified column number
-v, --verbose Display informational messages and warnings
-o Place the output into . Defaults to stdout.
Multiple files can be processed using the following syntax:
java -jar yuicompressor.jar -o '.css$:-min.css' *.css
java -jar yuicompressor.jar -o '.js$:-min.js' *.js

JavaScript Options
--nomunge Minify only, do not obfuscate
--preserve-semi Preserve all semicolons
--disable-optimizations Disable all micro optimizations

If no input file is specified, it defaults to stdin. In this case, the
'type'
option is required. Otherwise, the 'type' option is required only if the
input
file extension is neither 'js' nor 'css'.
make: *** [all] Error 1

mullex, 2015-07-09:
I can always debug it as long as you are available on this:)))

I commented out the following lines in juci-compile(i had to comment out mv statement there is an error complaining about a mssing .out file).
for file in `find bin/www/css/ -name "*.css"`; do
        echo "Compiling CSS ${file}.."
        #yui-compressor ${file} > ${file}.out
        #mv ${file}.out ${file}
done

Then make and make install worked!

But when i run sudo bash server start, got the following error:
[root@localhost juci-feed]# sudo bash server start
PATH:/home/alex/Projects/juci/juci-feed/build/bin:/sbin:/bin:/usr/sbin:/usr/bin
UBUS
RPCD
rpcd: connected to ubus
rpcd: session init
rpcd: uci init
rpcd: plugin init
rpcd: loading /usr/lib/rpcd/juci-ubus-core.so
rpcd: purge savedirs
rpcd: started
QUEST
NETIF
UHTTPD
connected as 502c3854
[root@localhost juci-feed]# chdir(main path): No such file or directory
chdir(main path): No such file or directory
Failed to load network config

mkschreder, 2015-07-09:
That is actually netifd that is giving that error and it's ok. The local
uhttpd should run in the background anyway. For development it is usually
better to use juci-local-server and connect to a real box running uhttpd
over ubus.

mullex, 2015-07-09:
It finally works fine and i can open 127.0.0.1:8080

What is the default password for admin?

mkschreder, 2015-07-09:
The same as for linux user admin.

closed 2015-10-28 by mkschreder