mkschreder/juci #3: Problems building luci-express

readysteadygo2006, 2015-07-17:
Hi,

I'm keen to see luci-express working, as it looks really interesting!

Ive been trying to build it for my  alix2 router, but seem to be having a number of
problems.

1. Do dont seem to be able to clone:
git clone http://ihgsp.inteno.se/git/iopsysAA.git

I thought:
https://github.com/mkschreder/iopsys-openwrt-inteno.git 
was a mirror of it, but when I look for:
package/inteno/webgui/juci

As per your discussion with mullex, it not present!

2. aditionally if I try to build what I clone from:
https://github.com/mkschreder/iopsys-openwrt-inteno.git 

I get am unable to access the juci-feeds:

src-git juci http://iopsys.inteno.se:juci-feed.git

Can you help? as Im keen to get this working!

Best Regards,
Paul

mkschreder, 2015-07-17:
>    1. aditionally if I try to build what I clone from:

The feed should be changed to
https://github.com/mkschreder/juci-openwrt-feed.git. I will update github
mirror..

As for the inteno sdk you should be able to clone. I just checked and the
repo is accessible to me from android. Yes the github is a mirror too but
the inteno repo is always latest one from day to day so you should now use
that one as long as you can access it.

readysteadygo2006, 2015-07-19:
Hi,

Had a chance to try to build sdk.

From message trial with mullex, I understood that broadcom packages 
were specific to your Inteno target, so deleted then from the 
package/iteno subdir.

Additionally assumed the same applied natalie-dect* packages, so again
deleted them from subdir.

Needed to make uhtttpd2 point ot your github, but after these changes
cmake got all the way through to building image.

Additionally made changes to mtd-utils discussed mullex.

I now get a failure when the grub script is running,shown below:

+ grub --batch --no-curses --no-floppy --device-map=/dev/null

    GNU GRUB  version 0.97  (640K lower / 3072K upper memory)

       [ Minimal BASH-like line editing is supported.   For
         the   first   word,  TAB  lists  possible  command
         completions.  Anywhere else TAB lists the possible
         completions of a device/filename. ]
grub> device (hd0) /home/pgrant/vidiwave/newgui/iopsysAA/bin/x86/openwrt-x86-alix2-combined-ext4.img
grub> geometry (hd0) 105 16 63
drive 0x80: C/H/S = 105/16/63, The number of sectors = 105840, /home/pgrant/vidiwave/newgui/iopsysAA/bin/x86/openwrt-x86-alix2-combined-ext4.img
   Partition num: 0,  Filesystem type is ext2fs, partition type 0x83
   Partition num: 1,  Filesystem type is ext2fs, partition type 0x83
grub> root (hd0,0)
 Filesystem type is ext2fs, partition type 0x83
grub> setup (hd0)
 Checking if "/boot/grub/stage1" exists... yes
 Checking if "/boot/grub/stage2" exists... yes
 Checking if "/boot/grub/e2fs_stage1_5" exists... yes
 Running "embed /boot/grub/e2fs_stage1_5 (hd0)"...  19 sectors are embedded.
succeeded
 Running "install /boot/grub/stage1 (hd0) (hd0)1+19 p (hd0,0)/boot/grub/stage2 /boot/grub/menu.lst"... succeeded
Done.
grub> quit
./gen_image_grub.sh: line 12: 32171 Segmentation fault      (core dumped) grub --batch --no-curses --no-floppy --device-map=/dev/null  <<EOF
device (hd0) $OUTPUT
geometry (hd0) $cyl $head $sect
root (hd0,0)
setup (hd0)
quit
EOF

Just to mention, I've set fs to ext4&squachfs via menuconfig

Any ideas?!

Best Regards,
Paul

mkschreder, 2015-07-20:
We never actually build grub. Inteno boxes use CFE. When an inteno box
firmware is built it is configured using ./iop_get_feeds.sh followed by
./genconfig_new <box> <profile>. For example "./genconfig_new cg300 open"
for an open version of firmware for cg300 router.

What I do know though, which could be causing sdk build issues for you, is
that the sdk is actually originally OpenWRT AA, with BB structure on top.
There could be old files in there which have never caused problems for
broadcom builds but are causing issues for you. Try updating sdk grub to
latest version maybe?

The SDK needs to actually be moved to openwrt trunk. Some effort has
already been put into preparing for this by at least moving most of the
inteno versions of packages into package/inteno folder. But the task has
not been completed yet so the sdk is not yet easily mergable with openwrt.

You could try copying these packages on top of a working version of openwrt
that works for your platform and then looking for any warnings of missing
package dependencies and moving them over too. The biggest caveat to watch
for is that there may still be some patched code left in some system
packages which some sdk package version relies on. This will not cause an
immediate failure but may cause wierd issues on a running system. Not very
optimal solution, basically, but may be an easy way to get past the error
at least :-)

mkschreder, 2015-08-31:
It is now possible to build juci on openwrt trunk by: 

- adding feed to feeds.conf.default
src-git juci https://github.com/mkschreder/juci-openwrt-feed.git

- updating and installing the feed: 
./scripts/feeds update juci
./scripts/feeds install -p juci -a

- selecting juci and juci-theme-inteno in config:
make menuconfig

It is however a major work in progress because stock openwrt is more generic and has much less tools than IOPSYS. The work to be done is mainly in making it work with default openwrt services just as wel as it already works on IOPSYS.

readysteadygo2006, 2015-09-07:
Great!

I'll take a look over next couple of days!

BR
Paul

closed 2015-10-28 by mkschreder