mkschreder/juci #5: I can sign in the juci

erniusias, 2015-08-12:
I just look the sign in page, after sign in , the
sid is always “RPC_DEFAULT_SESSION_ID”.  And I get this log "WARNING: JUCI.module() is deprecated! [core]"

mkschreder, 2015-08-12:
When you run from box or using juci-local-server?

erniusias, 2015-08-13:
Using juci-local-server, and I remove /usr/libexec/rpcd/juci-ubus-core-cgi since it block rpcd deamon

mkschreder, 2015-08-13:
That's because you are using old version of rpcd. I have patched it so it
loads shared libs correctly (only .so files, not programs). You can get it
from https://github.com/mkschreder/juci-rpcd

You may also want to get https://github.com/mkschreder/juci-uhttpd with
some fixes there as well. For a complete list of packages that have correct
version you can get https://github.com/mkschreder/juci-feed and just run
make and it will download all of them.

mkschreder, 2015-08-13:
The side effect of removing cgi program is that upgrade and config backup
will not work in webgui.

erniusias, 2015-08-13:
Thanks，I test juci backend on real openwrt router. Forward the ubus request to router with juci-local-server. So I use the juci-core from  juci-openwrt-feed. I can see the session login request is ok, But it  return to the sign in page.

mkschreder, 2015-08-13:
It is once again because you have their rpcd. It has to do with session
access call not returning same info as login call. The access call should
return your currently logged in user. If it doesn't find that info it will
log you out and go back to login page.

erniusias, 2015-08-14:
Thanks for your replay, I replaced the procd with juci-procd, this problem is solved. others I install the questd, then I can arrive at overview page, but I get a large amount of error log. I think I need more js and angular.js knowledge.

mkschreder, 2015-08-14:
Well you should install all dependencies from the juci feed. But you should
take them from iopsys (it is a fork of openwrt that we use on all our
routers) juci is an iopsys project so naturally there can be things
missing. But already there are only a few minor things. before it was much
more.

erniusias, 2015-08-21:
Thanks, I got the page after when switch the theme form vodafone to juci-inteno.

closed 2015-08-21 by erniusias