mkschreder/juci #6: Where to install services

erniusias, 2015-08-22:
I want to know where to install the services such as ueventd an juci-networkd. Others, if it need a init.d script.

mkschreder, 2015-08-22:
I've just added them so they're still experimental. But yes, eventually I
will add init.d scripts for them.

mkschreder, 2015-08-22:
ueventd is at the moment just a bunch of hotplug scripts that generate ubus
events to which juci-networkd and juci-eventd listen and respond. Networkd
is currently responsible for shoving newly connected network dongle into
the right place and eventd is responsible for keeping track of events until
gui retreives them.

The ultimate goal with all of this is to eventually provide a nice user
experience for a scenario like this: user plugs in a dongle, we identify it
as a network device, if gui is connected then we pop up a configuration
dialog in the gui asking what network the user wants to put the new device
into (and ask for pin if necessary). If gui is not connected then we
autoconfigure the dongle as wan device - but there are multiple ways to do
that and iopsys for example uses multiwan scheme for this. Settings are
saved and when user plugs in the dongle again it should be reconfigured
with same settings and "magically" come up without too much hassle. :-)

closed 2015-10-28 by mkschreder