mkschreder/juci #9: Can not establish ubus connection to router. If the router is rebooting then please wait a few minutes and try again.

lvzaina, 2015-09-24:
Hi, mkschreder

build success, but i got "Can not establish ubus connection to router. If the router is rebooting then please wait a few minutes and try again."

openwrt use 15.05 branch (Chaos Calmer)
   git clone git://git.openwrt.org/15.05/openwrt.git

~/source/openwrt_juci$ cat feeds.conf.default | grep juci
src-git juci https://github.com/mkschreder/juci-openwrt-feed.git;

devel has the same problems

mkschreder, 2015-09-24:
This usually happens when you don't have all necessary ubus calls running. In the latest devel I have added a new service for managing ubus scripts in /usr/lib/ubus. This can cause this issue. You should check that you have ubus scripts under /usr/lib/ubus/juci/ and that you have the same ubus objects in ubus list: 

ubus list -v "/juci*"
'/juci/diagnostics' @538d5efe
        "ping":{}
        "traceroute6":{}
        "traceroute":{}
        "ping6":{}
'/juci/ethernet' @d2f21ad3
        "adapters":{}
'/juci/iptv' @f6d27142
        "igmptable":{}
'/juci/macdb' @608de846
        "lookup":{}
'/juci/netmode' @a9ff23ee
        "select":{}
        "current":{}
'/juci/network' @44c01a63
        "clients":{}
        "load":{}
        "adapters":{}
        "nameservers":{}
        "nat_table":{}
....

If they are not running try to kill etc-ubus-daemon (will get a better name later) and then start it again directly "# ./etc-ubus-daemon". 

If you still have problems then give me the exact hash for the juci feed version you are using (cd feeds/juci; git log -1).

mkschreder, 2015-09-24:
I will add that service to the feed tomorrow.. then it will work again.

lvzaina, 2015-09-25:
Hi, mkschreder
thank you very much!
but i got login screen with blank page.

openwrt use 15.05 branch (Chaos Calmer)
git clone git://git.openwrt.org/15.05/openwrt.git

~/source/openwrt_juci$ cat feeds.conf.default | grep juci
src-git juci https://github.com/mkschreder/juci-openwrt-feed.git;devel

~/source/openwrt_juci/feeds/juci$ git log -1
commit e3e7e311dc32d2c33844afe7c9ae7910a22de9f5
Author: Martin Schröder <mkschreder.uk@gmail.com>
Date:   Fri Sep 25 15:19:21 2015 +0200

    iconnect: added initial package

Now what should I do ?

lvzaina, 2015-09-25:
using firebug, I got some information

WARNING: JUCI.module() is deprecated! [core]
http://192.168.1.1/js/01-juci.js
Line 49

WARNING: JUCI.module() is deprecated! [phone]
http://192.168.1.1/js/01-juci.js
Line 49

WARNING: JUCI.module() is deprecated! [internet]
http://192.168.1.1/js/01-juci.js
Line 49

RPC error (uci.configs): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
http://192.168.1.1/js/01-juci.js
Line 48

UCI failed to initialize!
http://192.168.1.1/js/01-juci.js
Line 49

$uci.sync() is deprecated and will be replaced with $uci.$sync() in future version to avoid config collisions. Please do not use it!
http://192.168.1.1/js/01-juci.js
Line 49

RPC error (uci.revert): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
http://192.168.1.1/js/01-juci.js
Line 48
......

mkschreder, 2015-09-25:
You still don't seem to have access to ubus. Can you see "/juci/*" objects
on ubus? Maybe something else is missing. Make sure you use juci-rpcd and
juci-uhttpd from the feed and not the stock ones.. Try to copy everything
in juci builddir "bin" folder directly to root on your box and restart
rpcd.

lvzaina, 2015-09-25:
i don't know `"/juci/*" objects on ubus`
>Try to copy everything in juci builddir "bin" folder directly to root on your box and restart rpcd.
i still got login screen with blank page.

```
root@OpenWrt:~# ps
  PID USER       VSZ STAT COMMAND
    1 root       976 S    /sbin/procd
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    5 root         0 SW<  [kworker/0:0H]
    7 root         0 SW   [rcu_sched]
    8 root         0 SW   [rcu_bh]
    9 root         0 SW   [migration/0]
   10 root         0 SW   [migration/1]
   11 root         0 SW   [ksoftirqd/1]
   13 root         0 SW<  [kworker/1:0H]
   14 root         0 SW<  [khelper]
   15 root         0 SW<  [perf]
   16 root         0 SW   [kworker/u4:1]
  187 root         0 SW<  [writeback]
  188 root         0 SW<  [bioset]
  190 root         0 SW<  [kblockd]
  312 root         0 SW<  [ata_sff]
  442 root         0 SW   [kswapd0]
  444 root         0 SW   [fsnotify_mark]
  541 root         0 SW<  [acpi_thermal_pm]
  605 root         0 SW   [scsi_eh_0]
  606 root         0 SW<  [scsi_tmf_0]
  609 root         0 SW   [scsi_eh_1]
  610 root         0 SW<  [scsi_tmf_1]
  615 root         0 SW   [scsi_eh_2]
  618 root         0 SW<  [scsi_tmf_2]
  619 root         0 SW   [scsi_eh_3]
  621 root         0 SW<  [scsi_tmf_3]
  675 root         0 SW<  [kpsmoused]
  690 root         0 SW<  [deferwq]
  693 root         0 SW   [kworker/1:3]
  708 root         0 SW<  [ext4-rsv-conver]
  769 root         0 SW<  [kworker/0:1H]
 1319 root       688 S    /sbin/ubusd
 1531 root       612 S    /sbin/askfirst /bin/ash --login
 1532 root      1112 S    /bin/ash --login
 1706 root         0 SW<  [ipv6_addrconf]
 2014 root       852 S    /sbin/etc-ubus-daemon
 2035 root       772 S    /sbin/logd -S 16
 2062 root      1068 S    {juci-eventd} /usr/bin/lua /sbin/juci-eventd
 2075 root      1052 S    {juci-netdevd} /usr/bin/lua /sbin/juci-netdevd
 2088 root      1080 S    {ubus-dect} /usr/bin/lua /sbin/ubus-dect
 2139 root      1060 S    /sbin/netifd
 2155 root       844 S    /usr/sbin/odhcpd
 2195 root       872 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300
 2239 root       756 S    /sbin/mountd -f
 2262 root      1104 S    /usr/sbin/ntpd -n -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.org -p 2.openwrt.pool.ntp.org -p 3.openwrt.pool.ntp.org
 2276 root      1028 S    /usr/sbin/uhttpd -f -h /www -r OpenWrt -x /cgi-bin -u /ubus -t 60 -T 30 -k 20 -A 1 -n 3 -N 100 -R -p 0.0.0.0:80 -p [::]:80
 2337 root      1100 S    udhcpc -p /var/run/udhcpc-eth0.pid -s /lib/netifd/dhcp.script -f -t 0 -i eth0 -C
 2433 nobody     820 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf -k -x /var/run/dnsmasq/dnsmasq.pid
 3248 root         0 SW   [kworker/0:0]
 3668 root         0 SW   [kworker/0:2]
 3821 root         0 SW   [kworker/1:1]
 3831 root         0 SW   [scsi_eh_4]
 3832 root         0 SW<  [scsi_tmf_4]
 3838 root         0 SW   [usb-storage]
 3909 root         0 SW   [kworker/u4:2]
 3910 root         0 SW<  [kworker/1:1H]
 4024 root         0 SW<  [ext4-rsv-conver]
 4046 root      1100 S    /sbin/rpcd
 4129 root       940 R    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300
 4130 root      1104 S    -ash
 4136 root      1100 R    ps
```

mkschreder, 2015-09-25:
All scripts from /usr/lib/ubus/* should appear as ubus objects with names
that start with /. You can list them using ubus list "/*" -v.

Next step is to check if you can log in with curl and access them over
http. You problem is somewhere between the uhttpd ubus plugin and the
scripts in /usr/lib/ubus.

mkschreder, 2015-10-09:
In devel etc-ubus-daemon is now called ubus-scriptd. You can build from
latest devel branch of the feed (make sure you have ";devel" after feed
url).

mkschreder, 2015-10-09:
There is now also a test branch of openwrt with juci included. You can
build juci branch from https://github.com/mkschreder/openwrt.git . It
worked fine to build yesterday..

lvzaina, 2015-10-16:
Thanks a lot.
but i got login screen with blank page.
```
$ git clone https://github.com/mkschreder/openwrt.git -b juci openwrt_juci
$ cd openwrt_juci/
$ git branch
* juci
$ ./scripts/feeds update -a
$ ./scripts/feeds install -a
$ make menuconfig
$ cat .config | grep juci
CONFIG_FEED_juci=y
CONFIG_PACKAGE_juci-ubus-core=y
# CONFIG_PACKAGE_juci-asterisk is not set
# CONFIG_PACKAGE_juci-broadcom-dsl is not set
# CONFIG_PACKAGE_juci-broadcom-ethernet is not set
# CONFIG_PACKAGE_juci-broadcom-vlan is not set
# CONFIG_PACKAGE_juci-broadcom-wl is not set
# CONFIG_PACKAGE_juci-ddns is not set
CONFIG_PACKAGE_juci-diagnostics=y
CONFIG_PACKAGE_juci-dnsmasq-dhcp=y
CONFIG_PACKAGE_juci-dropbear=y
CONFIG_PACKAGE_juci-ethernet=y
CONFIG_PACKAGE_juci-event=y
CONFIG_PACKAGE_juci-firewall-fw3=y
# CONFIG_PACKAGE_juci-igmpinfo is not set
# CONFIG_PACKAGE_juci-inteno-router is not set
CONFIG_PACKAGE_juci-jquery-console=y
# CONFIG_PACKAGE_juci-macdb is not set
# CONFIG_PACKAGE_juci-minidlna is not set
CONFIG_PACKAGE_juci-mod-status=y
CONFIG_PACKAGE_juci-mod-system=y
# CONFIG_PACKAGE_juci-natalie-dect is not set
# CONFIG_PACKAGE_juci-netmode is not set
CONFIG_PACKAGE_juci-network-netifd=y
# CONFIG_PACKAGE_juci-router-openwrt is not set
# CONFIG_PACKAGE_juci-samba is not set
CONFIG_PACKAGE_juci-sysupgrade=y
# CONFIG_PACKAGE_juci-upnp is not set
CONFIG_PACKAGE_juci-usb=y
CONFIG_PACKAGE_juci-theme-inteno=y
CONFIG_PACKAGE_juci=y
CONFIG_PACKAGE_juci-uhttpd=y
CONFIG_PACKAGE_juci-uhttpd-mod-ubus=y
CONFIG_PACKAGE_juci-rpcd=y
# CONFIG_PACKAGE_juci-rpcd-mod-file is not set
$ make V=99
```
using firebug
```
RPC error (uci.configs): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
```
minicom
```
root@OpenWrt:/usr/share/rpcd# ls -l *
acl.d:
-rw-r--r--    1 root     root           307 Oct 10 19:20 juci-diagnostics.json
-rw-r--r--    1 root     root           173 Oct 10 19:20 juci-dnsmasq-dhcp.json
-rw-r--r--    1 root     root           166 Oct 10 19:20 juci-dropbear.json
-rw-r--r--    1 root     root           148 Oct 10 19:20 juci-ethernet.json
-rw-r--r--    1 root     root           134 Oct 10 19:20 juci-event.json
-rw-r--r--    1 root     root           241 Oct 10 19:20 juci-firewall-fw3.json
-rw-r--r--    1 root     root           136 Oct 10 19:20 juci-mod-status.json
-rw-r--r--    1 root     root           966 Oct 10 19:20 juci-mod-system.json
-rw-r--r--    1 root     root           608 Oct 10 19:20 juci-network-netifd.json
-rw-r--r--    1 root     root           388 Oct 10 19:20 juci-sysupgrade.json
-rw-r--r--    1 root     root           409 Oct 10 19:20 juci-ubus-core.json
-rw-r--r--    1 root     root           129 Oct 10 19:20 juci-usb.json
-rw-r--r--    1 root     root           179 Oct 10 19:20 unauthenticated.json

menu.d:
-rw-r--r--    1 root     root          8470 Oct 10 19:20 juci-theme-inteno.json
root@OpenWrt:/usr/share/rpcd#
root@OpenWrt:/usr/share/rpcd# ubus call system board
{
        "kernel": "4.1.6",
        "hostname": "OpenWrt",
        "system": "Qualcomm Atheros QCA9533 ver 2 rev 0",
        "model": "Atheros AP147-010 reference board",
        "release": {
                "distribution": "OpenWrt",
                "version": "Bleeding Edge",
                "revision": "r47067",
                "codename": "designated_driver",
                "target": "ar71xx\/generic",
                "description": "OpenWrt Designated Driver r47067"
        }
}
root@OpenWrt:/usr/share/rpcd#
```

mkschreder, 2015-10-16:
I built one image today and it worked fine. Are you using latest feed?

If you get a blank page, it is usually temporary and probably due to devel
branch being slightly out of sync. You should primarily look in browser
console and see what happened there. The access denied error is always
recoverable. You should look for an exception somewhere higher up. You
could try doing /etc/init.d/rpcd enable && /etc/init.d/rpcd restart.

lvzaina, 2015-10-16:
```
source/openwrt_juci/feeds/juci$ git log -1
commit 3e42f05983ba6a4461e014345256866cf48e2fb1
Author: Martin Schröder <mkschreder.uk@gmail.com>
Date:   Fri Oct 16 07:44:07 2015 +0200

    ubus-scriptd: add libpthread dependency to makefile
lc@BB:~/source/openwrt_juci/feeds/juci$
```
browser console 
```
post ==>
{"jsonrpc":"2.0","method":"call","id":1,"params":["00000000000000000000000000000000","uci","configs"
,{}]}

response ==>
{"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
```
Very sorry, my English is poor. I found this error occurs:
```
#file: juci / src / js / uci.js 
544 $ rpc.uci.configs () 
#file: juci-uhttpd-2014-10-27 / ubus.c 
540 if (! Conf.ubus_noauth &&! Uh_ubus_allowed (data.sid, data.object, data.function)) { 
#file: juci-uhttpd-2014-10-27 / ubus.c 
498 if (ubus_lookup_id (ctx, "session", & id)) 
```
Thank you very much once again, this is really a strange question.

mkschreder, 2015-10-16:
Try clearing your browser cache.

lvzaina, 2015-10-16:
```
lvzaina@BB:~$ curl -X POST -H "Content-Type: application/json" -d '{"jsonrpc":"2.0","method":"call","id":1,"params":["00000000000000000000000000000000","/juci/network.status","arp",{}]}' http://192.168.1.1/ubus
{"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}lvzaina@BB:~$ 
lvzaina@BB:~$ 
lvzaina@BB:~$ curl -X POST -H "Content-Type: application/json" -d '{"jsonrpc":"2.0","method":"call","id":1,"params":["00000000000000000000000000000000","uci","configs",{}]}' http://192.168.1.1/ubus
{"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}lvzaina@BB:~$ 
```
I cleared the cache, but still get "Access denied", But also get additional information:
```
WARNING: JUCI.module() is deprecated! [core]
r.prototype.module()01-juci.js 
e = "core"
t = undefined
n = undefined
01-juci.js()
WARNING: JUCI.module() is deprecated! [internet]
r.prototype.module()01-juci.js 
e = "internet"
t = undefined
n = undefined
50-juci-firewall-fw3.js()
```

lvzaina, 2015-10-16:
I tried "uhttpd" add "-a" option
Browser console output:
```
Init UBUS -> 
Init UCI
Missing type definition for config rpcd
Missing type definition for config ubootenv
Missing type definition for config uhttpd
Missing type definition for config wireless
Init CONFIGc
invalid config name boardpanelc
```
and 
```
post ==>
{"jsonrpc":"2.0","method":"call","id":1,"params":["00000000000000000000000000000000","uci","revert",
{"config":"system"}]}
response ==>
RPC succeeded (uci.revert), but returned error: {"jsonrpc":"2.0","id":1,"result":[6,{"code":6}]}: Access denied

post ==>
{"jsonrpc":"2.0","method":"call","id":1,"params":["00000000000000000000000000000000","uci","get",{"config"
:"system"}]}
response ==>
RPC succeeded (uci.get), but returned error: {"jsonrpc":"2.0","id":1,"result":[6]}: Access denied
```
but
```
root@OpenWrt:/# ubus call uci get '{"config":"system"}'
{
        "values": {
                "cfg02e48a": {
                        ".anonymous": true,
                        ".type": "system",
                        ".name": "cfg02e48a",
                        ".index": 0,
                        "hostname": "OpenWrt",
                        "timezone": "UTC"
                },
                "ntp": {
                        ".anonymous": false,
                        ".type": "timeserver",
                        ".name": "ntp",
                        ".index": 1,
                        "server": [
                                "0.openwrt.pool.ntp.org",
                                "1.openwrt.pool.ntp.org",
                                "2.openwrt.pool.ntp.org",
                                "3.openwrt.pool.ntp.org"
                        ],
                        "enabled": "1",
                        "enable_server": "0"
                },
                "led_wan": {
                        ".anonymous": false,
                        ".type": "led",
                        ".name": "led_wan",
                        ".index": 2,
                        "name": "WAN",
                        "sysfs": "ap147:green:wan",
                        "trigger": "netdev",
                        "dev": "eth1",
                        "mode": "link tx rx"
                },
                "led_lan1": {
                        ".anonymous": false,
                        ".type": "led",
                        ".name": "led_lan1",
                        ".index": 3,
                        "name": "LAN1",
                        "sysfs": "ap147:green:lan1",
                        "trigger": "switch0",
                        "port_mask": "0x10"
                },
                "led_lan2": {
                        ".anonymous": false,
                        ".type": "led",
                        ".name": "led_lan2",
                        ".index": 4,
                        "name": "LAN2",
                        "sysfs": "ap147:green:lan2",
                        "trigger": "switch0",
                        "port_mask": "0x08"
                },
                "led_lan3": {
                        ".anonymous": false,
                        ".type": "led",
                        ".name": "led_lan3",
                        ".index": 5,
                        "name": "LAN3",
                        "sysfs": "ap147:green:lan3",
                        "trigger": "switch0",
                        "port_mask": "0x04"
                },
                "led_lan4": {
                        ".anonymous": false,
                        ".type": "led",
                        ".name": "led_lan4",
                        ".index": 6,
                        "name": "LAN4",
                        "sysfs": "ap147:green:lan4",
                        "trigger": "switch0",
                        "port_mask": "0x02"
                },
                "led_wlan2g": {
                        ".anonymous": false,
                        ".type": "led",
                        ".name": "led_wlan2g",
                        ".index": 7,
                        "name": "WLAN 2.4 GHz",
                        "sysfs": "ap147:green:wlan-2g",
                        "trigger": "phy0tpt"
                }
        }
}
root@OpenWrt:/#
```
In the router's shell, it works

mkschreder, 2015-10-16:
You should not need to do anything. The branch should compile and work by
default.

ubus in shell and ubus over uhttpd are different. Over uhttpd it checks for
access rights when you make the call.

What is your output of "ubus list"?

lvzaina, 2015-10-17:
```
root@OpenWrt:/# ubus list
/juci/diagnostics
/juci/ethernet
/juci/network
/juci/network.lua
/juci/network.status
/juci/system
/juci/system.conf
/juci/system.process
/juci/system.service
/juci/system.time
/juci/system.upgrade
/juci/system.user
/juci/usb
dhcp
juci.network.netdevd
juci.ui
log
network
network.device
network.interface
network.interface.lan
network.interface.loopback
network.interface.wan
network.interface.wan6
network.wireless
service
session
system
uci
root@OpenWrt:/# 
```

lvzaina, 2015-10-22:
I ues ie and chrome, I got some information
```
WARNING: JUCI.module() is deprecated! [core]
WARNING: JUCI.module() is deprecated! [core]
WARNING: JUCI.module() is deprecated! [core]
WARNING: JUCI.module() is deprecated! [internet]
Init UBUS -> 
Init UCI
RPC error (uci.configs): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
Init CONFIG
RPC error (uci.revert): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
RPC error (uci.get): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
Could not sync config boardpanel
RPC error (uci.revert): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
RPC error (uci.get): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
Could not sync config system
invalid config name juci
Using settings from config.json on router
Failed to verify session.
juci: loading menu from server..
RPC error (juci.ui.menu): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
SCRIPT5022: [$injector:modulerr] http://errors.angularjs.org/1.3.15/$injector/modulerr?p0=juci&p1=Error%3A%20%5B%24injector%3Aunpr%5D%20http%3A%2F%2Ferrors.angularjs.org%2F1.3.15%2F%24injector%2Funpr%3Fp0%3Dt%0A%20%20%20at%20Anonymous%20function%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A37%3A52294)%0A%20%20%20at%20r%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A37%3A51263)%0A%20%20%20at%20i%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A37%3A51527)%0A%20%20%20at%20r%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A37%3A50757)%0A%20%20%20at%20Anonymous%20function%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A37%3A50889)%0A%20%20%20at%20s%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A37%3A36799)%0A%20%20%20at%20f%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A37%3A50659)%0A%20%20%20at%20_t%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A37%3A52451)%0A%20%20%20at%20o%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A37%3A41773)%0A%20%20%20at%20%24%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A37%3A42253)
```
# cat feeds.conf.default | grep juci
src-git juci https://github.com/mkschreder/mks-openwrt-feed.git

mkschreder, 2015-10-22:
juci feed has multiple branches. "devel" is newest one. "for-14.07" is for
BB etc..

To build from devel your feed url would look like this: src-git juci
https://github.com/mkschreder/mks-openwrt-feed.git;devel

lvzaina, 2015-10-23:
I found this error is because nodejs version is too old! 
After updating nodejs I can see login page. 
Now I want to know which version of nodejs should I use? v4.2.1? 
The current version is v0.10.37

mkschreder, 2015-10-23:
Standard apt-get version should work. But remember to symlink so
/usr/bin/node -> is a link to -> /usr/bin/nodejs

lvzaina, 2015-10-27:
I found this error(can not find index.html) is because my system default shell is dash !
```
$ sudo rm /bin/sh
$ sudo ln -s /bin/bash /bin/sh
```
Now, I can see the WebGUI !

How to add multi-language support?

mkschreder, 2015-10-27:
What a coincidence. I fould this bug yesterday as well. I will update
affected files to use bash explicitly at the top of the file.

mkschreder, 2015-10-27:
To add a new language you create a new language file in po folder and
update it with the pot file you will have in that folder after you build.
Once you build gain the language will be included. You will then need to
add it to config.json in juci www root to see it in the language picker at
the bottom of the page.

lvzaina, 2015-10-27:
When I click on wifi, an error occurred.
```
<h1>
  Not Found
</h1>
The requested URL /pages/wifi-general.html was not found on this server.
```

mkschreder, 2015-10-27:
You need to select juci-broadcom-wl or juci-openwrt-wireless to get wifi
config support. Although you should not see that menu link. It should be
hidden if none of these plugins are installed.

mkschreder, 2015-10-27:
juci-broadcom-wl will not work unless you use an Inteno router. The new
juci-openwrt-wireless is in early stages of development right now.

lvzaina, 2015-10-27:
I chose "luci-openwrt-wireless",
```
$ git log -1
commit c00e8e1fe0b213a6973d560e79d9ea990e0e2fa6
Author: Martin Schröder <mkschreder.uk@gmail.com>
Date:   Sun Oct 25 19:31:57 2015 +0100

    rpcd: integrated upstream changes

$ git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/juci-cc
  remotes/origin/juci-old
  remotes/origin/master
$ cat feeds.conf.default | grep juci
src-git juci https://github.com/mkschreder/mks-openwrt-feed.git;devel
$ cat .config | grep juci
CONFIG_FEED_juci=y
CONFIG_PACKAGE_juci-ubus-core=y
# CONFIG_PACKAGE_juci-asterisk is not set
# CONFIG_PACKAGE_juci-broadcom-dsl is not set
# CONFIG_PACKAGE_juci-broadcom-ethernet is not set
# CONFIG_PACKAGE_juci-broadcom-vlan is not set
# CONFIG_PACKAGE_juci-broadcom-wl is not set
# CONFIG_PACKAGE_juci-ddns is not set
CONFIG_PACKAGE_juci-diagnostics=y
CONFIG_PACKAGE_juci-dnsmasq-dhcp=y
CONFIG_PACKAGE_juci-dropbear=y
CONFIG_PACKAGE_juci-ethernet=y
CONFIG_PACKAGE_juci-event=y
CONFIG_PACKAGE_juci-firewall-fw3=y
# CONFIG_PACKAGE_juci-freecwmp is not set
# CONFIG_PACKAGE_juci-igmpinfo is not set
# CONFIG_PACKAGE_juci-inteno-multiwan is not set
# CONFIG_PACKAGE_juci-inteno-router is not set
CONFIG_PACKAGE_juci-jquery-console=y
# CONFIG_PACKAGE_juci-macdb is not set
# CONFIG_PACKAGE_juci-minidlna is not set
CONFIG_PACKAGE_juci-mod-status=y
CONFIG_PACKAGE_juci-mod-system=y
# CONFIG_PACKAGE_juci-natalie-dect is not set
# CONFIG_PACKAGE_juci-netmode is not set
CONFIG_PACKAGE_juci-network-netifd=y
CONFIG_PACKAGE_juci-openwrt-wireless=y
CONFIG_PACKAGE_juci-router-openwrt=y
# CONFIG_PACKAGE_juci-samba is not set
# CONFIG_PACKAGE_juci-snmp is not set
CONFIG_PACKAGE_juci-sysupgrade=y
# CONFIG_PACKAGE_juci-upnp is not set
CONFIG_PACKAGE_juci-usb=y
CONFIG_PACKAGE_juci-theme-inteno=y
CONFIG_PACKAGE_juci=y
CONFIG_PACKAGE_juci-uhttpd=y
CONFIG_PACKAGE_juci-uhttpd-mod-ubus=y
CONFIG_PACKAGE_juci-rpcd=y
# CONFIG_PACKAGE_juci-rpcd-mod-file is not set
```

lvzaina, 2015-10-27:
I found some log
```
...
Missing type definition for config fstab
Missing type definition for config rpcd
Missing type definition for config ubootenv
Missing type definition for config uhttpd
Missing type definition for config wireless
...
```
These are wrong?

lvzaina, 2015-10-27:
```
root@OpenWrt:~# find /sys/class/net/**/*phy80211* | awk 'BEGIN{FS=\"/\"} { print
 $5 }'
awk: cmd. line:1: Unexpected token
```

lvzaina, 2015-10-27:
```
    local db_str = juci.shell("db show hw.board");
```
"db" ? What application should I install?

mkschreder, 2015-10-27:
That one will be replaced. db is a very application specific thing.

lvzaina, 2015-10-28:
hi mkschreder ,
i can't find "openwrt.wireless" in ubus list.
```
root@OpenWrt:/usr/lib/ubus/juci# ls
diagnostics       network.status    system.conf       system.upgrade
ethernet          openwrt.wireless  system.process    system.user
network           router            system.service    usb
network.lua       system            system.time
root@OpenWrt:/usr/lib/ubus/juci#
```

```
root@OpenWrt:/usr/lib/ubus/juci# ubus -v list "/juci*"
'/juci/diagnostics' @d44efb00
        "ping":{}
        "traceroute6":{}
        "traceroute":{}
        "ping6":{}
'/juci/ethernet' @a58385e8
        "adapters":{}
'/juci/network' @1dce5f8e
        "clients":{}
        "load":{}
        "nameservers":{}
        "nat_table":{}
'/juci/network.lua' @0a9776f8
        "nat_table":{}
        "clients":{}
        "services":{}
'/juci/network.status' @1c4b9156
        "ipv6routes":{}
        "ipv4routes":{}
        "arp":{}
'/juci/router' @6796b1cf
        "info":{}
'/juci/system' @4009774a
        "defaultreset":{}
        "log":{}
        "info":{}
        "filesystems":{}
        "reboot":{}
'/juci/system.conf' @679fc07b
        "restore":{}
        "clean":{}
'/juci/system.process' @fa1290f9
        "list":{}
'/juci/system.service' @0edb8bb3
        "status":{}
        "stop":{}
        "list":{}
        "start":{}
        "reload":{}
        "enable":{}
        "disable":{}
'/juci/system.time' @8d32fbdc
        "zonelist":{}
        "set":{}
        "get":{}
'/juci/system.upgrade' @bd3f9724
        "check":{}
        "test":{}
        "start":{}
        "clean":{}
'/juci/system.user' @396fb692
        "password":{}
'/juci/usb' @3f9bb4e0
        "list":{}
```

closed 2015-10-28 by lvzaina

lvzaina, 2015-10-28:
Thank you very much ~!
Now I've the solution to my problem is as follows:
### 1) Upgrade your nodejs ([url](https://nodejs.org/en/download/package-manager/))
```
$ curl --silent --location https://deb.nodesource.com/setup_4.x | sudo bash -
$ sudo apt-get install --yes nodejs
```
### 2) The default shell -> bash
```
$ sudo rm /bin/sh
$ sudo ln -s /bin/sh /bin/bash
```