mkschreder/juci #10: Build failed with latest feed

thesabbir, 2015-10-03:
Feed:
```sh
echo "src-git juci https://github.com/mkschreder/juci-openwrt-feed.git" >> feeds.conf.default
./scripts/feeds update juci
./scripts/feeds install -p juci -a
```

Make:

```sh
make package/juci/compile V=s                                                                                                                                  !1763
make[1]: Entering directory `/home/user/openwrt'
make[1]: *** No rule to make target `package/juci/compile'.  Stop.
make[1]: Leaving directory `/home/user/openwrt'
make: *** [package/juci/compile] Error 2
```
.config diff

```diff
1204,1205c1204,1205
< CONFIG_PACKAGE_libpthread=y
< CONFIG_PACKAGE_librt=y
---
> # CONFIG_PACKAGE_libpthread is not set
> # CONFIG_PACKAGE_librt is not set
1284,1330d1283
< # JUCI
< #
<
< #
< # Core
< #
< CONFIG_PACKAGE_juci=y
< CONFIG_JUCI_THEME_SELECTED=y
<
< #
< # Plugins
< #
< # CONFIG_PACKAGE_juci-asterisk is not set
< # CONFIG_PACKAGE_juci-broadcom-dsl is not set
< # CONFIG_PACKAGE_juci-broadcom-ethernet is not set
< # CONFIG_PACKAGE_juci-broadcom-vlan is not set
< # CONFIG_PACKAGE_juci-broadcom-wl is not set
< # CONFIG_PACKAGE_juci-ddns is not set
< CONFIG_PACKAGE_juci-diagnostics=y
< CONFIG_PACKAGE_juci-dnsmasq-dhcp=y
< CONFIG_PACKAGE_juci-dropbear=y
< # CONFIG_PACKAGE_juci-event is not set
< CONFIG_PACKAGE_juci-firewall-fw3=y
< # CONFIG_PACKAGE_juci-igmpinfo is not set
< # CONFIG_PACKAGE_juci-inteno-router is not set
< # CONFIG_PACKAGE_juci-jquery-console is not set
< CONFIG_PACKAGE_juci-macdb=y
< # CONFIG_PACKAGE_juci-minidlna is not set
< # CONFIG_PACKAGE_juci-mod-status is not set
< CONFIG_PACKAGE_juci-mod-system=y
< CONFIG_PACKAGE_juci-natalie-dect=y
< # CONFIG_PACKAGE_juci-netmode is not set
< # CONFIG_PACKAGE_juci-network-netifd is not set
< CONFIG_PACKAGE_juci-router-openwrt=y
< # CONFIG_PACKAGE_juci-samba is not set
< CONFIG_PACKAGE_juci-sysupgrade=y
< CONFIG_PACKAGE_juci-theme-inteno=y
< # CONFIG_PACKAGE_juci-upnp is not set
< # CONFIG_PACKAGE_juci-usb is not set
<
< #
< # Themes
< #
< CONFIG_PACKAGE_juci-theme-default=y
< CONFIG_JUCI_THEME_INTENO=y
<
< #
1830c1783
< CONFIG_PACKAGE_lua=y
---
> # CONFIG_PACKAGE_lua is not set
1834d1786
< # CONFIG_PACKAGE_lua-examples is not set
2110c2062
< CONFIG_PACKAGE_liblua=y
---
> # CONFIG_PACKAGE_liblua is not set
2200c2152
< CONFIG_PACKAGE_libubox-lua=y
---
> # CONFIG_PACKAGE_libubox-lua is not set
2202c2154
< CONFIG_PACKAGE_libubus-lua=y
---
> # CONFIG_PACKAGE_libubus-lua is not set
2204c2156
< CONFIG_PACKAGE_libuci-lua=y
---
> # CONFIG_PACKAGE_libuci-lua is not set
2220,2221c2172
< CONFIG_PACKAGE_libuv=y
< CONFIG_PACKAGE_libuv-lua=y
---
> # CONFIG_PACKAGE_libuv is not set
2240d2190
< # CONFIG_PACKAGE_luajit is not set
2930,2933d2879
< CONFIG_PACKAGE_juci-uhttpd=y
< # CONFIG_PACKAGE_uhttpd_debug is not set
< # CONFIG_PACKAGE_uhttpd_no_same_origin is not set
< CONFIG_PACKAGE_juci-uhttpd-mod-ubus=y
2945a2892
> # CONFIG_PACKAGE_uhttpd_debug is not set
2990d2936
< # CONFIG_PACKAGE_iconnect is not set
3125,3133d3070
< # UBUS
< #
< CONFIG_PACKAGE_etc-ubus-daemon=y
< CONFIG_PACKAGE_juci-rpcd=y
< # CONFIG_PACKAGE_juci-rpcd-mod-file is not set
< CONFIG_PACKAGE_juci-ubus-core=y
< CONFIG_JUCI_UBUS_CORE=y
<
< #
3344d3280
< # CONFIG_PACKAGE_network-address-from-cfe is not set

```

mkschreder, 2015-10-03:
> 1204,1205c1204,1205< CONFIG_PACKAGE_libpthread=y< CONFIG_PACKAGE_librt=y---> # CONFIG_PACKAGE_libpthread is not set> # CONFIG_PACKAGE_librt is not set1284,1330d1283< # JUCI< #<< #< # Core< #< CONFIG_PACKAGE_juci=y< CONFIG_JUCI_THEME_SELECTED=y<< #< # Plugins< #< # CONFIG_PACKAGE_juci-asterisk is not set< # CONFIG_PACKAGE_juci-broadcom-dsl is not set< # CONFIG_PACKAGE_juci-broadcom-ethernet is not set< # CONFIG_PACKAGE_juci-broadcom-vlan is not set< # CONFIG_PACKAGE_juci-broadcom-wl is not set< # CONFIG_PACKAGE_juci-ddns is not set< CONFIG_PACKAGE_juci-diagnostics=y< CONFIG_PACKAGE_juci-dnsmasq-dhcp=y< CONFIG_PACKAGE_juci-dropbear=y< # CONFIG_PACKAGE_juci-event is not set< CONFIG_PACKAGE_juci-firewall-fw3=y< # CONFIG_PACKAGE_juci-igmpinfo is not set< # CONFIG_PACKAGE_juci-inteno-router is not set< # CONFIG_PACKAGE_juci-jquery-console is not set< CONFIG_PACKAGE_juci-macdb=y< # CONFIG_PACKAGE_juci-minidlna is not set< # CONFIG_PACKAGE_juci-mod-status is not set< CONFIG_PACKAGE_juci-mod-system=y< CONFIG_PACKAGE_juci-natalie-dect=y< # CONFIG_PACKAGE_juci-netmode is not set< # CONFIG_PACKAGE_juci-network-netifd is not set< CONFIG_PACKAGE_juci-router-openwrt=y< # CONFIG_PACKAGE_juci-samba is not set< CONFIG_PACKAGE_juci-sysupgrade=y< CONFIG_PACKAGE_juci-theme-inteno=y< # CONFIG_PACKAGE_juci-upnp is not set< # CONFIG_PACKAGE_juci-usb is not set<< #< # Themes< #< CONFIG_PACKAGE_juci-theme-default=y< CONFIG_JUCI_THEME_INTENO=y<< #1830c1783< CONFIG_PACKAGE_lua=y---> # CONFIG_PACKAGE_lua is not set1834d1786< # CONFIG_PACKAGE_lua-examples is not set2110c2062< CONFIG_PACKAGE_liblua=y---> # CONFIG_PACKAGE_liblua is not set2200c2152< CONFIG_PACKAGE_libubox-lua=y---> # CONFIG_PACKAGE_libu [... cut ...] 36< # CONFIG_PACKAGE_iconnect is not set3125,3133d3070< # UBUS< #< CONFIG_PACKAGE_etc-ubus-daemon=y< CONFIG_PACKAGE_juci-rpcd=y< # CONFIG_PACKAGE_juci-rpcd-mod-file is not set< CONFIG_PACKAGE_juci-ubus-core=y< CONFIG_JUCI_UBUS_CORE=y<< #3344d3280< # CONFIG_PACKAGE_network-address-from-cfe is not set

make package/feeds/juci/juci/compile

mkschreder, 2015-10-03:
Actually you can check out my juci branch at
https://github.com/mkschreder/openwrt.git. It has all necessary patches now
for stock openwrt in order to compile an image with juci.

Many features of juci do not work properly on stock openwrt right now
though because of pimary focus on broadcom firmwares. But I'm assuming you
already know that..

thesabbir, 2015-10-03:
Instead of patched ubus, httpd, rpcd. Packages like juci-ubus juci-httpd, juci-rpc would have been better. 
Some queries:
* Why focus on broadcom?
* How much space does it takes (modules enough to replace luci-admin-full)

I was actually going to try out and take some idea. I am currently working on a mobile app for OpenWRT:
https://github.com/vertinteractive/oconnect

mkschreder, 2015-10-04:
I was able to run it on a hootoo with 4mb of flash. although without
modules like macdb that take space. Since the juci core is tiny, I think if
one puts some effort into keeping space usage to minimum then it can be
kept quite small. Main things that take up space are images and fonts and
they are theme speciffic.

I was actually thinking about juci-ubus etc, but I don't think that it is a
good idea in the long run because many packages depend on default ubus and
so we will need to have that as well. It is better to add extra things and
not create redundant packages with different names. That would lead to
having to maintain two packages for base package as well as conflicts
between them. I think a more reasonable future is that the patches will
mature and then eventually be included upstream as well.

The focus on broadcom is simply because my dayjob is working with broadcom
platform right now. So most of my time is spent on that. But I think there
is plenty of code online already for someone to implement support for
generic linux (ie using iw for wireless info etc) so it is just a matter of
writing the code to support that. Juci is also slowly moving towards being
completely modular and platform independent.

closed 2015-10-28 by mkschreder