mkschreder/juci #11: How to debug juci in host

congallenwang, 2015-10-13:
hi Schreder,

your juci project is great, now i can run it with DD in both X86/Mips machine. next i would like to learn develop my own view into Juci, and i prefer to  debug in host which is better and simple.
but when i run /juci-local-server on host, then the browser will complained 'Uncaught TypeError: Cannot read property 'system' of undefined'. i think it because no ubus service running properly in host.
would you give some idea how to debug juci on host? thanks.

br
Allen

mkschreder, 2015-10-13:
Usually you would have a working ubus running somewhere to which the locally running website can connect to. But yes it is possible to run all necessary programs completely locally and have a ubus instance running locally as well. For that you will need to compile ubusd, uci, rpcd and ubus-scriptd and run them locally. I would however much rather recommend that you build some sort of user mode linux or virtual machine image and run these services in their openwrt environment for best results.

mkschreder, 2015-10-13:
If you are lazy then I have prepared a dedicated juci branch of openwrt DD in https://github.com/mkschreder/openwrt.git. Don't forget to checkout juci branch and then select juci core package through menuconfig.

closed 2015-10-28 by mkschreder