mkschreder/juci #13: Build failed.(luaposix-release-v33.3.1)

lvzaina, 2015-11-04:
```
CC       ext/posix/posix.lo
In file included from ext/posix/posix.c:25:0:
ext/posix/sched.c: In function 'Psched_setscheduler':
ext/posix/sched.c:74:9: error: variable 'sched_param' has initializer but incomplete type
  struct sched_param sched_param = {0};
         ^
ext/posix/sched.c:74:9: warning: excess elements in struct initializer [enabled by default]
ext/posix/sched.c:74:9: warning: (near initialization for 'sched_param') [enabled by default]
ext/posix/sched.c:74:21: error: storage size of 'sched_param' isn't known
  struct sched_param sched_param = {0};
                     ^
ext/posix/sched.c:76:28: error: 'SCHED_OTHER' undeclared (first use in this function)
  int policy = optint(L, 2, SCHED_OTHER);
                            ^
ext/posix/sched.c:76:28: note: each undeclared identifier is reported only once for each function it appears in
ext/posix/sched.c:79:2: warning: implicit declaration of function 'sched_setscheduler' [-Wimplicit-function-declaration]
  return pushresult(L, sched_setscheduler(pid, policy, &sched_param), NULL);
  ^
In file included from ext/posix/posix.c:30:0:
ext/posix/sys/resource.c: In function 'luaopen_posix_sys_resource':
ext/posix/sys/resource.c:155:2: warning: overflow in implicit constant conversion [-Woverflow]
  LPOSIX_CONST( RLIM_INFINITY );
  ^
ext/posix/sys/resource.c:157:2: warning: overflow in implicit constant conversion [-Woverflow]
  LPOSIX_CONST( RLIM_SAVED_CUR );
  ^
ext/posix/sys/resource.c:160:2: warning: overflow in implicit constant conversion [-Woverflow]
  LPOSIX_CONST( RLIM_SAVED_MAX );
make[5]: *** [ext/posix/posix.lo] Error 1
```

Add "#include <pthread.h>" in luaposix-release-v33.3.1/ext/posix/sched.c file, it can be resolved.

mkschreder, 2015-11-04:
> Add "#include " in luaposix-release-v33.3.1/ext/posix/sched.c file, it can

Interesting. I do not have that problem in my builds. What was the fix you
did to make it work for you? Can you use git format-patch and send me a
patch for luaposix33 package in the juci feed?

lvzaina, 2015-11-10:
Is that OK?
```
From d59051c9a1413d38f5d4add71ff71e4a9f33c065 Mon Sep 17 00:00:00 2001
From: lvzaina <lvzaina@gmail.com>
Date: Tue, 10 Nov 2015 16:31:50 +0800
Subject: [PATCH] Build failed.(luaposix-release-v33.3.1)

---
 ext/posix/sched.c | 2 ++
 1 file changed, 2 insertions(+)

diff --git a/ext/posix/sched.c b/ext/posix/sched.c
index 13b196b..9ba2cb2 100644
--- a/ext/posix/sched.c
+++ b/ext/posix/sched.c
@@ -27,6 +27,8 @@
 #include <sched.h>
 #endif
 
+#include <pthread.h>
+
 #include "_helpers.c"
 
 
-- 
1.9.1

```

mkschreder, 2015-11-10:
patch added to the package feed (branch v1.15.11).

mkschreder, 2015-11-11:
Does it build without problems for you now?

lvzaina, 2015-11-12:
It is no problem.
thank you very much!

closed 2015-11-12 by lvzaina