mkschreder/juci #16: can not change "lan" ip address

lvzaina, 2015-11-09:
```
config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdcb:5bd4:6ddb::/48'

config interface 'lan'
        option ifname 'eth0'
        option force_link '1'
        option type 'bridge'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '10'

```
I can not change "lan" ip address,
I think it's because interface 'loopback' properties in the "ipaddr" and interface 'loopback' properties "ipaddr" repeated

mkschreder, 2015-11-09:
fixed in 1.15.11rc3. Let me know if you find any other issues. :-)

lvzaina, 2015-11-09:
Many thanks!
In addition, about "wlctl", where I can find this command?

mkschreder, 2015-11-09:
only on broadcom hardware. It talks directly to the broadcom wireless
driver.

closed 2015-11-10 by lvzaina