mkschreder/juci #17: [juci-openwrt-wireless]: Sometimes I do some operations, "ubus call / juci / wireless devices" will get incorrect results.

lvzaina, 2015-11-10:
### Sometimes I do some operations, "ubus call / juci / wireless devices" will get incorrect results.

Just start time (first start):
```
~# ubus call /juci/wireless devices
{
	"devices": [
		{
			"encryption": {
				"enabled": true,
				"auth_algs": [
					"OPEN"
				],
				"description": "WPA2 PSK (CCMP)",
				"wep": false,
				"auth_suites": [
					"PSK"
				],
				"wpa": 2,
				"pair_ciphers": [
					"CCMP"
				],
				"group_ciphers": [
					"CCMP"
				]
			},
			"bssid": "00:11:22:33:44:55",
			"hwmodes": {
				"a": false,
				"b": true,
				"ac": false,
				"g": true,
				"n": true
			},
			"mode": "Master",
			"mbssid_support": true,
			"type": "nl80211",
			"ssid": "test_123",
			"country": "US",
			"device": "wlan0",
			"txpower": 26,
			"channel": 11,
			"noise": -95,
			"frequency": 2462
		}
	]
}
```
After modifying lan ip address:
```
~# ubus call /juci/wireless devices
{
	"devices": [
		{
			"hwmodes": {
				"a": false,
				"b": true,
				"ac": false,
				"g": true,
				"n": true
			},
			"mode": "Unknown",
			"mbssid_support": true,
			"type": "nl80211",
			"country": "US",
			"device": "radio0"
		}
	]
}
```

Why "wlan0" will become "radio0"?

lvzaina, 2015-11-10:
ok, I added the " _option ifname 'wlan0'_ " in mac80211.sh file.
Problem solved.

mkschreder, 2015-11-10:
The device name comes from /sys/class/net/. You can check there to see if
you get weird behavior there.

The reason for missing firelds right after you make changes to config is
orobably because wireless is restarting so fields like ssid have not been
set yet when the call is made.

closed 2015-11-10 by lvzaina