mkschreder/juci #19: [juci-openwrt-wireless]: Sometimes "/juci/wireless clients" will get incorrect results.

lvzaina, 2015-11-10:
```
~$ curl -X POST -H "Content-Type: application/json" -d '{"jsonrpc": "2.0", "method":"call","params":["6d2f27bd7e596cda7aa1182029d7b59c","/juci/wireless","clients",{}], "id":"1"}' http://192.168.1.1/ubus
{"jsonrpc":"2.0","id":"1","result":[0,{"clients":[{"rx_bytes":21851,"authenticated":true,"tx_bytes":22243}],"error":"could not add json","json":"{\"clients\":[{\"rx_bytes\":21851,\"authenticated\":true,\"tx_bytes\":22243,\"tx_bitrate\":21.7,\"tx_packets\":175,\"tx_retries\":33,\"signal_avg\":-51,\"rx_packets\":385,\"device\":\"wlan0\",\"inactive_time\":18550,\"tdls_peer\":false,\"tx_failed\":0,\"wmm_wme\":true,\"rx_bitrate\":65,\"authorized\":true,\"preamble\":\"short\",\"expected_throughput\":\"1.473Mbps\",\"macaddr\":\"00:11:22:33:44:55\",\"mfp\":false,\"signal\":-51}]}"}]}
```
*"error":"could not add json"*
Sometimes it is normal, very strange

mkschreder, 2015-11-10:
try to run it as "/usr/lib/ubus/juc/wireless clients" and see if it prints
out anything other than just json as ouput.

lvzaina, 2015-11-10:
_/usr/lib/ubus/juci/wireless clients_ : **each is normal**
_curl -X POST ..._ : **most of them are wrong**

mkschreder, 2015-11-10:
check output of ubus call hostapd.<wliface> get_clients 

where wliface is name of each interface from /sys/class/net/ which has wireless extension on it.

lvzaina, 2015-11-10:
It's ok.
```
~# ubus call hostapd.wlan0 get_clients
{
	"freq": 2462,
	"clients": {
		"00:11:22:33:44:55": {
			"auth": true,
			"assoc": true,
			"authorized": true,
			"preauth": false,
			"wds": false,
			"wmm": true,
			"ht": true,
			"vht": false,
			"wps": false,
			"mfp": false,
			"aid": 1
		}
	}
}
```

lvzaina, 2015-11-10:
```
~# /usr/lib/ubus/juci/wireless clients
{"clients":[{"rx_bytes":16476,"authenticated":true,"tx_bytes":16487,"tx_bitrate":28.9,"tx_packets":135,"tx_retries":14,"signal_avg":-51,"rx_packets":293,"device":"wlan0","inactive_time":26530,"tdls_peer":false,"tx_failed":0,"wmm_wme":true,"rx_bitrate":58.5,"authorized":true,"preamble":"short","expected_throughput":"1.848Mbps","macaddr":"00:11:22:33:44:55","mfp":false,"signal":-51}]}
```

mkschreder, 2015-11-10:
what exactly is the issue? /usr/lib/ubus/juci/wireless clients returns extra info about signal strength that it gets from iw station info.

lvzaina, 2015-11-11:
hi mkschreder
I read this part of the code, not very clear.
Why blobmsg_add_json_from_string will return false?

*ubus-scriptd-d7cf4e1e820365073684f0e095e831ca7df0a360/src/script_object.c +93*
```
	if(stat(fname, &st) == 0){
		int exit_code = 0; 
		const char *resp = script_object_run_command(self, "%s %s '%s'", &exit_code, fname, method, blobmsg_format_json(msg, true)); 
		if(!blobmsg_add_json_from_string(&buf, resp)){
			blobmsg_add_string(&buf, "error", "could not add json"); 
			blobmsg_add_string(&buf, "json", resp); 
			return ubus_send_reply(ctx, req, buf.head); 
		}
	}
```

mkschreder, 2015-11-11:
It will return false if string is not valid json. Which may happen
sometimes. Reasons:

- Script author does not pipe stderr to null for juci.shell commands and it
messes up json.
- Json contains some construct that blobmsg_add_json_from_string does not
support (need to look more into this one if there are any..)
- script crashes during execution and dumps an error instead of valid json

and perhaps other reasons too.. I suspect also that json.encode in juci lua
lib does not correctly escape json in some cases, but this is rare and I
have not had much time to look into exactly which strings fail and at first
glance everything looks fine.. Anyway, there is something fishy going on
somewhere in this chain.

Planned fix is to eliminate the step of parsing stdout for lua scripts
completely and just have them return a lua object that will be directly
sent as response on ubus. The parsing step was originally intended for
being able to handle shell scripts and arbitrary programs as cgi on ubus
and for that it fits the purpose. For lua though, I probably will change it
in next version.

closed 2015-12-02 by mkschreder