mkschreder/juci #20: Any interest in hosting built packages?

max-b, 2015-11-10:
Some folks were just mentioning that they wanted to check out your UI, but didn't to build an entire kernel to test it out.

I know that a lot of your development workflow includes people customizing their own builds, but do you think it would be possible to host built packages for maybe just a handful of the most popular architectures?

mkschreder, 2015-11-11:
User Mode Linux could be an option..

mkschreder, 2015-11-11:
Ok, so I've successfully built a uml image. Please try it out from here https://github.com/mkschreder/juci/blob/uml/juci-1.15.11rc3-uml-x86_64.tar.gz?raw=true

You will need to install uml-utilities and then do this on host: 
# create a tap interface
tuntap -u <your user id> (eg. tuntap -u 1000)
# configure it to have ip 192.168.10.1
ifconfig tap0 192.168.10.1 up
# run the startup script
./run.sh

Then go to 192.168.10.10 and login with admin/admin to the gui. 

If you have a chance to test this it would be great. Let me know how it works for you. I have also included my .config for openwrt (build from my juciwrt repo) as well as the files directory for this image. You should be able to build your own if you want.

roger-, 2015-11-12:
Can this be added to the OpenWrt [packages feed](https://github.com/openwrt/packages)?

mkschreder, 2015-11-12:
It has it's own feed. Currently juci is in
https://github.com/mkschreder/mks-openwrt-feed.git. Use ;v1.15.11 after
feed url to use current version of juci..

roger-, 2015-11-12:
Why not add all those packages to the official feed?

That way there'd be binaries available for everyone (in the trunk daily builds) and people could install and use juci without having to compile everything from scratch (which probably puts a lot of people off from testing it).

mkschreder, 2015-11-12:
It is also now possible to build iopsys uml images with juci from
https://github.com/mkschreder/iopsys-open-cc.git. You can find instructions
on main page.

I can probably host prebuilt images, but in that case, which platforms are
you most interested in?

mkschreder, 2015-11-12:
Roger, because currently juci needs extra functionality from openwrt which
is available in forks of core openwrt packages and it also takes a while to
get juci to surpass luci in terms of functionality. I would say we are now
85% there. So pretty soon juci can become official interface for openwrt.

roger-, 2015-11-12:
Ah okay. For my own curiosity though: what core packages have been modified? Also, much flash does OpenWrt + juci require to install?

I'm not familiar with iopsys but one day I'll play around with it, thanks.

mkschreder, 2015-11-12:
Mostly minor bugfixes. Easiest if you check the diffs. Some of them are
probably even obsolete now that juci has become much more standalone and
independent of distro you use..

I did a test at getting it into a hootoo box with 4M flash and I managed to
make it fit. You can strip out quite a few things if size is an issue and
still have a working gui. I am however not currently actively trying to
make it fit into 4M so I haven't really been putting much time into it..

mkschreder, 2015-11-15:
Roger, I have in fact now condensed all the changes and put it in the feed. But to make openwrt install overrides for core packages from the feed you need to use -f option to ./scripts/feeds.. In CC it is supported. 

So juci will work now on stock openwrt CC by doing just this: 
git clone git://git.openwrt.org/15.05/openwrt.git openwrt
cd openwrt
cp feeds.conf.default feeds.conf
echo "src-git-full juci https://github.com/mkschreder/mks-openwrt-feed.git;v1.15.11" >> feeds.conf
./scripts/feeds update -a 
./scripts/feeds install -a 
# use -f to override uhttpd, rpcd and ubus
./scripts/feeds install -f -p juci -a

You will still need to have a working juci uci config though. For this there is juci.example.config..

One more step towards making it supersimple..  

When I have time I can add a few preconfigured targets to the feed..

roger-, 2015-11-15:
That's great! Does this mean that a binary package can be made for stock that overwrites the necessary core packages?

mkschreder, 2015-11-16:
No, you add my feed and make it override stock core packages and then juci
will work as expected. So if you install the juci feed with -f option
(force overrides - see above) then you will now see in console "Overriding
package..." and then when you configure your image with juci +
juci-theme-inteno + juci-ubus-core + any other plugins you need (which do
not conflict with eachother), you will get a working image even with stock
CC now.

I will add an example openwrt config soon with all necessary options
selected...

mkschreder, 2016-02-04:
Closing this. Reopen if any questions persist.

closed 2016-02-04 by mkschreder