mkschreder/juci #21: Could not sync config hosts

lvzaina, 2015-11-13:
Where are the hosts file?

mkschreder, 2015-11-13:
It needs to be created. It is only used in the gui though for associating
data with hosts.. I know it is currently missing in default builds.

lvzaina, 2015-11-13:
Thank you!
I found the "overview" pages, "lan" widget client has not been updated, and the client is offline.

mkschreder, 2015-11-13:
It is because in overview-net.js the client list is not being updated using JUCY.interval.repeat. Easy enough to fix.. I will update tomorrow.

lvzaina, 2015-11-21:
How can I change it?

mkschreder, 2015-11-27:
I have added auto update to it, but it will still only be removed once the arp table entry expires. This means that it will not disappear right away. Although there is functionality to do realtime monitoring and it is handled by juci-netmond. It will broadcast ubus events that look like this: 
{ "juci.netmond.client.up": {"leasetime":"1448667188","ipaddr":"192.168.1.245","hostname":"host","device":"br-lan","macaddr":"11:22:33:44:55:66"} }
{ "juci.netmond.client.down": {"leasetime":"1448667188","ipaddr":"192.168.1.245","hostname":"host","device":"br-lan","macaddr":"11:22:33:44:55:66"} }

If you wish to bind javascript to these, you can do $event.subscribe("juci.netmond.client.up", function()) in juci. The default widget right now though does not use these and instead simply relies on the information retreived from arp table (and ipv6 neighbor table for ipv6) to show clients that are active on the network.

closed 2015-12-02 by mkschreder