mkschreder/juci #32: *** No rule to make target 'package/feeds/juci/libzmq/compile', needed by 'package/feeds/juci/czmq/compile'.  Stop.

ogamyster, 2015-12-27:
Hi, mkschreder!
Compilation for ar71xx permanently comes to an end with an error:
make[2]: *** No rule to make target 'package/feeds/juci/libzmq/compile', needed by 'package/feeds/juci/czmq/compile'.  Stop.
make[2]: Leaving directory '/root/trunk'
package/Makefile:188: recipe for target '/root/trunk/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile' failed
make[1]: *** [/root/trunk/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/trunk'
/root/trunk/include/toplevel.mk:177: recipe run error for the purpose of "world"
make: *** [world] Error 2
What do I do not and how to correct it? Thanks)

mkschreder, 2015-12-27:
You don't currently need zmq at all. If it appears as dependency somewhere
then you can remove it for now.

And make sure you are using branch v1.15.12 of the juci feed for all latest
packages. I will update master after new years eve..

ogamyster, 2015-12-27:
Thanks for so early reply)!
I transferred branch last night. 
Where in config this ill-fated record)?
#
# JUCI
#

#
# C-backend
#
CONFIG_PACKAGE_juci-ubus-core=y

#
# Plugins
#
CONFIG_PACKAGE_juci-asterisk=m
# CONFIG_JUCI_ASTERISK is not set
# CONFIG_PACKAGE_juci-broadcom-dsl is not set
# CONFIG_JUCI_BROADCOM_DSL is not set
# CONFIG_PACKAGE_juci-broadcom-ethernet is not set
# CONFIG_JUCI_BROADCOM_ETHERNET is not set
# CONFIG_PACKAGE_juci-broadcom-vlan is not set
# CONFIG_JUCI_BROADCOM_VLAN is not set
# CONFIG_PACKAGE_juci-broadcom-wl is not set
# CONFIG_JUCI_BROADCOM_WL is not set
CONFIG_PACKAGE_juci-ddns=y
CONFIG_PACKAGE_juci-diagnostics=y
CONFIG_PACKAGE_juci-dnsmasq-dhcp=y
CONFIG_PACKAGE_juci-dropbear=y
CONFIG_PACKAGE_juci-ethernet=y
CONFIG_PACKAGE_juci-event=y
CONFIG_PACKAGE_juci-firewall-fw3=y
CONFIG_PACKAGE_juci-freecwmp=y
CONFIG_PACKAGE_juci-igmpinfo=y
CONFIG_PACKAGE_juci-inteno-multiwan=y
CONFIG_PACKAGE_juci-inteno-router=y
CONFIG_PACKAGE_juci-jquery-console=y
CONFIG_PACKAGE_juci-macdb=y
CONFIG_PACKAGE_juci-minidlna=m
CONFIG_PACKAGE_juci-mod-status=y
CONFIG_PACKAGE_juci-mod-system=y
CONFIG_PACKAGE_juci-natalie-dect=y
CONFIG_PACKAGE_juci-netmode=y
CONFIG_PACKAGE_juci-network-netifd=y
CONFIG_PACKAGE_juci-openwrt-wireless=y
CONFIG_PACKAGE_juci-router-openwrt=y
CONFIG_PACKAGE_juci-samba=m
CONFIG_PACKAGE_juci-snmp=y
CONFIG_PACKAGE_juci-sysupgrade=y
CONFIG_PACKAGE_juci-upnp=y
CONFIG_PACKAGE_juci-usb=y

#
# Themes
#
CONFIG_PACKAGE_juci-theme-inteno=y
CONFIG_PACKAGE_juci=y

ogamyster, 2015-12-27:
o_O! Simply I deleted directories)) Thanks a lot! I already was I decided that at me hands grow not from that place;)

ogamyster, 2015-12-27:
Damn it
======= JUCI BUILD ==========
./juci-compile: line 9: grunt: command not found
Makefile:121: recipe for target 'release' failed
make[4]: *** [release] Error 127
make[4]: Leaving directory '/root/trunk/build_dir/target-mips_34kc_musl-1.1.11/juci-322aacc2bd71fc5382de56fc7eeb88c455c821d9'
Makefile:107: recipe for target '/root/trunk/build_dir/target-mips_34kc_musl-1.1.11/juci-322aacc2bd71fc5382de56fc7eeb88c455c821d9/.built' failed
make[3]: *** [/root/trunk/build_dir/target-mips_34kc_musl-1.1.11/juci-322aacc2bd71fc5382de56fc7eeb88c455c821d9/.built] Error 2
make[3]: Leaving directory '/root/trunk/feeds/juci/juci'
package/Makefile:191: recipe for target 'package/feeds/juci/juci/compile' failed
make[2]: *** [package/feeds/juci/juci/compile] Error 2
make[2]: Leaving directory '/root/trunk'

mkschreder, 2015-12-27:
There is a juciwrt repo that I have set up before which properly installs
all feeds so that you can build an image.

https://github.com/mkschreder/juciwrt.git

I have not updated it for a while though so it is probably broken now, but
you can look there for hints. For example you need to install the feed with
-f option etc. You will also need to install /etc/config/juci from
juci.config.example provided in the juci repo, before you can see menus in
the gui. Normally we have a metapackage that installs menus and selects all
necessary plugins in order to build a working image. I will upload an
example later...

Also make sure you install build prerequisites using
./scripts/ubuntu-bootstrap.sh otherwise it will not build...

mkschreder, 2015-12-27:
Yay! ;-)

ogamyster, 2015-12-27:
"Also make sure you install build prerequisites using
./scripts/ubuntu-bootstrap.sh otherwise it will not build..."
Yes! I copied a script from github and everything departed)! Thanks once again!

ogamyster, 2015-12-27:
Still question, Martin... You didn't consider an option to pack files of localizations by separate packets? Also can perhaps use without compilation? I would try to make the translation for Russian, but there is one small, BUT - I am not a programmer as you already probably marked)), and for me it is a little difficult, especially, when I up to the end don't understand that I do;)

mkschreder, 2015-12-27:
Use https://github.com/mkschreder/juci/blob/v1.15.12rc4/scripts/initlang
and https://github.com/mkschreder/juci/blob/v1.15.12rc4/scripts/packlang

I have attached packed files.

You need to translate using "poedit" and open each ru.po followed by
"catalog > update from POT file" and pick corresponding template.pot in the
same directory to get latest strings into the po file.

ogamyster, 2015-12-27:
Thanks. I will try)

ogamyster, 2015-12-31:
Hi, Martin!
I can't launch juci
root@OpenWrt:~# ubus-scriptd
_for_each_file: reading /usr/lib/ubus-services/
_for_each_file: found /usr/lib/ubus-services//juci-netdevd, type: 8
_load_service: forking service /usr/lib/ubus-services//juci-netdevd
_for_each_file: found /usr/lib/ubus-services//juci-eventd, type: 8
_load_service: forking service /usr/lib/ubus-services//juci-eventd
_for_each_file: found /usr/lib/ubus-services//juci-netmond, type: 8
_load_service: forking service /usr/lib/ubus-services//juci-netmond
_service_thread: loading lua service /usr/lib/ubus-services//juci-eventd..
connected as 4ab2cebe
_service_thread: loading lua service /usr/lib/ubus-services//juci-netdevd..
_service_thread: loading lua service /usr/lib/ubus-services//juci-netmond..
_for_each_file: reading /usr/lib/ubus
_for_each_file: found /usr/lib/ubus/juci, type: 4
_for_each_file: reading /usr/lib/ubus/juci
_for_each_file: found /usr/lib/ubus/juci/network.lua, type: 8
parsing nat_table,clients,services
_parse_methods_json: warning: could not parse json!
script_object_load: unable to parse json, tryging comma list...
What did I pass?

mkschreder, 2016-02-04:
I think this was just a temporary issue. Reopen the ticket if any problems are still there.

closed 2016-02-04 by mkschreder