mkschreder/juci #33: RPC error (system.board): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}

animavitis, 2016-01-12:
Hi
I was able to build openwrtCC with JUCI for virtualbox and successfully run it. But after login page i get in console:

RPC error (system.board): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}request.error @ 01-juci.js:55
01-juci.js:56 Uncaught Error: [ng:areq] http://errors.angularjs.org/1.3.15/ng/areq?p0=OverviewPageCtrl&p1=not%20a%20function%2C%20got%20undefined
01-juci.js:42 Uncaught Error: [$rootScope:inprog] http://errors.angularjs.org/1.3.15/$rootScope/inprog?p0=%24digest

can u help me?

mkschreder, 2016-01-12:
take a look at access.json files in juci. There you specify which ubus
calls you want to allow access to over rpc. Most likely system.board call
is not allowed or allowed in a plugin that you did not enable. Then it
could be a bug.

animavitis, 2016-01-12:
now i have this in console:

Init UBUS -> 
01-juci.js:56 Init UCI
01-juci.js:56 init: config
01-juci.js:55 RPC succeeded (uci.revert), but returned error: {"jsonrpc":"2.0","id":1,"result":[6,{"code":6}]}: Access denied
01-juci.js:56 Synched config juci
01-juci.js:56 Using settings from config/juci on router
01-juci.js:56 juci: loading menu from server..
01-juci.js:56 CONF
01-juci.js:56 juci: angular init
01-juci.js:40 Uncaught Error: [ng:areq] http://errors.angularjs.org/1.3.15/ng/areq?p0=OverviewPageCtrl&p1=not%20a%20function%2C%20got%20undefined(anonymous function) @ 01-juci.js:40Rb @ 01-juci.js:40sb @ 01-juci.js:40(anonymous function) @ 01-juci.js:41(anonymous function) @ 01-juci.js:41r @ 01-juci.js:40B @ 01-juci.js:41g @ 01-juci.js:41(anonymous function) @ 01-juci.js:41(anonymous function) @ 01-juci.js:41k @ 01-juci.js:41Ia.link @ 01-juci.js:44$ @ 01-juci.js:41B @ 01-juci.js:41g @ 01-juci.js:41g @ 01-juci.js:41g @ 01-juci.js:41g @ 01-juci.js:41g @ 01-juci.js:41g @ 01-juci.js:41B @ 01-juci.js:41(anonymous function) @ 01-juci.js:41(anonymous function) @ 01-juci.js:42n.$eval @ 01-juci.js:42n.$digest @ 01-juci.js:42n.$apply @ 01-juci.js:42(anonymous function) @ 01-juci.js:40e @ 01-juci.js:41d @ 01-juci.js:40uc @ 01-juci.js:40(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:56(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1iterate @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:56(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1iterate @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:55j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29request.success @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
01-juci.js:40 Uncaught Error: [$rootScope:inprog] http://errors.angularjs.org/1.3.15/$rootScope/inprog?p0=%24digest

mkschreder, 2016-01-12:
How are you building? Did you enable minimal plugins and choose the default
theme? There is a makefile.local example file that gives you an example of
a package selection. You need to copy it to Makefile.local if you build
locally. Opnwrt build typically makes package selection using menuconfig,
therefore there is no need to use makefile.local when building with
openwrt.

You are most likely getting this error because you are not compiling in a
theme.

animavitis, 2016-01-13:
success :)

i was something wrnog with .config file .... i run:
 ./scrpits/feeds uninstall -a
 ./scrpits/feeds update -a
 ./scrpits/feeds install -a
 ./scrpits/feeds install -f -p juci -a
make menuconfig (and select JUCI stuff)
make

and its works!

thanks for help

closed 2016-01-13 by animavitis

mkschreder, 2016-01-13:
Awesome.

It could be that it did not properly override the packages the first time.

When installing juci feed you should lways see "Overriding package" message
for some packages.