mkschreder/juci #34: There is only overview page,no plugins' pages compiled in

blademaster98, 2016-01-13:
I selected and saved some plugin options  in menuconfig such as “ethernet mod” , “status mod” and so on..  but there is no internet-ehternet.html or other plugin html files copied in building DIR,  except "index.html".  
The .css files generated are also strange, all file sizes are zero except "01-juci.css" ,"50-minidlna.css" ,"50-juci-network-netifd.css" and “80-juci-theme-inteno.css”.
There is no use to copy file "juci.config.example“ to router, as you can see here, no html files copied firstly at all.
So, when opening JUCI in broswer, there is no navigation taps (Should have like Overview Phone Wifi Seetings Status... ) in TOP-Navigation Bar area. It's blank there with only logo IOPSYS. 
The overview page looks fine by the way. Router datas were shown normally.
Anybody help here? Thanks.

blademaster98, 2016-01-13:
Sorry My Bad! when I copy file "juci.config.example" to the router, two irrelevant lines were added at the bottom. So the file was not resolved normally I guess!

closed 2016-01-13 by blademaster98

mkschreder, 2016-01-13:
Well the menu parts of the config file are release-specific so it is not
meant to be installed automatically (I try to keep it up to date though)

It is instead meant to be installed to firmware by a special metapackage
that selects plugins, sets up users and creates menus...

It is this way because quite often a customer wants to only have a subset
of all available pages and having a static set of pages in plugins and
showing all of them does not address this problem.

If you or anyone else is interested in suggesting improvements then right
now all suggestions will be considered. I'm willing to go as far as making
juci a kind of an X server to which applications can connect and publish
gui parts. And then implement proper role based access control that will
hide parts of gui based on what user can and can not do...