mkschreder/juci #35: can not access default page

longwdl, 2016-01-25:
juci version: 1.15.12

1. access root page 192.168.1.1  
rediret to url: http://192.168.1.1/#!/login, then a js alert displayed.  
  ```
Error: [$injector:unpr] http://errors.angularjs.org/1.3.15/$injector/unpr?p0=%24eventsProvider%20%3C-%20%24events
```

2. in the console, there are other errors  
  ```
RPC error (uci.revert): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
Could not sync config netmode
Could not sync config hosts
Could not sync config firewall
```

can someone help me to go through it?

animavitis, 2016-01-25:
I had the same problem: please read https://github.com/mkschreder/juci/issues/33
I hope this will help:)

longwdl, 2016-01-26:
After change the version to master:98b8b35，I can see the login page。But there still some problem after login. I can see two js alert.

### console log
```
50-juci-jquery-console.js:3 juci: angular init
01-juci.js:40 Uncaught Error: [$injector:unpr] http://errors.angularjs.org/1.3.15/$injector/unpr?p0=%24eventsProvider%20%3C-%20%24events%20%3C-%20NavigationCtrl(anonymous function) @ 01-juci.js:40(anonymous function) @ 01-juci.js:41d @ 01-juci.js:41(anonymous function) @ 01-juci.js:41d @ 01-juci.js:41e @ 01-juci.js:41w.instance @ 01-juci.js:41(anonymous function) @ 01-juci.js:41r @ 01-juci.js:40B @ 01-juci.js:41(anonymous function) @ 01-juci.js:41g @ 01-juci.js:41(anonymous function) @ 01-juci.js:41(anonymous function) @ 01-juci.js:41k @ 01-juci.js:41Ia.link @ 01-juci.js:44$ @ 01-juci.js:41B @ 01-juci.js:41g @ 01-juci.js:41B @ 01-juci.js:41g @ 01-juci.js:41B @ 01-juci.js:41(anonymous function) @ 01-juci.js:41(anonymous function) @ 01-juci.js:42n.$eval @ 01-juci.js:42n.$digest @ 01-juci.js:42n.$apply @ 01-juci.js:42(anonymous function) @ 01-juci.js:40e @ 01-juci.js:41d @ 01-juci.js:40uc @ 01-juci.js:40(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:56(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1iterate @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:56(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1iterate @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:55j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29request.success @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
(index):1 Font from origin 'http://bootstrap-live-customizer.com' has been blocked from loading by Cross-Origin Resource Sharing policy: No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://192.168.1.1' is therefore not allowed access.
01-juci.js:40 Uncaught Error: [$rootScope:inprog] http://errors.angularjs.org/1.3.15/$rootScope/inprog?p0=%24digest
```

mkschreder, 2016-01-26:
Current devel is branch v2.16.01rc2 which will be merged into master soon.

mkschreder, 2016-01-26:
As an open poll: do you think it would make sense for juci to have it's own
backend server that would replace uhttpd, rpcd, ubus and ubus-scriptd, and
will make juci websocket based and only dependent on a single backend
server for improved portability and stability?

animavitis, 2016-01-31:
I think It would be great:)

mkschreder, 2016-02-04:
Current devel branch is now v2.16.02rc1

closed 2016-02-04 by mkschreder