mkschreder/juci #36: How to use Juci?

darcyg, 2016-02-15:
hi martin:
  How to use Juci?
  Both Juci and Luci can be used at the same time.
  I need add line
  Https://github.com/mkschreder/openwrt-feed src-git
  Https://github.com/mkschreder/juci-openwrt-feed src-git
   into Feeds.conf ?
thks

mkschreder, 2016-02-15:
You will have problem with two different http servers (lighttpd and uhttpd)
running at the same time trying to listen on the same port. If you can
configure them to run on different ports then it should work fine.

I will soon be removing dependency on uhttpd and you can then use any
server that has websocket support (websocket support for rpc calls is in
the making too).

On OpenWRT:
- add feed src-git juci https://github.com/mkschreder/juci-openwrt-feed.git
- ./scripts/feeds update -a
- ./scripts/feeds install -f -a -p juci
- make menuconfig
- make

For quickly enabling juci you can do this:
cat >> .config << END
CONFIG_PACKAGE_juci-ubus-core=y
CONFIG_PACKAGE_juci-ddns=y
CONFIG_PACKAGE_juci-dnsmasq-dhcp=y
CONFIG_PACKAGE_juci-dropbear=y
CONFIG_PACKAGE_juci-ethernet=y
CONFIG_PACKAGE_juci-event=y
CONFIG_PACKAGE_juci-firewall-fw3=y
CONFIG_PACKAGE_juci-freecwmp=y
CONFIG_PACKAGE_juci-minidlna=y
CONFIG_PACKAGE_juci-mod-status=y
CONFIG_PACKAGE_juci-mod-system=y
CONFIG_PACKAGE_juci-network-netifd=y
CONFIG_PACKAGE_juci-openwrt-wireless=y
CONFIG_PACKAGE_juci-samba=y
CONFIG_PACKAGE_juci-simple-gui=y
CONFIG_PACKAGE_juci-snmpd=y
CONFIG_PACKAGE_juci-sysupgrade=y
CONFIG_PACKAGE_juci-uhttpd=y
CONFIG_PACKAGE_juci-upnp=y
CONFIG_PACKAGE_juci-usb=y
CONFIG_PACKAGE_juci-theme-inteno=y
CONFIG_PACKAGE_juci=y
CONFIG_PACKAGE_juci-full-openwrt=y
END
make defconfig

If you still have problems then let me know which target you want to build
for and I'll send you a ready made .config file for openwrt and you can
then start with that..

darcyg, 2016-02-16:
I would also like to transplant Juci to other embedded systems. For example, s3c2440.
No routing function. Requires only WebUI system and RPC.
Can provide the lightest amount of system Demo?
thks!

mkschreder, 2016-02-16:
First get json backend on the box then add a webserver and copy over the
html/js application. I think it will be easier once I get the new websocket
based server in order. It will have everything in one place. I really don't
like having to depend on ubus for such basic things as session handling,
which should be made part of the application server.

The new backend server is here: https://github.com/mkschreder/jucid. Very
early beta though. It's just now taking shape. It is going to run lua
applications that will talk json over a websocket. The current juci
frontend can be put on top and simply use websocket rpc for all
communication with the backend. Even juci backend is already mostly lua so
it will only require minor changes to be loaded by the new server.
Precompiled lua is rather small and very efficient for embedded work.

If you want to collaborate on it by making something small for your arm9
platform then it would be quite useful I think because you will be taking
the project in a different direction from currently main target (routers).
Feel free to first play with a rather stable juci build from the feed and
then you can think about how the new server can make your app smaller and
more efficient.

darcyg, 2016-02-17:
Thks
I compiled JUCI to Openwrt 15.05. Is the most basic system.
There are the following questions:
1) /www directory, JUCI all files are.Gz format.
2) I am using uhttpd. is not juci-uhttpd. I want to put the openwrt source directory RPCD, ubox, UBUS, uhttpd and other projects. Replaced by juci-rpcd, juci-ubox, juci-ubus, juci-uhttpd related projects.
3) ubus2 seems to be a new project. Now can you use it?
4) I extract the /www directory under the *.gz file. Login prompt after use:

> `You do not have an RPC connection to the router!`
5) 5) start /etc/init.d/juci error

so I don't know how to start the Juci backend service.

darcyg, 2016-02-17:
replace rpcd,ubox,ubus,uhttpd project.
make error:

> Configuring kmod-e1000e.
Configuring kmod-ath9k-htc.
Configuring usbreset.
Configuring kmod-iwlagn.
Configuring kmod-button-hotplug.
Configuring kmod-net-ipw2100.
Configuring kmod-bluetooth_6lowpan.
Configuring ppp-mod-pppoe.
Collected errors:
 * opkg_install_cmd: Cannot install package libs.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for juci-full-openwrt:
 * 	juci-ddns * 	juci-dnsmasq-dhcp * 	juci-dropbear * 	juci-ethernet * 	juci-event * 	juci-firewall-fw3 * 	juci-simple-gui * 	juci-macdb * 	juci-minidlna * 	juci-mod-status * 	juci-mod-system * 	juci-network-netifd * 	juci-openwrt-wireless * 	juci-samba * 	juci-snmpd * 	juci-sysupgrade * 	juci-ubus-core * 	juci-uhttpd * 	juci-upnp * 	juci-usb * 	juci-utils * 	juci-igmpinfo * 	juci-theme-inteno *

mkschreder, 2016-02-17:
ubus2 is not used. gz is fine because they are served as is (as gz
compressed with correct mime type by juci-uhttpd and decompressed
automatically by browser). Current feed requires you to install override
packages (use ./scripts install -f -a -p juci command).

mkschreder, 2016-02-17:
you can do opkg install --force-depends if you install package separately.
If you build a full image though it should work fine.

Cannot install package "libs"? Looks like something weird is going on in
your build.

Easiest way to copy files over is probably to not use package system at all
and just do make in juci git repo and then copy all files put under ./bin
folder to root (scp -r ./bin/* root@box:/). Troubleshooting can proceed as
follows: restart ubus-scriptd, restart rpcd, make sure rpcd has right
config in /etc/config/rpcd (at least default one), make sure juci menu is
placed in /etc/config/juci (on full image it is generated by uci-defaults
script that can detect various subsystems and add corresponding menus to
menu structure).

Even local testing is possible meaning that you only need working rpc
backend on the box (and load all www files locally). You can then do
./juci-local-server --host 192.168.1.1 --port 1234 and then go to
localhost:1234 in browser and log in.

darcyg, 2016-02-17:
don't login system.

my openwrt system: x86

shell ps command: 
> ... 
 700 root         0 SW<  [kworker/0:1H]
 1269 root       688 S    /sbin/ubusd
 1299 root      1080 S    /bin/ash --login
 1714 root         0 SW<  [crypto]
 1737 root         0 SW<  [ipv6_addrconf]
 1761 root         0 SW<  [rpciod]
 1830 root         0 SW<  [nfsiod]
 1834 root         0 SW<  [bat_events]
 1837 root         0 SW<  [cryptodev_queue]
 1878 root         0 SW<  [cfg80211]
 2027 root         0 SW<  [krfcommd]
 2239 root       772 S    /sbin/logd -S 16
 2248 root       984 S    /sbin/rpcd
 2257 root      1124 S    /sbin/ubus-scriptd
 2269 root         0 Z    [juci-ubus-core-]
 2270 root      1800 S    {juci-eventd} /sbin/ubus-scriptd
 2272 root      1768 S    {juci-netmond} /sbin/ubus-scriptd
 2273 root      1728 S    {juci-switchd} /sbin/ubus-scriptd
 2274 root      1700 S    {juci-utils-spee} /sbin/ubus-scriptd
 2406 root      1036 S    /sbin/netifd
 2417 root       844 S    /usr/sbin/odhcpd
 2511 root       632 S    /usr/sbin/autossh -M 20000    -i /etc/dropbear/id_rsa -N -T -R 2222:localhost:22 user@host
 2571 root       872 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300
 2585 nobody     812 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf -k -x /var/run/dnsmasq/dnsmasq.pid
 2670 root      1060 S    /usr/sbin/telnetd -F -l /bin/login.sh
 2684 root      1040 S    /usr/sbin/uhttpd -f -h /www -r OpenWrt -x /cgi-bin -u /ubus -t 60 -T 30 -k 20 -A 1 -n 3 -N 100
 2716 root      1132 S    /usr/bin/rsync --daemon --no-detach
 2735 root      1064 S    /usr/sbin/ntpd -n -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.or
 7418 root       940 R    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300
...

my browser console error:

> Uncaught Error: Section mini_snmpd already defined. Please fix your code!
01-juci.js:55 Init UBUS -> 
50-juci-jquery-console.js:3 Init UCI
50-juci-jquery-console.js:3 RPC error (uci.configs): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}console.error @ 50-juci-jquery-console.js:3
50-juci-jquery-console.js:3 UCI failed to initialize!console.error @ 50-juci-jquery-console.js:3
50-juci-jquery-console.js:3 init: config
50-juci-jquery-console.js:3 RPC error (system.board): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}console.error @ 50-juci-jquery-console.js:3request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
Navigated to http://192.168.1.1/
01-juci.js:56 Uncaught Error: Section mini_snmpd already defined. Please fix your code!UCIConfig.$registerSectionType @ 01-juci.js:56(anonymous function) @ 50-juci-snmpd.js:1
01-juci.js:55 Init UBUS -> 
50-juci-jquery-console.js:3 Init UCI
50-juci-jquery-console.js:3 RPC error (uci.configs): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}console.error @ 50-juci-jquery-console.js:3request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
50-juci-jquery-console.js:3 UCI failed to initialize!console.error @ 50-juci-jquery-console.js:3(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
50-juci-jquery-console.js:3 init: config
50-juci-jquery-console.js:3 RPC error (system.board): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}console.error @ 50-juci-jquery-console.js:3request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
50-juci-jquery-console.js:3 RPC error (uci.revert): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}console.error @ 50-juci-jquery-console.js:3request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
50-juci-jquery-console.js:3 RPC error (uci.get): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}console.error @ 50-juci-jquery-console.js:3request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
50-juci-jquery-console.js:3 Could not sync config juciconsole.error @ 50-juci-jquery-console.js:3(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
[... cut ...]
        list languages 'fi'
        list languages 'sv'
>
>config juci_event 'juci_event'
        list filter firewall.ssh
        list filter network.netifd
        list filter network.dnmasq
        list filter system.kernel
        list filter system.ntp
        list filter iptv.mcpd
>
>config menu
        option path 'overview'
        option page 'overview'
>
>config menu
        option path 'internet'
        option page 'network'
        option redirect 'internet-ethernet'
        list acls 'user-support'

mkschreder, 2016-02-17:
use $p$admin as admin password. You are setting it wrong. If you want to
set directly then set it to a hash. But if you use $p$admin then openwrt
rpcd will use password for user admin. You can set it using "passwd admin".

darcyg, 2016-02-18:
My system does not have a admin account. I tested with root account, login authentication error.
Now I can't test. 3 hours later, I try again.
Load page is very slow. Why?

darcyg, 2016-02-18:
Root account or admin account can not log on.

`You do not have an RPC connection to the router!`

>Failed to load resource: the server responded with a status of 404 (Not Found)
Navigated to http://localhost/
favicon.ico:1 GET http://localhost/favicon.ico 404 (Not Found)
3Navigated to http://192.168.1.1/
01-juci.js:55 Init UBUS -> 
01-juci.js:56 Init UCI
01-juci.js:55 RPC error (uci.configs): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
01-juci.js:56 UCI failed to initialize!(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
01-juci.js:56 init: config
01-juci.js:30 POST http://192.168.1.1/ubus net::ERR_CONNECTION_RESETk.cors.a.crossDomain.send @ 01-juci.js:30n.extend.ajax @ 01-juci.js:30ajax @ 50-juci-sysupgrade.js:1$.extend.jsonRPC._doRequest @ 01-juci.js:53$.extend.jsonRPC.request @ 01-juci.js:53(anonymous function) @ 01-juci.js:55$.extend.jsonRPC.withOptions @ 01-juci.js:53rpc_request @ 01-juci.js:55obj.(anonymous function) @ 01-juci.js:55(anonymous function) @ 01-juci.js:56(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1iterate @ 01-juci.js:1async.eachSeries @ 01-juci.js:1_asyncMap @ 01-juci.js:1(anonymous function) @ 01-juci.js:1async.series @ 01-juci.js:1JUCIConfig.$init @ 01-juci.js:56(anonymous function) @ 01-juci.js:56(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1iterate @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
01-juci.js:55 RPC error (system.board): {"error":"JSONRPC: server returned empty string","version":"2.0"}request.error @ 01-juci.js:55$.extend.jsonRPC._requestError @ 01-juci.js:54$.ajax.error @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
01-juci.js:55 RPC error (uci.revert): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
01-juci.js:30 POST http://192.168.1.1/ubus net::ERR_CONNECTION_RESETk.cors.a.crossDomain.send @ 01-juci.js:30n.extend.ajax @ 01-juci.js:30ajax @ 50-juci-sysupgrade.js:1$.extend.jsonRPC._doRequest @ 01-juci.js:53$.extend.jsonRPC.request @ 01-juci.js:53(anonymous function) @ 01-juci.js:55$.extend.jsonRPC.withOptions @ 01-juci.js:53rpc_request @ 01-juci.js:55obj.(anonymous function) @ 01-juci.js:55(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29request.error @ 01-juci.js:55$.extend.jsonRPC._requestSuccess @ 01-juci.js:54$.ajax.success @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
01-juci.js:55 RPC error (uci.get): {"error":"JSONRPC: server returned empty string","version":"2.0"}request.error @ 01-juci.js:55$.extend.jsonRPC._requestError @ 01-juci.js:54$.ajax.error @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
01-juci.js:56 Could not sync config juci(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29request.error @ 01-juci.js:55$.extend.jsonRPC._requestError @ 01-juci.js:54$.ajax.error @ 01-juci.js:53j @ 01-juci.js:29k.fireWith @ 01-juci.js:29x @ 01-juci.js:30(anonymous function) @ 01-juci.js:30
01-juci.js:56 Using settings from config/juci on router
01-juci.js:30 POST http://192.168.1.1/ubus net::ERR_CONNECTION_RESETk.cors.a.crossDomain.send @ 01-juci.js:30n.extend.ajax @ 01-juci.js:30ajax @ 50-juci-sysupgrade.js:1$.extend.jsonRPC._doRequest @ 01-juci.js:53$.extend.jsonRPC.request @ 01-juci.js:53(anonymous function) @ 01-juci.js:55$.extend.jsonRPC.withOptions @ 01-juci.js:53rpc_request @ 01-juci.js:55obj.(anonymous function) @ 01-juci.js:55rpc.$authenticate @ 01-juci.js:55(anonymous function) @ 01-juci.js:56(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1iterate @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:1(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29k.fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 0 [... cut ...]
01-juci.js:56 GET http://192.168.1.1/fonts/fontawesome-webfont.woff2?v=4.3.0 net::ERR_CONNECTION_RESET

I don't know how to solve it.
How do I test JUCI-UBUS and uhttpd for normal work?

- Login page load slow(Load 1 ~ 2 minutes)
- browser RPC connect failed(Many http://192.168.1.1/ubus requests failed)
- Default uhttpd does not support GZ automatic compression(Need manual gzip -d *)

I suspect that my rpcd/ubus is a problem.
But I checked the compiled package, it was juci-rpcd/juci-ubus.

How do I check to confirm that uhttpd and UBUS are working properly？

mkschreder, 2016-02-18:
You're not supposed to need default uhttpd. Use at least openwrt CC and
install feed with -f flag. It should say "Overriding package uhttpd"..

darcyg, 2016-02-19:
thks! is ok!

closed 2016-02-19 by darcyg

happyuser220, 2016-08-19:
@mkschreder 
Hi Mkschreder:
I did what your said exactly ,but i can not login now.
the username is admin, the password is admin password of openwrt, right?
I use adduser to add the admin user and set password to 123456. but still can not login.
I use F12 to see the error info:
01-juci.js:57 UCI failed to initialize!(anonymous function) @ 01-juci.js:57j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29socket.onmessage @ 01-juci.js:56
01-juci.js:57 JUCI Init
01-juci.js:57 init: config
01-juci.js:56 Could not sync config juci(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29socket.onmessage @ 01-juci.js:56
01-juci.js:57 Using settings from config/juci on router
01-juci.js:57 Failed to verify session.
01-juci.js:57 juci: loading menu from server..
01-juci.js:57 CONF
01-juci.js:57 juci: angular init
01-juci.js:56 Could not sync config firewall(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29socket.onmessage @ 01-juci.js:56
01-juci.js:56 Could not sync config hosts(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29socket.onmessage @ 01-juci.js:56
01-juci.js:56 Could not sync config wireless(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29socket.onmessage @ 01-juci.js:56
01-juci.js:56 Could not sync config system(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29(anonymous function) @ 01-juci.js:56j @ 01-juci.js:29fireWith @ 01-juci.js:29e.(anonymous function) @ 01-juci.js:29socket.onmessage @ 01-juci.js:56
01-juci.js:56 Could not sync config upnpd

RPC connection established!
01-juci.js:56 login: got challenge: {"token":"43fac352"}
01-juci.js:58 Could not log in!

happyuser220, 2016-08-23:
@mkschreder I can login now, But it displays some error:

01-juci.js:57 juci: angular init
301-juci.js:56 Synched config firewall
01-juci.js:56 Could not sync config hosts(anonymous function) @ 01-juci.js:56
301-juci.js:56 Synched config wireless
01-juci.js:56 Synched config system
01-juci.js:56 Could not sync config upnpd(anonymous function) @ 01-juci.js:56
01-juci.js:56 Adding: {".type":"settings",".name":"settings"} to firewall: {"disabled":false,"ping_wan":false,"nat_enabled":true}
01-juci.js:56 Adding: {".type":"wifi-status",".name":"status"} to wireless: {"wlan":true,"wps":true,"schedule":false,"sched_status":false}
01-juci.js:56 Checking for errors...
01-juci.js:56 UCI error [firewall]: firewall.privoxy: Rule name can not be empty!(anonymous function) @ 01-juci.js:56
01-juci.js:56 Writing changes: []
01-juci.js:56 Adding: {".type":"upgrade",".name":"upgrade"} to system: {"fw_check_url":"","fw_path_url":"","fw_usb_path":"","fw_find_ext":"","fw_upload_path":""}
01-juci.js:56 Synched config network
01-juci.js:56 Added new section: {"section":"settings"}
01-juci.js:56 Checking for errors...
01-juci.js:56 UCI error [firewall]: firewall.privoxy: Rule name can not be empty!(anonymous function) @ 01-juci.js:56
01-juci.js:56 UCI error [network]: network.wwan: DHCP interface needs physical interface(anonymous function) @ 01-juci.js:56
01-juci.js:56 Writing changes: []
01-juci.js:56 Added new section: {"section":"status"}
01-juci.js:56 Checking for errors...
01-juci.js:56 UCI error [firewall]: firewall.privoxy: Rule name can not be empty!(anonymous function) @ 01-juci.js:56
01-juci.js:56 UCI error [network]: network.wwan: DHCP interface needs physical interface(anonymous function) @ 01-juci.js:56
01-juci.js:56 Writing changes: []
50-juci-openwrt-wireless.js:1 Uncaught TypeError: result.clients.map is not a function
01-juci.js:56 Added new section: {"section":"upgrade"}
01-juci.js:56 Checking for errors...
01-juci.js:56 UCI error [firewall]: firewall.privoxy: Rule name can not be empty!(anonymous function) @ 01-juci.js:56
01-juci.js:56 UCI error [network]: network.wwan: DHCP interface needs physical interface(anonymous function) @ 01-juci.js:56
01-juci.js:56 Writing changes: []
201-juci.js:56 Synched config wireless
201-juci.js:56 Synched config network
50-juci-openwrt-wireless.js:1 Uncaught TypeError: $scope.wireless.clients.map is not a function
201-juci.js:56 Synched config wireless
201-juci.js:56 Synched config network
01-juci.js:56 Synched config wireless