mkschreder/juci #39: How to use JUCI on OpenWRT?

jasonslam, 2016-03-30:
Hello,

I compiled OpenWRT following the guidelines in README.
However, the web interface is not functional:

[image]

Please note that this is after running juci-update manually. Without this, it the web root doesn't contain any HTML and no assets.

mkschreder, 2016-03-30:
You seem to have a broken build. There is a 01-juci.js file that also
should be included. You have plugins but you are missing core library.

jasonslam, 2016-03-30:
I selected everything related to JUCI in `make menuconfig` but I still cant get it to work.
However this command was not found: `CONFIG_PACKAGE_juci-full=y`
What is this supposed to do and how can I get the core library?

mkschreder, 2016-03-30:
It is not a command. It is a metapackage that will select most of other
required dependencies. The text goes into your .config and then you run
"make defconfig" to set all dependencies.

Don't worry if you have a little trouble building right now, I will do more
clean test builds by the end of the week and fix any issues that arise. I'm
currently away.

jasonslam, 2016-03-30:
Alright thanks for your help.
Unfortunately the line didn't solve the problem either.
It seems a little weird that I'm required to run `juci-update` for the HTML to build.
Also having to apply the fix https://github.com/mkschreder/juci-openwrt-feed/issues/4#issuecomment-200684706 manually quite often.

closed 2016-04-05 by jasonslam