mkschreder/juci #42: You do not have an RPC connection to the router!

zooboa, 2016-05-01:
**F12 in browser: **
Request URL:ws://192.168.200.254/websocket/
Request Method:GET
Status Code:503 Service Not Available

**Error log in /var/log/error.log(openwrt):**
root@Box:/tmp/log# cat error.log 
2016-04-09 11:37:20: (log.c.166) server started 
2016-04-09 11:37:20: (server.c.1032) WARNING: unknown config-key: url.access-deny (ignored) 
2016-04-09 11:37:20: (mod_websocket.c.340) /websocket/ is match WebSocket extension: /websocket/ 
2016-04-09 11:37:20: (mod_websocket_handshake.c.98) allowed origins are not specified 
2016-04-09 11:37:20: (mod_websocket.c.531) works as WebSocket Proxy 
2016-04-09 11:37:20: (mod_websocket.c.152) try to connect backend -> 127.0.0.1 : 1234 
2016-04-09 11:37:20: (mod_websocket.c.156) fail to connect 
2016-04-09 11:37:20: (mod_websocket.c.854) disconnected from client ( fd = 8 ) 
2016-04-09 11:37:25: (mod_websocket.c.340) /websocket/ is match WebSocket extension: /websocket/ 
2016-04-09 11:37:25: (mod_websocket_handshake.c.98) allowed origins are not specified 
2016-04-09 11:37:25: (mod_websocket.c.531) works as WebSocket Proxy 
2016-04-09 11:37:25: (mod_websocket.c.152) try to connect backend -> 127.0.0.1 : 1234 
2016-04-09 11:37:25: (mod_websocket.c.156) fail to connect 
2016-04-09 11:37:25: (mod_websocket.c.854) disconnected from client ( fd = 9 )

mkschreder, 2016-05-01:
You need lighttpd with websockets enabled and with forwarding set up to the
websocket server. This version is still beta. Stable version is 2.16.02
(without websocket backend).

mkschreder, 2016-05-01:
Lighttpd package is in the same repo, but make sure you install juci feed
before openwrt feed. Otherwise you will get default lighttpd and it will
not work. (in openwrt always install custom feeds forst and the main feed).

onexay, 2016-05-25:
akshay@Taxila:~/openwrt$ cat feeds.conf.default 
src-git-full juci https://github.com/mkschreder/juci-openwrt-feed.git
src-git packages https://github.com/openwrt/packages.git
src-git luci https://github.com/openwrt/luci.git
src-git routing https://github.com/openwrt-routing/packages.git
src-git telephony https://github.com/openwrt/telephony.git
src-git management https://github.com/openwrt-management/packages.git
src-git targets https://github.com/openwrt/targets.git
#src-git oldpackages http://git.openwrt.org/packages.git
#src-link custom /usr/src/openwrt/custom-feed

---

./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig

---

./scripts/feeds update juci
./scripts/feeds install -f -p juci -a
make defconfig 

---

make -j4

---

Getting the same error.

mkschreder, 2016-05-25:
You should specify version on the feed. Current stable is 2.16.02. Current
devel is 2.16.05.

ril3y, 2016-07-23:
@mkschreder can you explain how to specify the version on the feed?  I am not sure I follow.

mkschreder, 2016-07-23:
Hello,

Versions are specified using ";" after feed url followed by branch name.

...juci-openwrt-feed.git;v2.16.07

ril3y, 2016-07-23:
@mkschreder thanks for the reply.  After a day of failing I thought perhaps I could create a dockerfile and you can see my exact setup. 

https://hub.docker.com/r/ril3y/openwrt-juci/

This is the dockerfile that created this image.

```
FROM ubuntu:latest

RUN apt-get update && \
    apt-get install -y zip libxml-parser-perl apt-utils build-essential libncurses5-dev m4 flex git git-core zlib1g-dev unzip curl subversion gawk htop python libssl-dev quilt screen file wget vim && \
    apt-get autoremove && \
    apt-get autoclean;

#Install Nodejs to build JUCI
RUN curl -sL https://deb.nodesource.com/setup_6.x | apt-get install -y nodejs
RUN apt-get install -y npm  
RUN ln -s /usr/bin/nodejs /usr/bin/node
RUN npm install -g less uglifyjs yui-compressor grunt-cli

#Get OpenWRT Source
RUN git clone https://github.com/openwrt/openwrt.git /tmp/openwrt
WORKDIR /tmp/openwrt
RUN echo 'src-git-full juci https://github.com/mkschreder/juci-openwrt-feed.git;v2.16.07' | cat - feeds.conf.default  > temp && mv temp feeds.conf.default

#This is just so we can build as root.  You can add a user and do it via nonroot if you wish.
ADD build.sh build.sh
RUN chmod +x build.sh
```

Once all of that stuff runs all deps should be ready to build the image.  build.sh is below.
```
echo "  _____   _____  _______ __   _ _  _  _  ______ _______                               
 |     | |_____] |______ | \  | |  |  | |_____/    |                                  
 |_____| |       |______ |  \_| |__|__| |    \_    |                                  
                                                                                      
 _____ _     _ _______ _____      ______  _     _ _____        ______  _______  ______
   |   |     | |         |        |_____] |     |   |   |      |     \ |______ |_____/
 __|   |_____| |_____  __|__      |_____] |_____| __|__ |_____ |_____/ |______ |    \_
                                                                                      "
echo "=============================="
echo "Updating openWRT feeds."
echo "After feed updates are completed make menuconfig will automatically run."
echo "Select your packages and then the build script will continue."
echo "=============================="
./scripts/feeds update -a
./scripts/feeds install -a
               
               
echo "=============================="                                                                                      
echo "Running Make Menuconfig... "
echo "=============================="
make menuconfig

echo "=============================="
echo "Updating juci feeds and forcing override packages."
echo "=============================="
./scripts/feeds update juci
./scripts/feeds install -f -p juci -a

echo "CONFIG_PACKAGE_juci-full=y" >> .config 

echo "=============================="
echo "Running defconfig..."
echo "=============================="
make defconfig

export FORCE_UNSAFE_CONFIGURE=1
make V=s -j3

```

Once build starts it will run make menuconfig for you.  This is when I select my board and then ONLY select the JUCI all and the orangerpcd then leave everything else alone.  As you can see after you exit menuconfig it saves .config file then moves on to the juci feeds update and force install of juci.  I also echo the `CONFIG_PACKAGE_juci-full=y` to the .config for fun.  Run the `defconfig` and then make....

After all of this I flash the board but still get the:

`	
Please sign in
You do not have an RPC connection to the router!`

I was hoping you could take a look at this and see if there was something obvious standing out to you.  If you are familiar with docker you can try and build my images to see if it too is messed up for you as well.  Any help would be appreciated and thanks for all your great work.

mkschreder, 2016-07-23:
Very common problem: try installing juci feed first before you install
openwrt feed. I'm guessing that because you install all default packages
first, the feeds script will not install juci packages that are supposed to
replace default ones.

ril3y, 2016-07-24:
This line puts the feed at the top of the dress file. So I think it is as
you said to do?

Am I misunderstanding?

RUN echo 'src-git-full juci
https://github.com/mkschreder/juci-openwrt-feed.git;v2.16.07' | cat -
feeds.conf.default > temp && mv temp feeds.conf.default

mkschreder, 2016-07-24:
do feeds install -a after feeds install juci.

ril3y, 2016-07-24:
So like this?

echo "=============================="
echo "Updating openWRT feeds."
echo "After feed updates are completed make menuconfig will automatically
run."
echo "Select your packages and then the build script will continue."
echo "=============================="
./scripts/feeds update -a
./scripts/feeds install -a

echo "=============================="

echo "Running Make Menuconfig... "
echo "=============================="
make menuconfig

echo "=============================="
echo "Updating juci feeds and forcing override packages."
echo "=============================="
./scripts/feeds update juci
./scripts/feeds install -f -p juci -a
*./scripts/feeds install -a*

then

echo "CONFIG_PACKAGE_juci-full=y" >> .config

echo "=============================="
echo "Running defconfig..."
echo "=============================="
make defconfig

make V=s -j3

mkschreder, 2016-07-24:
You have feeds install -a further up. But yes first juci and then
everything else.

> do feeds install -a after feeds install juci.
>
>
> > This line puts the feed at the top of the dress file. So I think it is
as
> > you said to do?
> >
> > Am I misunderstanding?
> >
> > RUN echo 'src-git-full juci
> > https://github.com/mkschreder/juci-openwrt-feed.git;v2.16.07' | cat -
> > feeds.conf.default > temp && mv temp feeds.conf.default
> >
> >
> > > Very common problem: try installing juci feed first before you install
> > > openwrt feed. I'm guessing that because you install all default
> packages
> > > first, the feeds script will not install juci packages that are
> supposed
> > to
> > > replace default ones.
> > >
> > > > @mkschreder <https://github.com/mkschreder> thanks for the reply.
> > After
> > > a
> > > > day of failing I thought perhaps I could create a dockerfile and you
> > can
> > > > see my exact setup.
> > > >
> > > > https://hub.docker.com/r/ril3y/openwrt-juci/
> > > >
> > > > This is the dockerfile that created this image.
> > > >
> > > > FROM ubuntu:latest
> > > >
> > > >
> > > > RUN apt-get update && \
> > > > apt-get install -y zip libxml-parser-perl apt-utils build-essential
> > > libncurses5-dev m4 flex git git-core zlib1g-dev unzip curl subversion
> > gawk
> > > htop python libssl-dev quilt screen file wget vim && \
> > > > apt-get autoremove && \
> > > > apt-get autoclean;
> > > >
> > > > #Install Nodejs to build JUCI
> > > > RUN curl -sL https://deb.nodesource.com/setup_6.x | apt-get install
> -y
> > > nodejs
> > > > RUN apt-get install -y npm
> > > > RUN ln -s /usr/bin/nodejs /usr/bin/node
> > > > RUN npm install -g less uglifyjs yui-compressor grunt-cli
> > > >
> > > > #Get OpenWRT Source
> > > > RUN git clone https://github.com/openwrt/openwrt.git /tmp/openwrt
> > > > WORKDIR /tmp/openwrt
> > > > RUN echo 'src-git-full juci
> > > https://github.com/mkschreder/juci-openwrt-feed.git;v2.16.07' | cat -
> > > feeds.conf.default > temp && mv temp feeds.conf.default
> > > >
> > > > #This is just so we can build as root. You can add a user and do it
> via
> > > nonroot if you wish.
> > > > ADD build.sh build.sh
> > > > RUN chmod +x build.sh
> > > >
> > > > Once all of that stuff runs all deps should be ready to build the
> > image.
> > > > build.sh is below.
> > > >
> > > > echo " _____ _____ _______ __ _ _ _ _ ______ _______
> > > > | | |_____] |______ | \ | | | | |_____/ |
> > > > |_____| | |______ | \_| |__|__| | \_ |
> > > >
> > > > _____ _ _ _______ _____ ______ _ _ _____ ______ _______ ______
> > > > | | | | | |_____] | | | | | \ |______ |_____/
> > > > __| |_____| |_____ __|__ |_____] |_____| __|__ |_____ |_____/
> |______ |
> > > \_
> > > > "
> > > > echo "=============================="
> > > > echo "Updating openWRT feeds."
> > > > echo "After feed updates are completed make menuconfig will
> > > automatically run."
> > > > echo "Select your packages and then the build script will continue."
> > > > echo "=============================="
> > > > ./scripts/feeds update -a
> > > > ./scripts/feeds install -a
> > > >
> > > >
> > > > echo "=============================="
> > > > echo "Running Make Menuconfig... "
> > > > echo "=============================="
> > > > make menuconfig
> > > >
> > > > echo "=============================="
> > > > echo "Updating juci feeds and forcing override packages."
> > > > echo "=============================="
> > > > ./scripts/feeds update juci
> > > > ./scripts/feeds install -f -p juci -a
> > > >
> > > > echo "CONFIG_PACKAGE_juci-full=y" >> .config
> > > >
> > > > echo "=============================="
> > > > echo "Running defconfig..."
> > > > echo "=============================="
> > > > make defconfig
> > > >
> > > > export FORCE_UNSAFE_CONFIGURE=1
> > > > make V=s -j3
> > > >
> > > >
> > > > Once build starts it will run make menuconfig for you. This is when
I
> > > > select my board and then ONLY select the JUCI all and the orangerpcd
> > then
> > > > leave everything else alone. As you can see after you exit
menuconfig
> > it
> > > > saves .config file then moves on to the juci feeds update and force
> > > install
> > > > of juci. I also echo the CONFIG_PACKAGE_juci-full=y to the .config
> for
> > > > fun. Run the defconfig and then make....
> > > >
> > > > After all of this I flash the board but still get the:
> > > >
> > > >
> > > > Please sign in
> > > > You do not have an RPC connection to the router!
> > > >
> > > > I was hoping you could take a look at this and see if there was
> > something
> > > > obvious standing out to you. If you are familiar with docker you can
> > try
> > > > and build my images to see if it too is messed up for you as well.
> Any
> > > help
> > > > would be appreciated and thanks for all your great work.
> > > >
> > > > <https://github.com/mkschreder/juci/issues/42#issuecomment-234741948
> >,
> > > or mute
> > > > the thread
> > > > <
https://github.com/
> > > >
> > > >
> > >
> > or mute
> > > the thread
> > > <
> > >
> > >
> >
> or mute
> > the thread
> > <

> >
>
> >
>
> <https://github.com/mkschreder/juci/issues/42#issuecomment-234761390>, or
mute
> the thread
> <

<https://github.com/mkschreder/juci/issues/42#issuecomment-234783110>, or mute
the thread

ril3y, 2016-07-24:
Thanks for the help.

So in the spirit of not pounding my router into dust.

These are the exact steps. I am using to build this and still getting the
RPC error!

./scripts/feeds update juci
>
> ./scripts/feeds install -f -p juci -a
>
>
>> make defconfig
>
> ./scripts/feeds install -a
>
> make menuconfig
>
> make V=s -j3
>
>

During the make menuconfig I select juci full and the orange rpc then exit
and save.  Its pretty much defaults and JUCI.

I feel like the universe hates me right now :)

Thanks again for the help.

mkschreder, 2016-07-24:
did you delete your feeds folder before you started all over again? Then
also remember to select juci-full-openwrt and orange-rpcd in menuconfig.

ril3y, 2016-07-24:
I did not delete my feeds folder.  OK!  Great I will give it a go now.

Thanks for hanging in there with me!

ril3y

ril3y, 2016-07-26:
mkschreder,

I think I am close to getting this working however I think there might be
something wrong with your makefile for lighthttpd?

In file included from base.h:24:0,
>
>                  from mod_flv_streaming.c:1:
>
> *array.h:9:19: fatal error: pcre.h: No such file or directory*
>
> compilation terminated.
>
> Makefile:1348: recipe for target 'mod_flv_streaming.lo' failed
>
> make[7]: *** [mod_flv_streaming.lo] Error 1
>
> make[7]: Leaving directory
>> '/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/src'
>
> Makefile:945: recipe for target 'all' failed
>
> make[6]: *** [all] Error 2
>
> make[6]: Leaving directory
>> '/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/src'
>
> Makefile:435: recipe for target 'all-recursive' failed
>
> make[5]: *** [all-recursive] Error 1
>
> make[5]: Leaving directory
>> '/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231'
>
> Makefile:365: recipe for target 'all' failed
>
> make[4]: *** [all] Error 2
>
> make[4]: Leaving directory
>> '/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231'
>
> Makefile:316: recipe for target
>> '/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/.built'
>> failed
>
> make[3]: ***
>> [/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/.built]
>> Error 2
>
> make[3]: Leaving directory '/tmp/openwrt/feeds/juci/lighttpd'
>
> package/Makefile:196: recipe for target
>> 'package/feeds/juci/lighttpd/compile' failed
>
> make[2]: *** [package/feeds/juci/lighttpd/compile] Error 2
>
> make[2]: Leaving directory '/tmp/openwrt'
>
> package/Makefile:193: recipe for target
>> '/tmp/openwrt/staging_dir/target-mips_34kc_musl-1.1.14/stamp/.package_compile'
>> failed
>
> make[1]: ***
>> [/tmp/openwrt/staging_dir/target-mips_34kc_musl-1.1.14/stamp/.package_compile]
>> Error 2
>
> make[1]: Leaving directory '/tmp/openwrt'
>
> /tmp/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
>
> make: *** [world] Error 2
>
>
I have made sure I have libpcre3-dev installed on debian and ubuntu but
both are failing on build.

>
> *root@1cc1d8a1dbaa:/tmp/openwrt# locate pcre.h/usr/include/pcre.h*

I have poked around at this tonight for a few hours but I am coming up
empty here.  Do you know what to try next?  Thanks again for your help.

mkschreder, 2016-07-26:
Why is lighttpd_mod_flv selected? There is nothing here that selects it.
Try selecting libpcre as well then since that's what it reports as missing.

ril3y, 2016-07-26:
I will try what you suggested tonight.

I have no idea why flv is getting selected.  All I did was select my target
and juci-full in my .config file then did a make defconfig then make and
this is what I get.  I did not even run make menuconfig to select any
additional packages.
I will give it a go tonight.

Riley

mkschreder, 2016-07-26:
You did not select orange-rpcd though?

ril3y, 2016-07-26:
I thought defconfig would select that.  I can check that tonight.  However
I am still not sure why / if juci-full=y is selecting the flv package.

mkschreder, 2016-07-26:
No idea. But jsonrpc server is not selected by default. You need to select
that.

ril3y, 2016-07-26:
I will try that out.  Just curious but you have build juci with openwrt
with the new branch successfully right :)  If so I would love to see what
you did from the start to then end.

Riley

mkschreder, 2016-07-26:
I build with custom config most of the time. It's been a while since I
built from openwrt default. :-)

> I will try that out. Just curious but you have build juci with openwrt
> with the new branch successfully right :) If so I would love to see what
> you did from the start to then end.
>
> Riley
>
> On Tue, Jul 26, 2016 at 10:07 AM, Martin K. Schröder <
>
> > No idea. But jsonrpc server is not selected by default. You need to
select
> > that.
> >
> >
> >
> > > I thought defconfig would select that. I can check that tonight.
However
> > > I am still not sure why / if juci-full=y is selecting the flv package.
> > >
> > >
> > >
> > > On Tue, Jul 26, 2016 at 9:52 AM, Martin K. Schröder <
> > >
> > > > You did not select orange-rpcd though?
> > > >
> > > >
> > > >
> > > > > I will try what you suggested tonight.
> > > > >
> > > > > I have no idea why flv is getting selected. All I did was select
my
> > > > target
> > > > > and juci-full in my .config file then did a make defconfig then
make
> > > and
> > > > > this is what I get. I did not even run make menuconfig to select
any
> > > > > additional packages.
> > > > > I will give it a go tonight.
> > > > >
> > > > > Riley
> > > > >
> > > > > On Tue, Jul 26, 2016 at 2:49 AM, Martin K. Schröder <
> > > > >
> > > > > > Why is lighttpd_mod_flv selected? There is nothing here that
> > selects
> > > > it.
> > > > > > Try selecting libpcre as well then since that's what it reports
as
> > > > > missing.
> > > > > >
> > > > > > > mkschreder,
> > > > > > >
> > > > > > > I think I am close to getting this working however I think
there
> > > > might
> > > > > be
> > > > > > > something wrong with your makefile for lighthttpd?
> > > > > > >
> > > > > > > In file included from base.h:24:0,
> > > > > > > >
> > > > > > > > from mod_flv_streaming.c:1:
> > > > > > > >
> > > > > > > > *array.h:9:19: fatal error: pcre.h: No such file or
directory*
> > > > > >
> > > > > > > >
> > > > > > > > compilation terminated.
> > > > > > > >
> > > > > > > > Makefile:1348: recipe for target 'mod_flv_streaming.lo'
failed
> > > > > > > >
> > > > > > > > make[7]: *** [mod_flv_streaming.lo] Error 1
> > > > > > > >
> > > > > > > > make[7]: Leaving directory
'/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/src'
> > > > > > > >
> > > > > > > > Makefile:945: recipe for target 'all' failed
> > > > > > > >
> > > > > > > > make[6]: *** [all] Error 2
> > > > > > > >
> > > > > > > > make[6]: Leaving directory
'/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/src'
> > > > > > > >
> > > > > > > > Makefile:435: recipe for target 'all-recursive' failed
> > > > > > > >
> > > > > > > > make[5]: *** [all-recursive] Error 1
> > > > > > > >
> > > > > > > > make[5]: Leaving directory
'/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231'
> > > > > > > >
> > > > > > > > Makefile:365: recipe for target 'all' failed
> > > > > > > >
> > > > > > > > make[4]: *** [all] Error 2
> > > > > > > >
> > > > > > > > make[4]: Leaving directory
'/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231'
> > > > > > > >
> > > > > > > > Makefile:316: recipe for target
'/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/.built'
> > > > > > > >> failed
> > > > > > > >
> > > > > > > > make[3]: ***
[/tmp/openwrt/build_dir/target-mips_34kc_musl-1.1.14/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/.built]
> > > > > > > >> Error 2
> > > > > > > >
> > > > > > > > make[3]: Leaving directory
'/tmp/openwrt/feeds/juci/lighttpd'
> > > > > > > >
> > > > > > > > package/Makefile:196: recipe for target
> > > > > > > >> 'package/feeds/juci/lighttpd/compile' failed
> > > > > > > >
> > > > > > > > make[2]: *** [package/feeds/juci/lighttpd/compile] Error 2
> > > > > > > >
> > > > > > > > make[2]: Leaving directory '/tmp/openwrt'
> > > > > > > >
> > > > > > > > package/Makefile:193: recipe for target
'/tmp/openwrt/staging_dir/target-mips_34kc_musl-1.1.14/stamp/.package_compile'
> > > > > > > >> failed
> > > > > > > >
> > > > > > > > make[1]: ***
[/tmp/openwrt/staging_dir/target-mips_34kc_musl-1.1.14/stamp/.package_compile]
> Error 2
>
> make[1]: Leaving directory '/tmp/openwrt'
>
> /tmp/openwrt/include/toplevel.mk:192: recipe for target
> 'world'
> failed
>
> make: *** [world] Error 2
>
>
> I have made sure I have libpcre3-dev installed [...]
your
> > > help.
> > > > > > >
> > > > > > > <
> > > https://github.com/mkschreder/juci/issues/42#issuecomment-235152668
> > > > >,
> > > > > > or mute
> > > > > > > the thread
> > > > > > > <
https://github.com/
> > > > > > >
> > > > > >
> > > > > > >
> > > > > >
> > > > > > <
> > https://github.com/mkschreder/juci/issues/42#issuecomment-235177765
> > > >,
> > > > > or mute
> > > > > > the thread
> > > > > > <
> > > > > >
> > > > > >
> > > > >
> > > > > <
https://github.com/mkschreder/juci/issues/42#issuecomment-235271809
> > >,
> > > > or mute
> > > > > the thread
> > > > > <
> > > > >
> > > >
> > > > >
> > > >
> > > > <https://github.com/mkschreder/juci/issues/42#issuecomment-235273820
>,
> > > or mute
> > > > the thread
> > > > <
> > > >
> > > >
> > >
> > or mute
> > > the thread
> > > <
or mute
> > the thread
> > <
<https://github.com/mkschreder/juci/issues/42#issuecomment-235283679>, or
mute the thread

ril3y, 2016-07-26:
Ok well that is good to know.  Perhaps there is an issue.  I will keep on
running this down.  What do you typically compile for?  A different distro?

I will try some more tonight.

ril3y, 2016-07-27:
Hi Martin,

Sadly, I am at a loss here.  I have tried pretty much everything to get
lighthttpd to compile without the above mentioned error.  streaming flv is
not selected in the .config file for sure however it fails trying to build
it.

./scripts/feeds update juci
./scripts/feeds install -f -p juci -a

edit .config to set orangerpc and juci-full to = y

make defconfig

make V=s

Fails on building.... I have even left the default target options along
(ath7xx/9xx).  I would very much interested if you gave the exact steps
above a go and saw if it compiled for you?  I have tried ubuntu and debian
but nada!  I would really like to user your UI as I think its pretty
awesome and much more modern.  Any other ideas?  I would be willing to try.

mkschreder, 2016-08-31:
I have updated readme (in v2.16.08 branch) with exact steps for building and testing uml target. Fixed several issues along the way while getting that uml target working.

closed 2016-08-31 by mkschreder