mkschreder/juci #44: uci failed to initialize

DanialCasvyn, 2016-05-26:
please help mkschreder
i got this error 
sorry for my bad english
RPC init
01-juci.js:50 connecting to rpc server at (ws://192.168.1.1/websocket/)
01-juci.js:50 RPC connection established!
01-juci.js:50 RPC initialized!
01-juci.js:50 UCI Init
01-juci.js:50 Init UCI
01-juci.js:50 UCI failed to initialize!(anonymous function) @ 01-juci.js:50
01-juci.js:50 JUCI Init
01-juci.js:51 init: config
01-juci.js:50 Could not sync config juci(anonymous function) @ 01-juci.js:50
01-juci.js:51 Using settings from config/juci on router
01-juci.js:50 Failed to verify session.
01-juci.js:50 juci: loading menu from server..
01-juci.js:38 Uncaught Error: [$injector:modulerr] http://errors.angularjs.org/1.3.15/$injector/modulerr?p0=juci&p1=Error%3A%2…20%20%20at%20_t%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A38%3A32716)
https://cr-input.mxpnl.net/data?_channel_id=&_partner_id=39571&_sub_id=0000&_app_version=1.0.23&_app=cs-dca Failed to load resource: net::ERR_NAME_NOT_RESOLVED

mkschreder, 2016-05-26:
Make sure you have installed the lighttpd from juci repo with webocket
support and not from openwrt feed.

DanialCasvyn, 2016-05-26:
i have installed the lighttpd from juci repo with websocket
i've check folder bin/../packages/juci and lighttpd was there

mkschreder, 2016-05-26:
If you can't get devel (2.16.05) working then use stable 2.16.02 instead.
You should remove your feeds folder, then update all feeds and then do
./scripts/feeds install -f -a -p juci. Only after that install the rest of
the openwrt packages. That would make sure that you have all the right
packages first and then duplicates will not be installed if they exist.

DanialCasvyn, 2016-05-26:
thanks mkschreder, i wil try

DanialCasvyn, 2016-05-26:
i try compile juci 2.16.02 and i get error when compile luaposix33 

Package luaposix33 is missing dependencies for the following libraries:
libncursesw.so.5

ncurses already compile
i've check folder libncursesw, libncursesw.so.5 was there. but it's link shared library file, target to libncursesw.so.5.9 (this file was there)

mkschreder, 2016-05-26:
Yes some people have been having that problem. For some reason it tries to
link against ncursesw instead of libncurses. I don't know what causes it
for some.

DanialCasvyn, 2016-05-28:
i try compile again juci 2.16.02
and i get this error in console
Init UBUS -> 
01-juci.js:50 Init UCI
01-juci.js:50 Missing type definition for config rpcd
01-juci.js:50 init: config
01-juci.js:49 RPC error (system.board): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}request.error @ 01-juci.js:49
01-juci.js:49 RPC error (uci.revert): {"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}request.error @ 01-juci.js:49
http://192.168.1.1/favicon.ico Failed to load resource: the server responded with a status of 404 (Not Found)
01-juci.js:50 Synched config juci
01-juci.js:50 Using settings from config/juci on router
01-juci.js:49 RPC error (session.access): {"jsonrpc":"2.0","id":1,"error":{"code":-32700,"message":"Parse error"}}request.error @ 01-juci.js:49
01-juci.js:49 Session access call failed: you will be logged out!(anonymous function) @ 01-juci.js:49
01-juci.js:50 Failed to verify session.
01-juci.js:50 juci: loading menu from server..
01-juci.js:38 Uncaught Error: [$injector:modulerr] http://errors.angularjs.org/1.3.15/$injector/modulerr?p0=juci&p1=Error%3A%2…20%20%20at%20_t%20(http%3A%2F%2F192.168.1.1%2Fjs%2F01-juci.js%3A38%3A19686)
https://cr-input.mxpnl.net/data?_channel_id=&_partner_id=39571&_sub_id=0000&_app_version=1.0.23&_app=cs-dca Failed to load resource: net::ERR_NAME_NOT_RESOLVED

mkschreder, 2016-08-31:
v2.16.02 no longer maintained.

closed 2016-08-31 by mkschreder

ajeecai, 2017-03-25:
I got exact the same error when try to build with latest tree commit#592e32a, so what is the idea?