mkschreder/juci #47: rpcd - fatal: reference is not a tree: 72048bb2c8c23fbb57c7d9e6de4ab7373036da5a

Kota-Kun, 2016-06-20:
How to fix it??
```
make[3]: Leaving directory `/home/kota/juciwrt/package/network/utils/iwinfo
make[3]: Entering directory `/home/kota/juciwrt/package/system/rpcd'
mkdir -p /home/kota/juciwrt/dl
echo "Checking out files from the git repository..."; mkdir -p /home/kota/juciwrt/tmp/dl && cd /home/kota/juciwrt/tmp/dl && rm -rf rpcd && [ \! -d rpcd ] && git clone https://github.com/mkschreder/juci-rpcd.git rpcd --recursive && (cd rpcd && git checkout 72048bb2c8c23fbb57c7d9e6de4ab7373036da5a && git submodule update) && echo "Packing checkout..." && rm -rf rpcd/.git && 	tar czf /home/kota/juciwrt/tmp/dl/rpcd-2015-01-10-72048bb2c8c23fbb57c7d9e6de4ab7373036da5a.tar.gz rpcd && mv /home/kota/juciwrt/tmp/dl/rpcd-2015-01-10-72048bb2c8c23fbb57c7d9e6de4ab7373036da5a.tar.gz /home/kota/juciwrt/dl/ && rm -rf rpcd; 
Checking out files from the git repository...
Cloning into 'rpcd'...
remote: Counting objects: 647, done.
remote: Total 647 (delta 0), reused 0 (delta 0), pack-reused 647
Receiving objects: 100% (647/647), 137.50 KiB | 0 bytes/s, done.
Resolving deltas: 100% (423/423), done.
Checking connectivity... done.
fatal: reference is not a tree: 72048bb2c8c23fbb57c7d9e6de4ab7373036da5a
make[3]: *** [/home/kota/juciwrt/dl/rpcd-2015-01-10-72048bb2c8c23fbb57c7d9e6de4ab7373036da5a.tar.gz] Error 128
make[3]: Leaving directory `/home/kota/juciwrt/package/system/rpcd'
make[2]: *** [package/system/rpcd/compile] Error 2
make[2]: Leaving directory `/home/kota/juciwrt'
make[1]: *** [/home/kota/juciwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.11/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/kota/juciwrt'
make: *** [world] Error 2
kota@kota:~/juciwrt$ 
```

Kota-Kun, 2016-06-20:
**Fixed.**
chenge version in juciwrt/package/system/rpcd/Makefile
**from** 
`PKG_SOURCE_VERSION:=72048bb2c8c23fbb57c7d9e6de4ab7373036da5a`
**to**
`PKG_SOURCE_VERSION:=50b1a51f32bbabb450e0fcabb4225e967651b4bb`

mkschreder, 2016-07-05:
This is not necessary because since many months ago the feed already overrides that system package (if the feed is installed correctly).

mkschreder, 2016-08-19:
rpcd is no longer relevant.

closed 2016-08-19 by mkschreder