mkschreder/juci #49: build error openwrt + juci

Kota-Kun, 2016-06-22:
```
Makefile:1348: recipe for target 'mod_extforward.lo' failed
make[7]: *** [mod_extforward.lo] Error 1
make[7]: Leaving directory '/home/kota/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/src'
Makefile:945: recipe for target 'all' failed
make[6]: *** [all] Error 2
make[6]: Leaving directory '/home/kota/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/src'
Makefile:435: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/kota/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231'
Makefile:365: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/kota/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231'
Makefile:316: recipe for target '/home/kota/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/.built' failed
make[3]: *** [/home/kota/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/lighttpd-23644e8f150baac0aa5199a6820234d8e7898231/.built] Error 2
make[3]: Leaving directory '/home/kota/openwrt/feeds/juci/lighttpd'
package/Makefile:191: recipe for target 'package/feeds/juci/lighttpd/compile' failed
make[2]: *** [package/feeds/juci/lighttpd/compile] Error 2
make[2]: Leaving directory '/home/kota/openwrt'
package/Makefile:188: recipe for target '/home/kota/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/home/kota/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kota/openwrt'
/home/kota/openwrt/include/toplevel.mk:181: ошибка выполнения рецепта для цели «world»
make: *** [world] Ошибка 2
kota@kota:~/openwrt$ 
```

mkschreder, 2016-06-23:
Can not check right now, but from the above it seems like a lighttpd issue. Perhaps you can compare with their trunk and see whether current feed version is missing anything? You can use meld to compare just that file. Do it between the code in your builddir and the lighttpd official repo. If you fix it yourself then submit a patch.

Kota-Kun, 2016-06-23:
Compared to the original repository, changes a lot. I myself can not make changes. Therefore, waiting for your help)

mkschreder, 2016-06-23:
For now perhaps you can just disable that module in your build? Are you
using it for anything?

Kota-Kun, 2016-06-23:
He did not need to juci?
If not, turn it off. As I understand it is enabled by default.

Kota-Kun, 2016-06-23:
DEPENDS

juci-openwrt-feed/juci/Makefile

```
define Package/juci
  SECTION:=juci
  CATEGORY:=JUCI
  MENU:=1
  TITLE:=JUCI Web Interface
  DEPENDS:=+lighttpd +lighttpd-mod-websocket +lighttpd-mod-proxy +lighttpd-mod-cgi 
endef
```

mkschreder, 2016-08-30:
Why are you building lighttpd as module? That will only build package but not install it on your system. Anyways, try building it as built in and see if that helps (it should be selected using star *, not M),

mkschreder, 2016-08-31:
Check out my latest readme.md instructions in v2.16.08 branch.

closed 2016-08-31 by mkschreder