mkschreder/juci #50: Juci working partially

cfcnightmare, 2016-06-28:
Hello
Trying to make Juci working for about 5 month?,no success...
Target X86 (64)
System is Ubuntu 16.04 (14.04.12.05):
npm install
npm -g install nodejs
npm -g install express

adduser openwrt
passwd openwrt
su openwrt

git clone git://git.openwrt.org/15.05/openwrt.git
chown -R openwrt openwrt
cd openwrt
nano feeds.conf.default

src-git juci https://github.com/mkschreder/juci-openwrt-feed.git;v2.16.02
src-git packages https://github.com/openwrt/packages.git;for-15.05
src-git luci https://github.com/openwrt/luci.git;for-15.05
src-git routing https://github.com/openwrt-routing/packages.git;for-15.05
src-git telephony https://github.com/openwrt/telephony.git;for-15.05
src-git management https://github.com/openwrt-management/packages.git;for-15.05
#src-git targets https://github.com/openwrt/targets.git
#src-git oldpackages http://git.openwrt.org/packages.git
#src-svn xwrt http://x-wrt.googlecode.com/svn/trunk/package
#src-svn phone svn://svn.openwrt.org/openwrt/feeds/phone
#src-svn efl svn://svn.openwrt.org/openwrt/feeds/efl
#src-svn xorg svn://svn.openwrt.org/openwrt/feeds/xorg
#src-svn desktop svn://svn.openwrt.org/openwrt/feeds/desktop
#src-svn xfce svn://svn.openwrt.org/openwrt/feeds/xfce
#src-svn lxde svn://svn.openwrt.org/openwrt/feeds/lxde
#src-link custom /usr/src/openwrt/custom-feed

./scripts/feeds update -a
./scripts/feeds install -a
./scripts/feeds install -f -p juci -a

Overriding 3 packages...
Make menuconfig
Select juci-full ONLY

make V=s -j 3

successful compile!

openwrt-x86-64-combined-ext4 write to usb disk with physdiskwrite...

Openwrt loads on target PC
 
From PC in local network:
192.168.1.1

See Login screen
Using root login & pass.
[image]

And here:
JS errors
[image]
[image]

Ok waiting....

Can click only system.
If i click other button -  blank screen.

Tried with openwrt versions:
15.05,14.07,trunk
Tried Juci branches:
2.16.02,2.16.03.2.16.04, for-15.07

Dear Mkschreder can you say what im doing incorrect?

mkschreder, 2016-06-28:
nr1: always install custom feeds first, then default openwrt feeds. This is
true for any custom feed that contains overrides.
nr2: select a theme.

https://github.com/openwrt-management/packages.git;for-15.05

<https://cloud.githubusercontent.com/assets/9579109/16409936/b85b1098-3d28-11e6-86ef-23079d6343cc.png>

<https://cloud.githubusercontent.com/assets/9579109/16409954/c8b43942-3d28-11e6-9f96-0922ddbbbbc4.png>
[image: juci js error1]
<https://cloud.githubusercontent.com/assets/9579109/16409955/c8b60880-3d28-11e6-8904-8e319c9fc484.png>

cfcnightmare, 2016-06-28:
so i need to do:
./scripts/feeds update -a
./scripts/feeds install -juci
./scripts/feeds install -a
./scripts/feeds install -f -p juci -a

like this?
Theme is selected by default when i select juci-full in menuconfig

mkschreder, 2016-07-05:
Yes that looks correct. I have been busy. I will go over build instructions from the beginning when I have time. 

You should be able to use theme-openwrt. Inteno theme is not there anymore because inteno has their own heavily customized version of JUCI for iopsys. I have therefore removed many of the plugins that were previously in the source tree because the do not work anyway without the customized base system. Also main tree now supports building external plugins so you can have plugins in a different tree and build it using the main juci Makefile. Unfortunately current acrobatics in the build process also mess up openwrt package versioning which makes it more difficult to install packages separately (and for example building an image using image builder). This is pretty low on my list of priorities right now. The issue is that plugin packages are mostly dummy packages. All juci code actually goes into the main juci package. This is not correct. 

It is difficult to keep updated with openwrt default because the systems I work with are all heavily customized, but I try to keep things generic so that only minimal work is required to port packages between different system. It is unfortunate that you are having problems building on OpenWRT. I can only fix it when I have time. If you can make sure it works on your system and submit a patch for all necessary fixes then it will work sooner on your setup (and also you will be living up to the spirit of open source :-)).

closed 2016-08-30 by mkschreder