mkschreder/juci #51: Why compile juci firmware into the management interface are all blank, what all don't show?

daofeng2015, 2016-07-17:
[image]
Why？

mkschreder, 2016-07-17:
What's output of "grep juci .config"?

daofeng2015, 2016-07-17:
[qca-ttl打印.txt](https://github.com/mkschreder/juci/files/367909/qca-ttl.txt)
This file?

mkschreder, 2016-07-17:
No. The one in your openwrt root folder.

mkschreder, 2016-07-17:
Also do "cat feeds.conf" and paste it here.

daofeng2015, 2016-07-17:
`
src-git packages https://github.com/openwrt/packages.git;for-15.05
#src-git luci https://github.com/openwrt/luci.git;for-15.05
src-git-full juci https://github.com/mkschreder/juci-openwrt-feed.git
src-git routing https://github.com/openwrt-routing/packages.git;for-15.05
src-git telephony https://github.com/openwrt/telephony.git;for-15.05
src-git management https://github.com/openwrt-management/packages.git;for-15.05
#src-git targets https://github.com/openwrt/targets.git
#src-git oldpackages http://git.openwrt.org/packages.git
#src-svn xwrt http://x-wrt.googlecode.com/svn/trunk/package
#src-svn phone svn://svn.openwrt.org/openwrt/feeds/phone
#src-svn efl svn://svn.openwrt.org/openwrt/feeds/efl
#src-svn xorg svn://svn.openwrt.org/openwrt/feeds/xorg
#src-svn desktop svn://svn.openwrt.org/openwrt/feeds/desktop
#src-svn xfce svn://svn.openwrt.org/openwrt/feeds/xfce
#src-svn lxde svn://svn.openwrt.org/openwrt/feeds/lxde
#src-link custom /usr/src/openwrt/custom-feed`

mkschreder, 2016-07-17:
And "grep juci .config"?

welcoco, 2016-07-18:
I have the same problem,and I found "JUCI_MOBILE_BUNDLE" is undefined ,but I found this 
   library: prepare
	@echo "======= JUCI LIBRARY ========="
	@mkdir -p $(JUCI_LIB_PATH)
	@echo "Including jquery and async.js into JUCI library!"
	@echo "var JUCI_MOBILE_BUNDLE=true;">> $(JUCI_LIB_OUTPUT)
	@awk 'FNR==1{print ""}1' $(JUCI_LIB_OBJECTS) >> $(JUCI_LIB_OUTPUT)
	@echo "Library has been built: $(JUCI_LIB_OUTPUT)"

so ,is the makefile is not execute here？

mkschreder, 2016-07-18:
Is fixed since a while back in branch v2.16.07.

Latest feed branch v2.16.07 should also point to a working copy.

welcoco, 2016-07-19:
document.write("<script langrage=javascrikpt src='/lib/*.js'></script>");

daofeng2015, 2016-07-25:
Thank you, has been solved, the browser to empty the cache have to do is log in again, thank you!

welcoco, 2016-08-04:
@daofeng2015  How do you solve this problem? I still have this problem,I revise the Makefile,but there is no effect.

mkschreder, 2016-08-04:
Try this config: 

# cat >> .config
CONFIG_FEED_juci=y
CONFIG_PACKAGE_juci-ddns=y
CONFIG_PACKAGE_juci-diagnostics=y
CONFIG_PACKAGE_juci-dnsmasq-dhcp=y
CONFIG_PACKAGE_juci-dropbear=y
CONFIG_PACKAGE_juci-ethernet=y
CONFIG_PACKAGE_juci-event=y
CONFIG_PACKAGE_juci-firewall-fw3=y
CONFIG_PACKAGE_juci-freecwmp=y
# CONFIG_PACKAGE_juci-igmpinfo is not set
CONFIG_PACKAGE_juci-jquery-console=y
# CONFIG_PACKAGE_juci-macdb is not set
CONFIG_PACKAGE_juci-minidlna=y
# CONFIG_PACKAGE_juci-minisnmpd is not set
CONFIG_PACKAGE_juci-mod-status=y
CONFIG_PACKAGE_juci-mod-system=y
CONFIG_PACKAGE_juci-network-netifd=y
CONFIG_PACKAGE_juci-openwrt-wireless=y
# CONFIG_PACKAGE_juci-realtime-graphs is not set
CONFIG_PACKAGE_juci-samba=y
CONFIG_PACKAGE_juci-simple-gui=y
CONFIG_PACKAGE_juci-snmpd=y
CONFIG_PACKAGE_juci-sysupgrade=y
CONFIG_PACKAGE_juci-uhttpd=y
CONFIG_PACKAGE_juci-upnp=y
CONFIG_PACKAGE_juci-usb=y
# CONFIG_PACKAGE_juci-utils is not set
CONFIG_PACKAGE_juci-theme-openwrt=y
CONFIG_PACKAGE_juci=y
CONFIG_PACKAGE_juci-full-openwrt=y
CONFIG_PACKAGE_orange-rpcd=y
^D

# make defconfig

daofeng2015, 2016-08-04:
No CONFIG_PACKAGE_orange-rpcd=y

mkschreder, 2016-08-30:
What does browser console say?

closed 2016-11-05 by daofeng2015