mkschreder/juci #52: v2.16.08 question

cfcnightmare, 2016-08-04:
Hello
Successfuly compiled v2.16.08
Had one issue with luaposix but changed to v33.4.0 in Makefile and its compiled....
But, i cant access GUI.
In v2.16.01 i enter only login root and thats it , but here its dont working...
Can you help me with this?
And can you change luaposix33 version by default?

mkschreder, 2016-08-19:
pushed new command line util that will set the backend password. The syntax is "orangectl passwd user pass". This script is part of orangerpcd package in juci feed. Note that passwords are just stored as sha1 right now so switching to a salted hash is one of the todo items for this.

shufenzhang, 2016-08-19:
Hi Martin,

Thanks for your quick fix for login issue. Could you tell what specific steps I need to include your fix?

Best Regards,
Shufen

mkschreder, 2016-08-19:
You can pull latest v2.16.08  juci-openwrt-feed. But be careful, it
includes some changes that have not been tested yet (I have improved uci
permissions validation on the backend today which needs a few days to sink
in. I will also add more improvements to it. But it can possibly cause some
permission issues right now since client can no longer see all uci
configs.).

shufenzhang, 2016-08-19:
Hi Martin,

Thanks, so it might be better if tell me the specific file I can modify with and just add this line. Otherwise the whole build will be broken again. What do you think?
Is this JuCi more complicated than the Luci for us to use and to extend?

Thanks,
Shufen

cfcnightmare, 2016-08-21:
Orangectl passwd root admin   - not found

mkschreder, 2016-08-21:
Installed by this makefile:
https://github.com/mkschreder/juci-openwrt-feed/blob/v2.16.08/orange-rpcd/Makefile

cfcnightmare, 2016-08-21:
I cleaned all and updated from tree 2.16.08.
In usr/bin i see it but it dont works.

cfcnightmare, 2016-08-21:
And again.... I want to ask you to change luaposix version in Makefile from 33.3.1 to 33.4.0 because of error in compiling process. Its  not comfortable to change it every time after distclean.....

shufenzhang, 2016-08-21:
Hi Marti,

I could not avoid build error.

Thanks,

Shufen

/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt/scripts/download.pl "/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt/dl" "release-v34.1.0.tar.gz" "unknown" "" "https://github.com/luaposix/luaposix/archive/"
--2016-08-21 15:29:07--  https://github.com/luaposix/luaposix/archive/release-v34.1.0.tar.gz
Resolving github.com (github.com)... 192.30.253.112
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/luaposix/luaposix/tar.gz/release-v34.1.0 [following]
--2016-08-21 15:29:07--  https://codeload.github.com/luaposix/luaposix/tar.gz/release-v34.1.0
Resolving codeload.github.com (codeload.github.com)... 192.30.253.120
Connecting to codeload.github.com (codeload.github.com)|192.30.253.120|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-08-21 15:29:08 ERROR 404: Not Found.

Download failed.
--2016-08-21 15:29:08--  http://mirror2.openwrt.org/sources/release-v34.1.0.tar.gz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-08-21 15:29:08 ERROR 404: Not Found.

Download failed.
--2016-08-21 15:29:08--  http://downloads.openwrt.org/sources/release-v34.1.0.tar.gz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-08-21 15:29:08 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[3]: *** [/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt/dl/release-v34.1.0.tar.gz] Error 2
make[3]: Leaving directory `/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt/feeds/juci/luaposix33'
make[2]: *** [package/feeds/juci/luaposix33/compile] Error 2
make[2]: Leaving directory `/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt'
make[1]: *** [/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt'
make: *** [world] Error 2
[szhang@szhang-linux openwrt]$

https://github.com/mkschreder/juci-openwrt-feed/blob/v2.16.08/orange-rpcd/Makefile

shufenzhang, 2016-08-21:
Hi Martin,

Sorry that I typed your name wrong in the previous email and got compilation error.

Not sure if this can be fixed.

Thanks,

Shufen

________________________________
From: Shufen Zhang
Sent: Sunday, August 21, 2016 3:33:40 PM
To: mkschreder/juci
Subject: Re: [mkschreder/juci] v2.16.08 question (#52)

Hi Marti,

I could not avoid build error.

Thanks,

Shufen

/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt/scripts/download.pl "/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt/dl" "release-v34.1.0.tar.gz" "unknown" "" "https://github.com/luaposix/luaposix/archive/"
--2016-08-21 15:29:07--  https://github.com/luaposix/luaposix/archive/release-v34.1.0.tar.gz
Resolving github.com (github.com)... 192.30.253.112
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/luaposix/luaposix/tar.gz/release-v34.1.0 [following]
--2016-08-21 15:29:07--  https://codeload.github.com/luaposix/luaposix/tar.gz/release-v34.1.0
Resolving codeload.github.com (codeload.github.com)... 192.30.253.120
Connecting to codeload.github.com (codeload.github.com)|192.30.253.120|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-08-21 15:29:08 ERROR 404: Not Found.

Download failed.
--2016-08-21 15:29:08--  http://mirror2.openwrt.org/sources/release-v34.1.0.tar.gz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-08-21 15:29:08 ERROR 404: Not Found.

Download failed.
--2016-08-21 15:29:08--  http://downloads.openwrt.org/sources/release-v34.1.0.tar.gz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-08-21 15:29:08 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[3]: *** [/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt/dl/release-v34.1.0.tar.gz] Error 2
make[3]: Leaving directory `/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt/feeds/juci/luaposix33'
make[2]: *** [package/feeds/juci/luaposix33/compile] Error 2
make[2]: Leaving directory `/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt'
make[1]: *** [/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/szhang/Projects/Adaptrum/openwrt_juci_bb/openwrt'
make: *** [world] Error 2
[szhang@szhang-linux openwrt]$

https://github.com/mkschreder/juci-openwrt-feed/blob/v2.16.08/orange-rpcd/Makefile

mkschreder, 2016-08-22:
> luaposix/archive/"
> luaposix/archive/release-v34.1.0.tar.gz
> release-v34.1.0 [following]
> luaposix/luaposix/tar.gz/release-v34.1.0
> sources/release-v34.1.0.tar.gz
> 08/orange-rpcd/Makefile

Luaposix updated to 33.4.0.

@shufenzhang: you have changed your luaposix version to a version that has
not been released yet.

cfcnightmare, 2016-08-22:
Orangectl command dont working.
Compiled today version of openwrt-juci-feed and when rtying to execute it console wrote : not found!  But in usr/bin i see the time orangectl. And in orangercpd  file  (etc/init.d) i see a lot of unknown symbols :)

cfcnightmare, 2016-08-22:
If you want Martin i can add this file and you can look :)

closed 2016-08-23 by cfcnightmare