mkschreder/juci #53: Error: Cannot find module './index'

shufenzhang, 2016-08-10:
Anyone see this error? 
Running "nggettext_compile:all" (nggettext_compile) task

Done.
Compiling JS bin/www/js/01-juci.js..
Compiling JS bin/www/js/01-juci.tpl.js..
Compiling JS bin/www/js/99-translations.js..
Compiling CSS bin/www/css/01-juci.css..
which: no yui-compressor in (/path/OpenWrt_JuCi/juciwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.11/host/bin:/path/OpenWrt_JuCi/juciwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.11/bin:/path/OpenWrt_JuCi/juciwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.11/bin:/path/OpenWrt_JuCi/juciwrt/staging_dir/host/bin:/path/OpenWrt_JuCi/juciwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.11/bin:/path/OpenWrt_JuCi/juciwrt/staging_dir/host/bin:/path/OpenWrt_JuCi/juciwrt/staging_dir/host/bin:/usr/lib64/qt-3.3/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin)

module.js:340
    throw err;
          ^
Error: Cannot find module './index'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:280:25)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/usr/bin/yuicompressor:10:18)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
make[4]: *** [release] Error 8
make[4]: Leaving directory `/path/OpenWrt_JuCi/juciwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.11/juci-74d402a457e6cbabeb6e1e7acf75f8274a9eb274'
make[3]: *** [/path/OpenWrt_JuCi/juciwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.11/juci-74d402a457e6cbabeb6e1e7acf75f8274a9eb274/.built] Error 2
make[3]: Leaving directory `/path/OpenWrt_JuCi/juciwrt/feeds/juci/juci'
make[2]: *** [package/feeds/juci/juci/compile] Error 2
make[2]: Leaving directory `/path/OpenWrt_JuCi/juciwrt'
make[1]: *** [/path/OpenWrt_JuCi/juciwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.11/stamp/.package_compile] Error 2
make[1]: Leaving directory `/path/OpenWrt_JuCi/juciwrt'
make: *** [world] Error 2

mkschreder, 2016-08-11:
> at Object. (/usr/bin/yuicompressor:10:18)

... no yui-compressor in ...

Means yui-compressor is not installed. Note that this is very very old
version you are using. Is this stuff downloaded from juciwrt repo?

shufenzhang, 2016-08-11:
Hi Markin,

Thanks for your reply!!
I think so, but I also tried to install this yui-compressor separately, but it didn’t fix the problem either.
But for whatever under this directory are all downloaded from the repo.

Best Regards,
Shufen

mkschreder, 2016-08-11:
use juci-openwrt-feed together with openwrt cc.

shufenzhang, 2016-08-11:
How to use them together?
Is it to run “git clone https://github.com/mkschreder/juci-openwrt-feed” under openwrt source directory?

Thanks,
Shufen

mkschreder, 2016-08-11:
add it as a feed to feeds.conf in openwrt.

shufenzhang, 2016-08-11:
Thanks, I did it. It worked. I will let you know later while I proceed ..

Great thanks,
Shufen

<mailto:juci@noreply.github.com%3cmailto:%0b>> juci@noreply.github.com<mailto:juci@noreply.github.com>>>
<mailto:notifications@github.com%3c%0b>> mailto:notifications@github.com>>

closed 2016-08-19 by mkschreder