mkschreder/juci #54: default GUI password !

cfcnightmare, 2016-08-23:
Issue isnt gone with todays update.
Orangectl : not found
The better solution i think is to make root user is user for GUI by default. Then we can change password inside GUI.....

mkschreder, 2016-08-30:
Your error with orangectl is in fact misleading. It is because your system does not have /bin/bash (default config only has sh). So it looks like the script itself is missing, when in fact the shell requested by the script is missing. I have changed that to use sh and I have also updated the script with adduser command. 

Now you can do: 
orangectl adduser admin
orangectl passwd admin admin

You will need to have sha1sum selected in busybox custom settings in your config, since it is required for setting the password hash.

cfcnightmare, 2016-08-30:
I will compiled new version  today and look ...
Anyway i have the script and selected bash and whoami in menuconfig.... But had another issue with selecting that modules. When its compiled there is no rpcd connecting to a router (GUI says)

mkschreder, 2016-08-30:
I don't know what that could be but try todays update. It could have been a
problem with libblobpack. I have added many new unit tests and code
coverage to that lib now so another issue that I have seen with rpc server
crashing is also fixed now. I have also fixed overview widget in the
default theme so it does not result in javascript error on login.

Also, build now tested in user mode linux and builds fine with openwrt cc.
You should be fine with todays updates.

cfcnightmare, 2016-08-30:
Ok man, i will try today.... No its 1.21 am there :)

closed 2016-08-31 by cfcnightmare