mkschreder/juci #55: Juci GUI problem

TadeuGRibeiro, 2016-08-24:
Hello Martin, I'm beginer with openwrt and I loved this interface JUCI... but I have a problem with WRTNode compilation. I do compile and get successfull results. But after update I get this problem show in follow picture.

[image]

Can you help me with this problem?

Thanks a lot.

mkschreder, 2016-08-24:
You have luci redirect in cache so clear your browser cache.

TadeuGRibeiro, 2016-08-24:
I tried this, but the problem persists.... I tried do the access by my smartphone and same problem...

mkschreder, 2016-08-24:
Make sure luci packages are not selected in your build.

TadeuGRibeiro, 2016-08-24:
I removed LUCI packages by MAKE MENUCONFIG...

TadeuGRibeiro, 2016-08-25:
Hi Martin... Worked! I disabled LUA in config and success!
But now I get other problem....

[image]

Do you have any adea?

Thanks!

mkschreder, 2016-08-30:
No idea why you get blank page. Todays test build with uml linux works fine so try building with todays update to the feed and see if you still have problem.

mkschreder, 2016-08-30:
What does browser console say?

TadeuGRibeiro, 2016-08-31:
I will try rebuild.

The console say: 
        RPC init
        connection to rpc server at (ws://192.168.8.1/websocket/)
        RPC connection established!

Thanks and best regards!

TadeuGRibeiro, 2016-08-31:
Martin, I rebuild with last update and get this erro.... Thanks and best regards.

``e_ws_server.o -MD -MP -MF .deps/orangerpcd-orange_ws_server.Tpo -c -o orangerpcd-orange_ws_server.o `test -f 'orange_ws_server.c' || echo './'`orange_ws_server.c
orange_ws_server.c:119:56: error: 'enum lws_callback_reasons' declared inside parameter list [-Werror]
 static int _ubus_socket_callback(struct lws *wsi, enum lws_callback_reasons reason, void *_user, void *in, size_t len){
                                                        ^
orange_ws_server.c:119:56: error: its scope is only this definition or declaration, which is probably not what you want [-Werror]
orange_ws_server.c:119:77: error: parameter 2 ('reason') has incomplete type
 static int _ubus_socket_callback(struct lws *wsi, enum lws_callback_reasons reason, void *_user, void *in, size_t len){
                                                                             ^
orange_ws_server.c: In function '_ubus_socket_callback':
orange_ws_server.c:121:15: error: implicit declaration of function 'lws_get_protocol' [-Werror=implicit-function-declaration]
  const struct lws_protocols *proto = lws_get_protocol(wsi); 
               ^
orange_ws_server.c:121:38: error: initialization makes pointer from integer without a cast [-Werror]
  const struct lws_protocols *proto = lws_get_protocol(wsi); 
                                      ^
orange_ws_server.c:130:2: error: implicit declaration of function 'lws_get_socket_fd' [-Werror=implicit-function-declaration]
  int32_t peer_id = lws_get_socket_fd(wsi); 
  ^
orange_ws_server.c:133:57: error: dereferencing pointer to incomplete type
    struct ubus_srv_ws *self = (struct ubus_srv_ws*)proto->user; 
                                                         ^
orange_ws_server.c:139:4: error: implicit declaration of function 'lws_get_peer_addresses' [-Werror=implicit-function-declaration]
    lws_get_peer_addresses(wsi, peer_id, hostname, sizeof(hostname), ipaddr, sizeof(ipaddr)); 
    ^
orange_ws_server.c:144:4: error: implicit declaration of function 'lws_callback_on_writable' [-Werror=implicit-function-declaration]
    lws_callback_on_writable(wsi);  
    ^
orange_ws_server.c:152:57: error: dereferencing pointer to incomplete type
    struct ubus_srv_ws *self = (struct ubus_srv_ws*)proto->user; 
                                                         ^
orange_ws_server.c:162:57: error: dereferencing pointer to incomplete type
    struct ubus_srv_ws *self = (struct ubus_srv_ws*)proto->user; 
                                                         ^
orange_ws_server.c:180:5: error: implicit declaration of function 'lws_write' [-Werror=implicit-function-declaration]
     int n = lws_write(wsi, &frame->buf[LWS_SEND_BUFFER_PRE_PADDING]+frame->sent_count, len, flags);
     ^
orange_ws_server.c:199:4: error: implicit declaration of function 'lws_rx_flow_control' [-Werror=implicit-function-declaration]
    lws_rx_flow_control(wsi, 1); 
    ^
orange_ws_server.c:206:57: error: dereferencing pointer to incomplete type
    struct ubus_srv_ws *self = (struct ubus_srv_ws*)proto->user; 
                                                         ^
orange_ws_server.c:214:4: error: implicit declaration of function 'lws_is_final_fragment' [-Werror=implicit-function-declaration]
    if(lws_is_final_fragment(wsi)){
    ^
orange_ws_server.c: In function '_websocket_destroy':
orange_ws_server.c:286:2: error: implicit declaration of function 'lws_context_destroy' [-Werror=implicit-function-declaration]
  if(self->ctx) lws_context_destroy(self->ctx); 
  ^
orange_ws_server.c: In function '_websocket_listen':
orange_ws_server.c:321:17: error: assignment from incompatible pointer type [-Werror]
  info.protocols = self->protocols; 
                 ^
orange_ws_server.c:323:17: error: 'LWS_SERVER_OPTION_VALIDATE_UTF8' undeclared (first use in this function)
  info.options = LWS_SERVER_OPTION_VALIDATE_UTF8;
                 ^
orange_ws_server.c:323:17: note: each undeclared identifier is reported only once for each function it appears in
orange_ws_server.c:325:2: error: implicit declaration of function 'lws_create_context' [-Werror=implicit-function-declaration]
  self->ctx = lws_create_context(&info); 
  ^
orange_ws_server.c:325:12: error: assignment makes pointer from integer without a cast [-Werror]
  self->ctx = lws_create_context(&info); 
            ^
orange_ws_server.c: In function '_websocket_server_thread':
orange_ws_server.c:339:3: error: implicit declaration of function 'lws_service' [-Werror=implicit-function-declaration]
   lws_service(self->ctx, 10); 
   ^
orange_ws_server.c: In function 'orange_ws_server_new':
[... 21 lines cut ...]
   ^
orange_ws_server.c:426:42: error: excess elements in struct initializer [-Werror]
   .per_session_data_size = sizeof(struct ubus_srv_ws_client*),
                                          ^
orange_ws_server.c:426:42: error: (near initialization for '(anonymous)') [-Werror]
orange_ws_server.c:427:3: error: unknown field 'user' specified in initializer
   .user = self
   ^
orange_ws_server.c:428:2: error: excess elements in struct initializer [-Werror]
  };
  ^
orange_ws_server.c:428:2: error: (near initialization for '(anonymous)') [-Werror]
orange_ws_server.c:428:2: error: invalid use of undefined type 'struct lws_protocols'
cc1: all warnings being treated as errors
make[5]: *** [orangerpcd-orange_ws_server.o] Error 1
[... cut ...]
make[3]: *** [/home/tadeu/compilacoes/23-08/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/orange-rpcd-fd0af57bbef7b95a4a03e9de8278de943ae337d6/.built] Error 2
make[3]: Leaving directory `/home/tadeu/compilacoes/23-08/openwrt/feeds/juci/orange-rpcd'
make[2]: *** [package/feeds/juci/orange-rpcd/compile] Error 2
make[2]: Leaving directory `/home/tadeu/compilacoes/23-08/openwrt'
make[1]: *** [/home/tadeu/compilacoes/23-08/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/tadeu/compilacoes/23-08/openwrt'
make: ** [world] Erro 2

cfcnightmare, 2016-08-31:
try:
make clean
make dirclean
make distclean
./scripts/feeds uninstall -a
./scripts/feeds update -a
./scripts/feeds update juci
./scripts/feeds install -f -p juci -a
./scripts/feeds install -a

install ubuntu-bootstrap script (look at juci tree,root needed)
then npm init in openwrt folder

nano feeds.conf.default
add juci-openwrt-feed there
make menuconfig
select all juci packages
save
exit
make defconfig
make V=s -j numberofcorestouse+1 (make V=s -j 5 if cores is 4)

TadeuGRibeiro, 2016-09-01:
I tried this commands but same errors... One point, for compiler I remove all references of LUA in .config file, JUCI has some dependece of LUA?

Thanks!

mkschreder, 2016-09-01:
You can not just remove dependencies. Run make defconfig to fix any missing
dependencies. Juci rpc backend is done in lua so yes lua is necessary.

Did you have any problems building lua? Why did you remove it? Did you try
to build and test a user mode linux image first as outlined in the v2.16.08
readme?

cfcnightmare, 2016-09-01:
Martin today build need fix..
It shows: you do have rpcd connection to a router

TadeuGRibeiro, 2016-09-02:
Yes, I got this error when I build....

make[5]: Entering directory `/home/tadeu/compilacoes/23-08/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/luaposix-release-v33.3.1'
  CC       ext/posix/posix.lo
In file included from ext/posix/ctype.c:22:0,
                 from ext/posix/posix.c:13:
./ext/include/_helpers.c:38:18: fatal error: term.h: No such file or directory
 #include <term.h>
                  ^
compilation terminated.
make[5]: *** [ext/posix/posix.lo] Error 1
make[5]: Leaving directory `/home/tadeu/compilacoes/23-08/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/luaposix-release-v33.3.1'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/tadeu/compilacoes/23-08/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/luaposix-release-v33.3.1'
make[3]: *** [/home/tadeu/compilacoes/23-08/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/luaposix-release-v33.3.1/.built] Error 2
make[3]: Leaving directory `/home/tadeu/compilacoes/23-08/openwrt/feeds/juci/luaposix33'
make[2]: *** [package/feeds/juci/luaposix33/compile] Error 2
make[2]: Leaving directory `/home/tadeu/compilacoes/23-08/openwrt'
make[1]: *** [/home/tadeu/compilacoes/23-08/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/tadeu/compilacoes/23-08/openwrt'
make: ** [world] Erro 2

Because this I removed LUA dependecies.

Thanks.

mkschreder, 2016-09-02:
It should be fixed in v2.16.08. Had to update luaposix package.

mkschreder, 2016-09-02:
I will merge v2.16.08 into master soon..

TadeuGRibeiro, 2016-09-02:
I tryed updated luaposix33 with: ./scripts/feeds updater luaposix33 or ./scripts/feeds update luaposix and got same error. Whats the command I need use for update?
Thanks.

mkschreder, 2016-09-02:
Here is a script that builds uml image:
https://github.com/mkschreder/juci-openwrt-feed/blob/master/build-uml-image.sh

I'm testing it right now...

Here is a bootstrap script:
https://github.com/mkschreder/juci-openwrt-feed/blob/master/openwrt-bootstrap.sh

You can study both and see if you are missing anything..

Luaposix33 was update in v2.16.08 of the feed. Most examples build from
master but to have 2.16.08 for example you need to add ;v2.16.08 after feed
url in your feeds.conf.

TadeuGRibeiro, 2016-09-03:
I updated all files and now I got a new problem. I read in other topic about this, when the solution is put JUCI feeds before Openwrt, but dont works.

[image]

mkschreder, 2016-09-04:
It seems you do not have websocket on the /websocket/ path of the
webserver. This typically happens if you have wrong lighttpd (without
websocket support). Most probably cause is not having the correct package
(the one from juci-openwrt-feed) installed. You can do "./scripts/feeds
uninstall -a" and then install the packages again by doing "./scripts/feeds
install -f -a -p juci; ./scripts/feeds install -a; make defconfig; make".

The user mode linux build script that I have uploaded to the feed repo
works and you can use that as a reference. I did notice a few other
problems on the vanilla openwrt build that need fixing, but they are all
behind the login screen. This issue does not occur on the uml build so
please make sure that your build environment has been set up correctly and
that correct packages are being built.

cfcnightmare, 2016-09-04:
Why we need to use uml image? Why i cant use standard X86 image (ext4). 
In you instructions of creating uml image see incorrect commands (i tried)
1) sudo tunctl -n <your user id>         this command not exists
2) if i do like you wrote i got squash.img but i need ext4, because it didnt start (shows kernel panic)

mkschreder, 2016-09-04:
> 1) sudo tunctl -n this command not exists

tunctl is part of uml-utilities package.

Anyways, uml build script is a good example of a script that works and
actually builds a working image from scratch. You can build whatever image
you want yourself, but uml build script shows you what steps to take even
if the final result may not be what you are after. It is convenient because
it builds a standard executable and a rootfs file. All you have to do is
run the executable to start openwrt - no need to use an emulator etc..
Although to access it's network interface you do need tunctl to set up a
tun device and also you need to configure route to your image - all fairly
basic network configuration steps. If any utilities are missing like tunctl
then just install them. Uml image script should build you an ext 4
filesystem too (it sets that explicitly). Uml image is not meant to be a
starting point for a project - use it as a build process reference only.

TadeuGRibeiro, 2016-09-05:
I build again, I download all files again and get same error. Now I tried enabled CONFIG_PACKAGE_libwebsockets-cyassl=y
CONFIG_PACKAGE_libwebsockets-openssl=y
in .config file, but I get this new error.

[image]

If I enable only CONFIG_PACKAGE_libwebsockets-openssl=y I get sucess with complilation, but I get RPC connection fail....

mkschreder, 2016-09-05:
Did you build according to the instructions?

TadeuGRibeiro, 2016-09-05:
Yes... But I dont know whats is wrong...

mkschreder, 2016-09-05:
Post exact commands you ran.

TadeuGRibeiro, 2016-09-05:
make clean
make dirclean
make distclean
./scripts/feeds uninstall -a
./scripts/feeds update -a
./scripts/feeds update juci
./scripts/feeds install -f -p juci -a
./scripts/feeds install -a

make menuconfig
make defconfig
make V=s -j1

mkschreder, 2016-09-05:
That is not complete list. For example it does not say which version of
juci feed you are using. Also it does not say which version of openwrt you
are building on. And neither does it say exactly what you are enabling in
menuconfig.

If you base your build on uml build script then all these questions are
crystal clear and answered inside the build script.

TadeuGRibeiro, 2016-09-05:
I dont know whats openwrt version is... I cloned this repository: https://github.com/WRTnode/openwrt.git and for juci I added src-git packages https://github.com/openwrt/packages.git;for-15.05 in feeds.config.default

My feeds complete is:

src-git packages https://github.com/openwrt/packages.git;for-15.05
#src-git luci https://github.com/openwrt/luci.git;for-15.05
src-git routing https://github.com/openwrt-routing/packages.git;for-15.05
src-git telephony https://github.com/openwrt/telephony.git;for-15.05
src-git management https://github.com/openwrt-management/packages.git;for-15.05
src-git wrtnode https://github.com/wrtnode/openwrt-packages.git
#src-git targets https://github.com/openwrt/targets.git
src-git oldpackages https://git.openwrt.org/packages.git
#src-svn xwrt http://x-wrt.googlecode.com/svn/trunk/package
#src-svn phone svn://svn.openwrt.org/openwrt/feeds/phone
#src-svn efl svn://svn.openwrt.org/openwrt/feeds/efl
#src-svn xorg svn://svn.openwrt.org/openwrt/feeds/xorg
#src-svn desktop svn://svn.openwrt.org/openwrt/feeds/desktop
#src-svn xfce svn://svn.openwrt.org/openwrt/feeds/xfce
#src-svn lxde svn://svn.openwrt.org/openwrt/feeds/lxde
#src-link custom /usr/src/openwrt/custom-feed
src-git-full juci https://github.com/mkschreder/juci-openwrt-feed.git

im menuconfig I only select my WRTNode as Target, select all juci packages and libwebsockets-cyassl and libwebsockets-openssl. After make menuconfig and make V=s -j1.

TadeuGRibeiro, 2016-09-05:
I try run uml build script but...
[image]

cfcnightmare, 2016-09-05:
And thats my commands
cd
rm -rf openwrt
git clone git://git.openwrt.org/15.05/openwrt.git
echo "src-git-full juci https://github.com/mkschreder/juci-openwrt-feed.git" >> feeds.conf.default

Then commented luci in feeds

./scripts/feeds update -a 
./scripts/feeds install -f -a -p juci
./scripts/feeds install -a

Then:

Make menuconfig
Selected sha1sum, ALL juci packages

make V=s -j 5

RPC connection problem....
DO i need to select libwebsockets-cyassl && libwebsockets-openssl ???

mkschreder, 2016-09-05:
> juci-openwrt-feed.git" >> feeds.conf.default

No. In fact you should not be selecting all juci packages either. Select
only juci-full-openwrt and orange-rpcd and also busybox ip and sha1sum
utility (see the uml build script - it's all there). You do not need to
select ANYTHING other than whats in the uml build script.

cfcnightmare, 2016-09-05:
Im making hardware server right Now on xeon e5-2670 (8c/16t) when its compiled i dive you access and you will try... But 99% result will still the same i think. Its not hardware to make a build but the rpcd error is boring...

shufenzhang, 2016-09-05:
Hi Martin,

Could  you give a link to your UML build instruction?

Thanks,
Shufen

mkschreder, 2016-09-05:
Checks out fresh openwrt cc and calls make:
https://github.com/mkschreder/juci-openwrt-feed/blob/master/build-uml-image.sh
Configures openwrt build with juci:
https://github.com/mkschreder/juci-openwrt-feed/blob/master/openwrt-bootstrap.sh

TadeuGRibeiro, 2016-09-05:
Martin, I tried use build-uml-image.sh but I got this 
[image]
error.

mkschreder, 2016-09-05:
You are unable to build cmake host package? Strange..

Besides, you are still dicking around and doing something wierd because the
script builds using just make -j8 which does not generate so much extra
output. You sure about consistency of that repo?

TadeuGRibeiro, 2016-09-05:
Is very strange... I tried with -j8, but I got error and I changed for V=s -j1 to show the errors....
This repo is consistent , without LUCI work fine....

TadeuGRibeiro, 2016-09-06:
Martin I formatted my computer, reinstalled everything and tried again. I'm use build-uml-image.sh without any change , but in the process of finishing I got this problem...
Can you help me? I do not know what I do wrong...

[image]

Thanks a lot.

cfcnightmare, 2016-09-06:
Wait pls, im compiling now. I wrote if get the same

mkschreder, 2016-09-06:
I see you are using ubuntu.. maybe try symlinking your /bin/sh to bash
instead of dash? It is an issue when building openwrt in general
sometimes..

Also post the actual error that you get this time when you run make with
V=s.

cfcnightmare, 2016-09-06:
make[4]: Entering directory '/home/openwrt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/orange-rpcd-2.16.08-c4ad159efa5fc40f610fa0cb11e25fda19f131ad'
Making all in src
make[5]: Entering directory '/home/openwrt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/orange-rpcd-2.16.08-c4ad159efa5fc40f610fa0cb11e25fda19f131ad/src'
/bin/sh ../libtool  --tag=CC   --mode=link x86_64-openwrt-linux-uclibc-gcc -Wall -fPIC -Wno-format-y2k -W -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wwrite-strings -Wswitch -Wcast-align -Wchar-subscripts -Winline -Wnested-externs -Wredundant-decls -Wmissing-field-initializers -Wextra -Wformat=2 -Wno-format-nonliteral -Wpointer-arith -Wno-missing-braces -Wno-unused-parameter -Wno-unused-variable -Wno-inline -std=gnu99 -fstack-protector-all -Wall -Werror -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result   -L/home/openwrt/openwrt/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/lib -L/home/openwrt/openwrt/staging_dir/target-x86_64_uClibc-0.9.33.2/lib -L/home/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/lib  -o orangerpcd orangerpcd-main.o -lutype -lblobpack -lorange -lrt
OpenWrt-libtool: link: x86_64-openwrt-linux-uclibc-gcc -Wall -fPIC -Wno-format-y2k -W -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wwrite-strings -Wswitch -Wcast-align -Wchar-subscripts -Winline -Wnested-externs -Wredundant-decls -Wmissing-field-initializers -Wextra -Wformat=2 -Wno-format-nonliteral -Wpointer-arith -Wno-missing-braces -Wno-unused-parameter -Wno-unused-variable -Wno-inline -std=gnu99 -fstack-protector-all -Wall -Werror -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -o .libs/orangerpcd orangerpcd-main.o  -L/home/openwrt/openwrt/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/lib -L/home/openwrt/openwrt/staging_dir/target-x86_64_uClibc-0.9.33.2/lib -L/home/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/lib /home/openwrt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/orange-rpcd-2.16.08-c4ad159efa5fc40f610fa0cb11e25fda19f131ad/src/.libs/liborange.so -lblobpack -lutype -lpthread -lwebsockets -lcrypt -llua -luci -lrt
/home/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/lib/gcc/x86_64-openwrt-linux-uclibc/4.8.3/../../../../x86_64-openwrt-linux-uclibc/bin/ld: cannot find -lssp_nonshared
/home/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/lib/gcc/x86_64-openwrt-linux-uclibc/4.8.3/../../../../x86_64-openwrt-linux-uclibc/bin/ld: cannot find -lssp
/home/openwrt/openwrt/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/lib/libuci.so: warning: the use of `mktemp' is dangerous, better use `mkstemp'
collect2: error: ld returned 1 exit status
Makefile:484: recipe for target 'orangerpcd' failed
make[5]: *** [orangerpcd] Error 1
make[5]: Leaving directory '/home/openwrt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/orange-rpcd-2.16.08-c4ad159efa5fc40f610fa0cb11e25fda19f131ad/src'
Makefile:391: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/openwrt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/orange-rpcd-2.16.08-c4ad159efa5fc40f610fa0cb11e25fda19f131ad'
Makefile:65: recipe for target '/home/openwrt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/orange-rpcd-2.16.08-c4ad159efa5fc40f610fa0cb11e25fda19f131ad/.built' failed
make[3]: *** [/home/openwrt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/orange-rpcd-2.16.08-c4ad159efa5fc40f610fa0cb11e25fda19f131ad/.built] Error 2
make[3]: Leaving directory '/home/openwrt/openwrt/feeds/juci/orange-rpcd'
package/Makefile:191: recipe for target 'package/feeds/juci/orange-rpcd/compile' failed
make[2]: *** [package/feeds/juci/orange-rpcd/compile] Error 2
make[2]: Leaving directory '/home/openwrt/openwrt'
package/Makefile:188: recipe for target '/home/openwrt/openwrt/staging_dir/target-x86_64_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/home/openwrt/openwrt/staging_dir/target-x86_64_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/openwrt'
/home/openwrt/openwrt/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2
openwrt@ubuntu:~/openwrt$

mkschreder, 2016-09-06:
Right so no stack smashing protector on your gcc. That was disappointing. I
guess I will need to remove that flag.

Btw, check if there is an option in your .config called SSP_SUPPORT and see
if enabling it fixes your build.

cfcnightmare, 2016-09-06:
opened new topic? lets go there...
Its too much comments here already :)

cfcnightmare, 2016-09-06:
another error...
moved to juci-openwrt-feed issues now