mkschreder/juci #57: compile error: libblobpack-1.0.1/src/*.h': No such file or directory

shufenzhang, 2016-08-31:
Hi Martin,
Just updated my source, mainly juci package, but got the following compile error:

Could you take a look and how to fix it?

Thanks,
Shufen
-------
cp: cannot stat '/home/szhang/Projects/Adaptrum/Endor/cc_15_04/ChaosCalmer15.04/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/libblobpack-1.0.1/src/*.h': No such file or directory
make[3]: *** [/home/szhang/Projects/Adaptrum/Endor/cc_15_04/ChaosCalmer15.04/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/stamp/.libblobpack_installed] Error 1
make[3]: Leaving directory `/home/szhang/Projects/Adaptrum/Endor/cc_15_04/ChaosCalmer15.04/feeds/juci/libblobpack'
make[2]: *** [package/feeds/juci/libblobpack/compile] Error 2

mkschreder, 2016-08-31:
I think I just forgot to update the version. It should actually use commit
has when building, but it has not been doing that so you in fact have old
files there which have not been deleted even though you have updated the
feed. So just delete that build_dir/target../libblobpack-1.0.1/ folder,
rebuild and it should work.

mkschreder, 2016-08-31:
I have fixed the package makefile to always use different folder when
commit hash changes from now on.

shufenzhang, 2016-08-31:
Hi Martin,

Thanks for your reply, I built successfully after deleting dir and update Juci. But now, what is the correct login/password? I tried orange/orange, it didn't work. What exactly I need to do to be able to login? Could you please help? Step by step instructions would be mostly preferred!! :-) 

Appreciated!
Shufen

mkschreder, 2016-08-31:
orangectl adduser admin
orangectl passwd admin admin

shufenzhang, 2016-08-31:
Hi Martin, thanks for your quick reply.
Did configure admin passwd failed? 
-----------------------------
root@OpenWrt:/# orangectl passwd admin admin
No sha1sum utility present on this system!
root@OpenWrt:/# 
----------------
What do I need to add or enable? 

Thanks, really eager to login to this JuCi..
Shufen

cfcnightmare, 2016-08-31:
because you need to select sha1sum in busybox in menuconfig!

mkschreder, 2016-08-31:
It is in the new build instructions. It is CONFIG_BUSYBOX_CONFIG_SHA1SUM=y

shufenzhang, 2016-08-31:
Hi Martin,

I will do it and let you know the result.

Thanks,
Shufen

mkschreder, 2016-08-31:
Actually with sha1sum installed juci-full-openwrt default package should
automatically add admin/admin default login for you on first boot.

shufenzhang, 2016-08-31:
Hi Martin,
I recompiled image, cmd orangect works. and admin was indeed created initially i guess I got "User Already exists!" message. But I still failed to login.
I got 2 lines of error messags on the login screen:
1. You do not have an RPC connection to the router!
2. Could not connect to 10.20.20.101!

I must have still missed something??

Thanks,
Shufen

mkschreder, 2016-09-01:
Default image ip is 192.168.1.1. You seem to be connecting to 10.20.20.101.
Could that be the issue?

shufenzhang, 2016-09-01:
Hi Martin,

No, I don’t think that is the reason. I changed to use dhcp and that’s how it got that ip address. Ping worked both directions.

Ok, just tried using 192.168.1.1, the same behavior as before, saying “You do not have an RPC connection to the router” and “Could not connect to 192.168.1.1”.
What else can I do?

Thanks,
shufen

mkschreder, 2016-09-01:
Get rpc server client from here (script in the repo named "orange"):
https://github.com/mkschreder/orangerpcd

Then do this:
./orange --host 192.168.1.1 --path "/websocket/" --username admin
--password admin list "*" '{}'.

If you do not see list dump of available rpc objects then for some reason
the rpc server is not running (either not installed or something else).

cfcnightmare, 2016-09-01:
root@OpenWrt:/etc# ./orange.sh --host 192.168.1.1 --path "/websocket/" --username admin --password admin list "*" '{}'.
env: can't execute 'node': No such file or directory
root@OpenWrt:/etc# ./orange.sh
env: can't execute 'node': No such file or directory
root@OpenWrt:/etc#

shufenzhang, 2016-09-02:
Hi Martin,

Got rpc server client from:
https://github.com/mkschreder/orangerpcd

I did see the addition of liborange from make menuconfig.

But running image after built, and no change from GUI. Same as of seeing both

"You do not have an RPC connection to the router!" and "Could not connect to 192.168.1.1!"

From console command error:

root@OpenWrt:/# ./orange --host 192.168.1.1 --path "/websocket/" --username admi
n --password admin list "*" '{}'.
/bin/ash: ./orange: not found
root@OpenWrt:/#

Thanks,

Shufen

https://github.com/mkschreder/orangerpcd

mkschreder, 2016-09-02:
> auth/AERqiWWhZATUYuVMKwG-

client script is meant to be run from that folder of that repo along with
supplied node_modules. Obviously you will have to install nodejs too on
your pc.

shufenzhang, 2016-09-02:
Hi Martin,

Thanks for your reply. So we will need to install node.js on the host(build) machine or on the target board?
Could you give some detailed instruction as how to install this node.js and what else other requirements?

Thanks very much,
Shufen

<mailto:notifications@github.com%3c%0b>> > mailto:notifications@github.com>>

mkschreder, 2016-09-02:
> auth/APfja3kXcLxoHbiyIN7Axo-

host.

You can use rpc client to connect from host to box and debug any rpc
issues. So you only need nodejs on host.

shufenzhang, 2016-09-02:
Hi Martin,

That’s too generic as I am pretty new in these stuff and would like to follow your step by step instruction to make and see it work.

Thanks,
Shufen

<mailto:notifications@github.com%3cmailto:%0b>> notifications@github.com<mailto:notifications@github.com>>>
<mailto:juci@noreply.github.com%3cmailto:%0b>> juci@noreply.github.com<mailto:juci@noreply.github.com>>>

closed 2021-04-20 by mkschreder