mkschreder/juci #61: CAN NOT LOGIN THE WEB

adersonleo, 2016-09-18:
I think i do everything right with compile,  and I can see the login page, I also use orangectl adduser and orangectl passwd user password the set the password, but I still can not login the WEB, I compile after your step.
I really like this project,
thank you!

adersonleo, 2016-09-18:
It's always show "Please enter correct username and password!"

mkschreder, 2016-09-18:
maybe you don't have sha1sum cli utility? Check uml build script in the
juci-feed-openwrt repo for required steps to select packages for a basic
build.

adersonleo, 2016-09-19:
@mkschreder 
Thank you for your reply!
I do all the step again, and lucky, I am already login, But I still have one problem , I can login with Mircosoft Eage Brower,When I try to login with Chrome. It's still show  Please enter correct username and password!, I login with the same password in Eage
Thank you!

Pillar1989, 2016-09-24:
Hi mkschreder
I have installed all the software package which you pointed, but i still cannt login the web. Must I add admin user ? I cannt use default root account.

Pillar1989, 2016-09-24:
Thanks other issue, I'm login, I need do two commands.
orangectl adduser admin
orangectl passwd admin admin

Where can we find detail deploy documents ?

mkschreder, 2016-09-24:
Check the user mode linux build script. It explains all the steps you need
to take when starting with a fresh openwrt repo to get all the packages
installed correctly.

closed 2021-04-20 by mkschreder