mkschreder/juci #63: How to use rpc call ubus interface?

Pillar1989, 2016-09-28:
It seem JUCI only allow call lua api now. How to use rpc call ubus interface? eg:
root@mylinkit:/usr/lib/orange/api/juci# ubus list
dhcp
iwinfo
log
network
network.device
network.interface
network.interface.lan
network.interface.loopback
network.interface.wan
network.wireless
rpc-sys
service
session
system
uci
root@mylinkit:/usr/lib/orange/api/juci# ubus list -v iwinfo
'iwinfo' @b488fe88
        "devices":{}
        "info":{"device":"String"}
        "scan":{"device":"String"}
        "assoclist":{"device":"String","mac":"String"}
        "freqlist":{"device":"String"}
        "txpowerlist":{"device":"String"}
        "countrylist":{"device":"String"}
        "phyname":{"section":"String"}
root@mylinkit:/usr/lib/orange/api/juci# ubus call iwinfo scan '{"device":"ra0"}'

mkschreder, 2016-09-28:
Juci allows you to export ubus calls. It is used for things like uci. 

Example here: https://github.com/mkschreder/orangerpcd/blob/master/plugins/system.lua

Basically create a lua object that then exports a wraper for the ubus objects. Then you can do rpc access control for these the same way as for the other lua objects. It works pretty well.

Pillar1989, 2016-09-29:
I opened your send console.log. But I still cannt do your example.
[image]

Pillar1989, 2016-09-29:
Wow, I get a bad feeling. I need to do :
```shell
/etc/init.d/orange restart
/etc/init.d/rpcd restart
```
then I'll get the right result.
why ??

Pillar1989, 2016-10-05:
Hello again.
I had added a lua script.
```lua
root@mylinkit:/usr/lib/orange/api# cat wireless.lua 
#!/usr/bin/lua

-- JUCI Lua Backend Server API
-- Copyright (c) 2016 Martin Schröder <mkschreder.uk@gmail.com>. All rights reserved. 
-- This module is distributed under GNU GPLv3 with additional permission for signed images.
-- See LICENSE file for more details. 

local ubus = require("orange/ubus"); 

return ubus.bind("iwinfo", { "scan", "info" });  
```
But when I call it. 

$rpc.wireless.scan({"device":"ra0"})
50-juci-jquery-console.js:3 websocket > {"jsonrpc":"2.0","id":25,"method":"call","params":["0e3447488782c7b175ad39ac98516bb8","/wireless","scan",{"device":"ra0"}]}

Object {}
50-juci-jquery-console.js:3 {"jsonrpc":"2.0","id":25,"error":{"code":1,"str":"Permission denied"}}

So How add admin authority to my lua script.

mkschreder, 2016-10-05:
You need to define permissions in an acl

rpc /iwinfo scan x
rpc /iwinfo info x

and then add that filename (or wildcard) to the user in etc/config/orange.
Then restart the server and you should be able to access your functions.

closed 2021-04-20 by mkschreder