mkschreder/juci #64: Installation on Mofi4500

RVN-BR, 2016-10-10:
Dear All,

I'm a OpenWRT newbie (or even less...lol)... My experience with OpwenWRT or other router firmwares is more limited to some other platforms, but I recently was assigned a job to test a mofi4500 router for an application we are deploying.

I understand it is based on a openwrt base and it uses a (ugly and hard to use btw) LuCI interface...  My objective is to get a bit more familiarized with it, to the point where I will add some sort of HTTP API to control the router/modem...  I dont really care much for the interface, but the Juci project could help at least get a better understanding of working with it as well as being a stepping stone to using something external to poll/control the router/modem (it will be used mainly as a celular gateway.  the one I have has a builtin modem).

I'm a bit unsure about how to install juci, and if its even ompatible...  The "manual" has a getting started on openwrt page which unfortunately is returning a 404, so I guess its either not finished or just a broken link...

I see the main instructions on the Github page for a "user-mode-linux juci test build" is this what I should be following?  I assume I can just SSH into the device to execute the commands, or add a package if that exists in a "prepackaged" format, but I'm admittedly lost...

The unit is running mofi's firmware v2.0-std, and says Kernel Version 3.10.14.

Here is part of the system log, shows memory, cpu, and some other things... The router also has an option to flash a "non-mofi image", but I'm not sure if that would be a wise path or not at this point...

```
Fri Sep 30 13:55:38 2016 kern.notice kernel: [    0.000000] Linux version 3.10.14 (mofi@mofi-build) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 unknown) ) #3 Fri Sep 30 10:55:27 PDT 2016
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] Analog PMU set to hw control
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] Digital PMU set to hw control
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] SoC Type: MT7620
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000] 
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000]  The CPU feqenuce set to 580 MHz
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000]  PCIE: bypass PCIe DLL.
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000]  PCIE: Elastic buffer control: Addr:0x68 -> 0xB4
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000]  disable all power about PCIe
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] Software DMA cache coherency
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] Determined physical RAM map:
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000]  memory: 10000000 @ 00000000 (usable)
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000] Zone ranges:
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000]   Normal   [mem 0x00000000-0x0fffffff]
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000] Movable zone start for each node
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000] Early memory node ranges
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000]   node   0: [mem 0x00000000-0x0fffffff]
Fri Sep 30 13:55:38 2016 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Fri Sep 30 13:55:38 2016 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 8036f910, node_mem_map 81000000
Fri Sep 30 13:55:38 2016 kern.debug kernel: [    0.000000]   Normal zone: 512 pages used for memmap
Fri Sep 30 13:55:38 2016 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Fri Sep 30 13:55:38 2016 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, 4-way, VIPT, linesize 32 bytes.
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Sep 30 13:55:38 2016 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Fri Sep 30 13:55:38 2016 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
Fri Sep 30 13:55:38 2016 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS1,115200n8 root=/dev/mtdblock5 rootfstype=squashfs,jffs2
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] Writing ErrCtl register=0006d5f3
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] Readback ErrCtl register=0006d5f3
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] Memory: 255880k/262144k available (2812k kernel code, 6264k reserved, 707k data, 192k init, 0k highmem)
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] NR_IRQS:128
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.000000] console [ttyS1] enabled, bootconsole disabled
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.004000] Calibrating delay loop... 385.02 BogoMIPS (lpj=770048)
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.036000] pid_max: default: 32768 minimum: 301
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.040000] Mount-cache hash table entries: 512
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.044000] pinctrl core: initialized pinctrl subsystem
Fri Sep 30 13:55:38 2016 kern.info kernel: [    0.048000] NET: Registered protocol family 16
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.052000] RALINK_GPIOMODE = 1a311d
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.056000] RALINK_GPIOMODE = 18311d
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.056000] PPLL_CFG1=0xe64000
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.060000] MT7620 PPLL lock
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.060000] PPLL_DRV =0x80080504
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.064000] start PCIe register access
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.564000] RALINK_RSTCTRL = 2400000
Fri Sep 30 13:55:38 2016 kern.warn kernel: [    0.568000] RALINK_CLKCFG1 = 75afffc0
```

mkschreder, 2016-10-10:
I have not been updating the manual as often as I should be. Actually it's
good that you mention it because both the readme and the other docs need a
major cleanup.

JUCI pretty much replaces LUCI as a frontend to uci configuration databse
used by for example OpenWRT. JUCI however was never designed to replace
luci on OpenWRT. I created JUCI project as an alternative for my customers
for whom I do various other OpenWRT related work.

This has led to the documentation being a little lacking. So I put together
the UML build guide as sortof the latest tested quick example of how to add
JUCI to a vanilla OpenWRT build. There are a few important steps such as
using latest version of libwebsockets from the juci feed in order to build
orangerpcd and using an extended version of lighttpd (also in the juci
feed) for websocket support. All these steps are included in the uml
script. Basically from that script you can add the necessary stepts to your
own openwrt bootstrap script and then continue from there. Even if you need
to do a little bootstrapping to set up menu, theme etc, for the most part I
don't think you will have trouble using JUCI on any platform that is
supported by openwrt. There is a minor crash issue when you build
orangerpcd with musl libc instead of uclibc but there is a patch for that
mentioned in another ticket so if you have that issue let me know and I'll
dig up that patch for you.

RVN-BR, 2016-10-10:
Thanks for clearing that up @mkschreder !

I actually just realized I'm much further behind than I should lol...  I dont have a openwrt directory anywhere obvious on the router (checked home directory, as well as a few other directories I would imagine it would be)...  I guess I'll have to play around a lot more with the firmware as is in order to grasp a bit more of the concepts...  I'm fine working with linux etc, but havent really done much in WRT or other custom firmwares except for some stuff with Ubiquiti equipment...

Even the current packages seem a bit out of date, and the OPKG sources are all no longer available so its going to need a bit of stepping back and digging before I can start doing stuff... its a pitty Mofi doesnt povide any sort of documentation and there appears to be very few users, or at least no useful forum that I have come across so far...  Only things are good comments on hardware capability but always harsh remarks about how difficult it is to get setup due to its lacky gui...

I'll see what I learn and if I do end up having success I'll post my findings here....

closed 2021-04-20 by mkschreder