mkschreder/juci #65: compile Juci on my ubuntu PC

Pillar1989, 2016-10-12:
Hi,
I want to compile Juci on my ubuntu PC.
1, cp example-Makefile.local  Makefile.local
2,make debug

But I get fallowing error. I think I missed some configure selection. Do you know how to fix it ??
```shell
# we use existing node modules to be able to compile offline
tar -xzf node_modules.tar.gz
#npm install --production
-e [JS]	juci -> tmp/01-juci.js
/home/pillar/MTK/juci/juci/
-e [LESS]	juci -> tmp/juci-compiled-styles.css
-e [CSS]	juci -> tmp/01-juci.css
make: *** [tmp/01-juci.css.js] Error 1
```

mkschreder, 2016-10-12:
What are you going to use it for on ubuntu? If you just want to develop
then consider using juci-local-server.

Pillar1989, 2016-10-12:
Yes, I want to develop it. If we run the juci-local-server ,we can see it also call "make debug DEFAULT_THEME=y" to compile the source code. end also print this error.

mkschreder, 2016-10-12:
maybe you are missing lessc (the less -> css compiler) ?

Pillar1989, 2016-10-12:
I have it. 
```shell
pillar@server:~/MTK/juci$ which lessc 
/usr/local/bin/lessc
pillar@server:~/MTK/juci$ 
pillar@server:~/MTK/juci$ 
pillar@server:~/MTK/juci$ cat /usr/local/bin/lessc
#!/usr/bin/env node

var path = require('path'),
    fs = require('../lib/less-node/fs'),
    os = require("os"),
    errno,
    mkdirp;

try {
    errno = require('errno');
} catch (err) {
    errno = null;
}
...........
```
I also strace openwrt build process. I can see it build po/template.pot first.
```shell
# we use existing node modules to be able to compile offline
tar -xzf node_modules.tar.gz
#npm install --production
-e [POT]	juci -> /home/pillar/MTK/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/juci-d35aa0df0ceb77266678be37e0e1bbf77d4b8691/juci//po/template.pot
-e [JS]	juci -> tmp/01-juci.js
-e [LESS]	juci -> tmp/juci-compiled-styles.css
-e [CSS]	juci -> tmp/01-juci.css
-e [HTML]	juci -> tmp/01-juci.tpl.js
cat tmp/01-juci.js tmp/01-juci.css.js tmp/01-juci.tpl.js > bin/www/js/01-juci.js

```

mkschreder, 2016-10-12:
Have you checked scripts/ubuntu-bootstrap.sh?

mkschreder, 2016-10-12:
We are not using grunt any more though and not bower either. I need to
update that script.

Do you have msguniq? It is part of gettext package.

Pillar1989, 2016-10-12:
Yes ,I have msguniq, But when I run scripts/ubuntu-bootstrap.sh, I got error.
```shell
0 upgraded, 0 newly installed, 0 to remove and 506 not upgraded.
npm WARN package.json juci@0.0.1 No repository field.
npm WARN prefer global coffee-script@1.10.0 should be installed with -g
npm ERR! peerinvalid The package grunt does not satisfy its siblings' peerDependencies requirements!
npm ERR! peerinvalid Peer grunt-angular-gettext@2.1.11 wants grunt@~0.4.1

npm ERR! System Linux 3.16.0-45-generic
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /home/pillar/MTK/juci
npm ERR! node -v v0.10.25
npm ERR! npm -v 1.3.10
npm ERR! code EPEERINVALID
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pillar/MTK/juci/npm-debug.log
npm ERR! not ok code 0
^Cnpm http GET https://registry.npmjs.org/mocha
^C^Cnpm http GET https://registry.npmjs.org/uglify-js
^C^Cpillar@server:~/MTK/juci$ ^C
pillar@server:~/MTK/juci$ which msguniq
/usr/bin/msguniq

```

mkschreder, 2016-10-12:
> npm ERR! peerinvalid The package grunt does not satisfy its siblings' peerDependencies requirements!npm ERR! peerinvalid Peer grunt-angular-gettext@2.1.11 wants grunt@~0.4.1npm ERR! System Linux 3.16.0-45-genericnpm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"npm ERR! cwd /home/pillar/MTK/jucinpm ERR! node -v v0.10.25npm ERR! npm -v 1.3.10npm ERR! code EPEERINVALIDnpm ERR! npm ERR! Additional logging details can be found in:npm ERR!     /home/pillar/MTK/juci/npm-debug.lognpm ERR! not ok code 0^Cnpm http GET https://registry.npmjs.org/mocha^C^Cnpm http GET https://registry.npmjs.org/uglify-js^C^Cpillar@server:~/MTK/juci$ ^Cpillar@server:~/MTK/juci$ which msguniq/usr/bin/msguniq

Wow that was an old nodejs version. I have 4.3.2. You definitely need to
update these. Although I don't know if that is causing your problem.

Can you run "make SHELL="sh -x" debug" and check exact command that
actually fails?

Pillar1989, 2016-10-13:
When I update my ubuntu to 16.04, I can use juci-local-server . You are right. because of my nodejs version is too low.

closed 2016-10-13 by Pillar1989