mkschreder/juci #66: how to get the custom lighttpd

jaapjolman, 2016-10-15:
Dear mkschreder,

i am trying to build juci but i cannot for the life of me find out how to get the custom version of lighttpd in the code, it keeps picking the default one

Pillar1989, 2016-10-15:
https://github.com/mkschreder/juci-lighttpd

mkschreder, 2016-10-15:
./scripts/feeds install -f -a -p juci

uninstall all feeds first then install the other again *after* this step.

jaapjolman, 2016-10-15:
i just noticed that your version of lighttpd is 1.4.33 and the one of openwrt is 1.4.41 may this cause an override on the ./scripts/feeds install -f -a -p juci

because if i do 
./scripts/feeds install -f -a -p juci
it shows your package in make menuconfig but if i do ./scripts/feeds install -a after your command it will show the default openwrt package.

mkschreder, 2016-10-15:
CC or DD?

The behaviour is incorrect then. Already installed packages should never be
overridden without -f. Perhaps you need to reverse the order of the ops if
the reverse behaviour works for you? What if you change version and set it
to a number that is larger than openwrt ver?

closed 2021-04-20 by mkschreder