mkschreder/juci #67: Compiling Juci on Ubuntu to be used on OpenWRT router

exander77, 2016-12-12:
I seem to have problem understanding how to build Juci on Ubuntu so I can use on my OpenWRT router as LuCI replacement.

Could you point me to the right direction?

mkschreder, 2016-12-13:
You should build as part of openwrt. Check this script to see how to add
the necessary feed and select required packages:
https://github.com/mkschreder/juci-openwrt-feed/blob/master/openwrt-bootstrap.sh

There is also a uml (user mode linux) build script in the feed repo above
that will build openwrt as a native executable and also build a rootfs
image that you can pass to it and run openwrt as a linux process (uml image
is rather limited though)

exander77, 2016-12-15:
There is no way to build it separately and install it from packages? It has some platform dependencies?

exander77, 2016-12-15:
I tried to build juci repository and it went fine, it created bin directory with etc, sbin, usr, www, can I make a package and install this on my OpenWRT router?

mkschreder, 2016-12-15:
You can of course. You just need to figure out the peculiarities of your
particular system yourself.

closed 2021-04-20 by mkschreder