mkschreder/juci #71: I visit the home page and can not see anything. The homepage is blank.Also the console does not have any errors

anoopantony8, 2017-01-13:
I followed these steps during the setup of JUCI-openwrt

mkschreder, 2017-01-13:
Use openwrt-bootstrap script in the juci-openwrt-feed repo as reference.

anoopantony8, 2017-01-16:
@mkschreder I have used that script

djhamzas, 2017-02-14:
I have a same problem can any one help me , when i cheked with fireburg i found  the flowing errors ;

"NetworkError: 404 Not Found - http://192.168.1.168/websocket/"
Firefox ne peut établir de connexion avec le serveur à l’adresse ws://192.168.1.168/websocket/.
	

01-juci.js (ligne 56, col. 1718)
"NetworkError: 404 Not Found - http://192.168.1.168/fonts/fontawesome-webfont.ttf?v=4.3.0"

Help me please

mkschreder, 2017-02-15:
> fontawesome-webfont.ttf?v=4.3.0"

If websocket url is not found it means you are using wrong lighttpd. Use
the one in the juci feed repo and make sure that default openwrt one is
overridden properly.

closed 2021-04-20 by mkschreder