mkschreder/juci #72: Compile error juci on lede project

perry-amsterdam, 2017-01-15:
Hi,

I tried to compile juci on the latest lede project a couple of weeks ago it worked.
But now i got the following compile error:

Greeting from Amsterdam, Perry

juci/gl-inet-6416a-v1/openwrt-src/build_dir/target-mips_24kc_musl-1.1.15/libblobpack-2.16.09-a024c6e3a2512741d94e3a48d72e734de7f73ff3:libblobpack-2.16.09-a024c6e3a2512741d94e3a48d72e734de7f73ff3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT libblobpack_la-blob.lo -MD -MP -MF .deps/libblobpack_la-blob.Tpo -c blob.c  -fPIC -DPIC -o .libs/libblobpack_la-blob.o
blob.c: In function 'blob_field_init':
blob.c:42:17: error: implicit declaration of function 'htobe32' [-Werror=implicit-function-declaration]
  attr->id_len = htobe32(len);
                 ^
blob.c:42:2: error: nested extern declaration of 'htobe32' [-Werror=nested-externs]
  attr->id_len = htobe32(len);
  ^
blob.c: In function 'blob_put_u16':
blob.c:162:8: error: implicit declaration of function 'htobe16' [-Werror=implicit-function-declaration]
  val = htobe16(val);
        ^
blob.c:162:2: error: nested extern declaration of 'htobe16' [-Werror=nested-externs]
  val = htobe16(val);
  ^
blob.c: In function 'blob_put_u64':
blob.c:172:8: error: implicit declaration of function 'htobe64' [-Werror=implicit-function-declaration]
  val = htobe64(val);
        ^
blob.c:172:2: error: nested extern declaration of 'htobe64' [-Werror=nested-externs]
  val = htobe64(val);
  ^
cc1: all warnings being treated as errors
Makefile:442: recipe for target 'libblobpack_la-blob.lo' failed
make[5]: *** [libblobpack_la-blob.lo] Error 1

mkschreder, 2017-01-15:
These are in endian.h system header. I remember there was someone else
having the issue but not sure how they fixed it. Could it be that you have
switched libc version?

perry-amsterdam, 2017-01-21:
Hi,

I compiled the latest lede trunk version for Raspberri PI2, with the default libs.

Perry

p4u, 2017-02-18:
Got the same problem here. Also with last libblobpck rev d0fe5175995c550f6961d2675d80a96e44242475

p4u, 2017-02-18:
This fixed the problem for me https://github.com/mkschreder/libblobpack/pull/3

closed 2021-04-20 by mkschreder