mkschreder/juci #77: error libblobpack

fahdigunbound, 2017-06-01:
blob_field.c: In function 'blob_field_set_raw_len':
blob_field.c:49:19: error: implicit declaration of function 'htobe32' [-Werror=implicit-function-declaration]
  attr->id_len &= ~htobe32(BLOB_FIELD_LEN_MASK);
                   ^
blob_field.c:49:2: error: nested extern declaration of 'htobe32' [-Werror=nested-externs]
  attr->id_len &= ~htobe32(BLOB_FIELD_LEN_MASK);
  ^
blob_field.c: In function 'blob_field_get_u16':
blob_field.c:125:9: error: implicit declaration of function 'be16toh' [-Werror=implicit-function-declaration]
  return be16toh(*tmp);
         ^
blob_field.c:125:2: error: nested extern declaration of 'be16toh' [-Werror=nested-externs]
  return be16toh(*tmp);
  ^
blob_field.c: In function 'blob_field_get_u32':
blob_field.c:138:9: error: implicit declaration of function 'be32toh' [-Werror=implicit-function-declaration]
  return be32toh(*tmp);
         ^
blob_field.c:138:2: error: nested extern declaration of 'be32toh' [-Werror=nested-externs]
  return be32toh(*tmp);
  ^
cc1: all warnings being treated as errors
Makefile:449: recipe for target 'libblobpack_la-blob_field.lo' failed
make[5]: *** [libblobpack_la-blob_field.lo] Error 1
make[5]: Leaving directory '/home/fahdi/openwrt/build_dir/target-x86_64_musl-1.1.16/libblobpack-2.16.09-d0fe5175995c550f6961d2675d80a96e44242475/src'
Makefile:381: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/fahdi/openwrt/build_dir/target-x86_64_musl-1.1.16/libblobpack-2.16.09-d0fe5175995c550f6961d2675d80a96e44242475'
Makefile:45: recipe for target '/home/fahdi/openwrt/build_dir/target-x86_64_musl-1.1.16/libblobpack-2.16.09-d0fe5175995c550f6961d2675d80a96e44242475/.built' failed
make[3]: *** [/home/fahdi/openwrt/build_dir/target-x86_64_musl-1.1.16/libblobpack-2.16.09-d0fe5175995c550f6961d2675d80a96e44242475/.built] Error 2
make[3]: Leaving directory '/home/fahdi/openwrt/feeds/juci/libblobpack'
package/Makefile:196: recipe for target 'package/feeds/juci/libblobpack/compile' failed
make[2]: *** [package/feeds/juci/libblobpack/compile] Error 2
make[2]: Leaving directory '/home/fahdi/openwrt'
package/Makefile:193: recipe for target '/home/fahdi/openwrt/staging_dir/target-x86_64_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/fahdi/openwrt/staging_dir/target-x86_64_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/fahdi/openwrt'
/home/fahdi/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2

i tired always error like that.. 
@mkschreder Please help me

seaving, 2017-06-19:
I know that.

seaving, 2017-06-19:
Hello, I have also encountered this situation, Have you solved it?

dman776, 2018-05-09:
Me too.   Any help?

mcellos, 2018-11-26:
1 have one problem with package too

flavio@Flavio-Note:~/openwrt/build_dir/target-i386_pentium4_musl/libblobpack-9cc36022$ make
Making all in src
make[1]: Entering directory '/home/flavio/openwrt/build_dir/target-i386_pentium4_musl/libblobpack-9cc36022/src'
/bin/bash ../libtool  --tag=CC   --mode=compile gcc -DPACKAGE_NAME=\"libblobpack.so\" -DPACKAGE_TARNAME=\"libblobpack-so\" -DPACKAGE_VERSION=\"0.1.0\" -DPACKAGE_STRING=\"libblobpack.so\ 0.1.0\" -DPACKAGE_BUGREPORT=\"mkschreder.uk@gmail.com\" -DPACKAGE_URL=\"\" -DPACKAGE=\"libblobpack-so\" -DVERSION=\"0.1.0\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DHAVE_UNISTD_H=1 -I.     -std=gnu99 -Wall -Werror -Wno-format-y2k -W -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch -Wshadow -Wcast-align -Wchar-subscripts -Winline -Wnested-externs -Wredundant-decls -Wmissing-field-initializers -Wextra -Wformat=2 -Wno-format-nonliteral -Wpointer-arith -Wno-missing-braces -Wno-unused-parameter -Wno-unused-variable -Wno-inline -g -O2 -MT libblobpack_la-ujsonenc.lo -MD -MP -MF .deps/libblobpack_la-ujsonenc.Tpo -c -o libblobpack_la-ujsonenc.lo `test -f 'ujsonenc.c' || echo './'`ujsonenc.c
libtool: compile:  gcc -DPACKAGE_NAME=\"libblobpack.so\" -DPACKAGE_TARNAME=\"libblobpack-so\" -DPACKAGE_VERSION=\"0.1.0\" "-DPACKAGE_STRING=\"libblobpack.so 0.1.0\"" -DPACKAGE_BUGREPORT=\"mkschreder.uk@gmail.com\" -DPACKAGE_URL=\"\" -DPACKAGE=\"libblobpack-so\" -DVERSION=\"0.1.0\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DHAVE_UNISTD_H=1 -I. -std=gnu99 -Wall -Werror -Wno-format-y2k -W -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch -Wshadow -Wcast-align -Wchar-subscripts -Winline -Wnested-externs -Wredundant-decls -Wmissing-field-initializers -Wextra -Wformat=2 -Wno-format-nonliteral -Wpointer-arith -Wno-missing-braces -Wno-unused-parameter -Wno-unused-variable -Wno-inline -g -O2 -MT libblobpack_la-ujsonenc.lo -MD -MP -MF .deps/libblobpack_la-ujsonenc.Tpo -c ujsonenc.c  -fPIC -DPIC -o .libs/libblobpack_la-ujsonenc.o
ujsonenc.c: In function 'Buffer_EscapeStringValidated':
ujsonenc.c:408:12: error: this statement may fall through [-Werror=implicit-fallthrough=]
         if (enc->encodeHTMLChars)
            ^
ujsonenc.c:420:7: note: here
       case 30:
       ^~~~
ujsonenc.c: In function 'Buffer_EscapeStringUnvalidated':
ujsonenc.c:200:12: error: this statement may fall through [-Werror=implicit-fallthrough=]
         if (enc->encodeHTMLChars)
            ^
ujsonenc.c:211:7: note: here
       case '/':
       ^~~~
cc1: all warnings being treated as errors
Makefile:477: recipe for target 'libblobpack_la-ujsonenc.lo' failed
make[1]: *** [libblobpack_la-ujsonenc.lo] Error 1
make[1]: Leaving directory '/home/flavio/openwrt/build_dir/target-i386_pentium4_musl/libblobpack-9cc36022/src'
Makefile:381: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

closed 2021-04-20 by mkschreder