mkschreder/juci #78: UCI failed!

TadeuGRibeiro, 2017-06-29:
Hi Martin,

Long time that I don't write to you... In past I got sucess with JUCI compiling with OpenWRT. Now I'm working with another project that I used LEDE, but now I got an erro with UCI, I read all topics in your GitHub and try all things that you suggest for this and don't works.... please, can you help me?

Thanks a lot.

[image]

mkschreder, 2017-06-30:
check if RPC.uci.* methods exist (you can do so in the browser console).

TadeuGRibeiro, 2017-06-30:
I did and got this result...

[image]

TadeuGRibeiro, 2017-07-03:
Hi Martin, any idea?

Tks.

TadeuGRibeiro, 2017-07-13:
anyone?

mkschreder, 2017-07-13:
Sorry to hear you haven't solved it yet. What kind of debugging have you
done yourself so far? This type of error is fairly common when openwrt repo
has not been setup correctly from the start. Did you follow roughly the
same process as outlined in juci-openwrt-feed docs example?

https://github.com/mkschreder/juci-openwrt-feed/blob/master/build-uml-image.sh

TadeuGRibeiro, 2017-07-13:
It's OK. Now I have just Browser console, and I dont get nothing that I can use... I think that OpenWRT setup are right... but whats type of wrong setup I can do? Do you have any idea? I follow all steps of juci-openwrt-feeds, but same result... I try in other machine and same error.... I dont know how I can debug more deeply...

mkschreder, 2017-07-13:
Keep in mind that you need to run make distclean before you do the
bootstrap. The bootstrap is necessary so you get correct versions of the
packages. I have it on my todo list right now to make juci build on latest
lede. I know there are a few problems that need to get fixed first.

TadeuGRibeiro, 2017-07-14:
This is an noob question, but I have problem with this now.... if I run make distclean, before set my target and the run the file openwrt-bootstrap.sh, the make defconfig is executed and set my target to ar71xx... I dont know how I can change the files to solve this...

mkschreder, 2017-07-14:
ar71xx is default openwrt target. The example script should set the target
to user-mode-linux (UML). You should select your own target using make
menuconfig. The openwrt-bootstrap.sh is just an example that shows you the
steps you need to take to ensure that you are using all the compatible
versions of packages. You probably should not use it as your main build
script without making some changes to it.

TadeuGRibeiro, 2017-07-14:
Right, I see it now. Thks.
I tried rebuild step by step that documentation describe, but I got same problem: UCI failed to initialize!

I don't have any idea that I can do to solve this... :(

mkschreder, 2017-07-15:
I just pushed latest feed. There was no problem with the feed other than
some packages needed to be update to latest version.

Try this on latest lede:
* add juci feed to feeds.conf
* ./scripts/feeds update -a
* ./scripts/feeds uninstall -a
* ./scripts/feeds install -fap juci
* ./scripts/feeds install -a
* select juci-full-openwrt in menuconfig
* make

I just tested this and it works just fine. Make sure you clear browser
cache too (you may have some broken version cached from before).

mkschreder, 2017-07-15:
Ok! I think I see the same problem you have. It is caused because openwrt
rpcd crashes. Why I don't know yet. If you do /etc/init.d/rpcd restart you
should be able to access the ui.

TadeuGRibeiro, 2017-07-16:
I tried run /etc/init.d/rpcd restart but nothing... Then I runned command ps to check which process are running and I got the results as below.

 /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300
  765 root      4288 S    /usr/sbin/lighttpd -f /etc/lighttpd.conf
  869 root      1180 S    udhcpc -p /var/run/udhcpc-eth0.2.pid -s /lib/netifd/dhcp.script -f -t 0 -i eth0.2 -C
  889 root      1180 S <  /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.org -p 2.openwrt.poo
  932 root      1180 S    udhcpc -p /var/run/udhcpc-apcli0.pid -s /lib/netifd/dhcp.script -f -t 0 -i apcli0 -C
 1016 root      6768 S    /usr/bin/orangerpcd -p /usr/lib/orange/api/
 1032 dnsmasq   1048 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf -k -x /var/run/dnsmasq/dnsmasq.pid
 1113 root      1124 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300
 1114 root      1184 S    -ash
 1143 root      1612 S    /sbin/rpcd

RPCD and OrangeRpcd is running....

mkschreder, 2017-07-16:
So if you run "ubus list" you can see uci in the list but you still get UCI
failed to initialize in the browser console?

TadeuGRibeiro, 2017-07-17:
I run ubus list and get this results....
[image]

lfdominguez, 2017-10-04:
Is like this??? #80

closed 2021-04-20 by mkschreder