mkschreder/juci #80: Compiling JUCI in LEDE

lfdominguez, 2017-10-04:
Hello, I'm using LEDE and i want put the JUCI interface into it. I compile without problems JUCI with LEDE, i was add the feed into the build of lede and:

1. `make menuconfig`
2. Select the arch arm for qemu (for testing)
3. Select the package juci-openwrt-full
4. Select the package coreutils-sha1sum
5. Select the package lighttpd-websockets
6. `make defconfig`
7. `make`

All compile ok and generate all files, i was created a tap device for communication and execute with qemu the image:

`qemu-system-arm -machine virt -hda /STORAGE/GIT_CLONES/lede/bin/targets/armvirt/32/lede-armvirt-32-root.squashfs.gz  -kernel /STORAGE/GIT_CLONES/lede/bin/targets/armvirt/32/lede-armvirt-32-zImage-initramfs  -nographic -append "root=/dev/sda console=ttyS0" -device virtio-net-pci,netdev=lan -netdev tap,id=lan,ifname=ledetap0,script=no,downscript=no -device virtio-net-pci,netdev=wan -netdev user,id=wan -m 64`

The system boot ok, and the JUCI first login works but, nothing more, the spins are forever, and if I update the page, then the spin is loading forever...

First login:
[image]

Welcome page, the spinners never ending:
[image]
[image]

If Update (F5), the spinner never ending:
[image]

I dont found any log, only this in /var/log/error.log:
2017-10-04 14:26:39: (log.c.166) server started 
2017-10-04 14:26:39: (server.c.1032) WARNING: unknown config-key: url.access-deny (ignored) 
2017-10-04 14:27:21: (mod_websocket.c.340) /websocket/ is match WebSocket extension: /websocket/ 
2017-10-04 14:27:21: (mod_websocket_handshake.c.98) allowed origins are not specified 
2017-10-04 14:27:21: (mod_websocket.c.531) works as WebSocket Proxy 
2017-10-04 14:27:21: (mod_websocket.c.152) try to connect backend -> 127.0.0.1 : 5303 
2017-10-04 14:27:21: (mod_websocket.c.180) connected 192.168.1.101 : 58272 ( fd = 7 ) -> 127.0.0.1 : 5303 ( fd = 14 ) 
2017-10-04 14:27:21: (mod_websocket_handshake.c.477) append X-Forwarded-Proto: http 
2017-10-04 14:27:21: (mod_websocket_handshake.c.493) append X-Forwarded-For: 192.168.1.101:58272 
2017-10-04 14:27:21: (mod_websocket_handshake.c.506) append X-Forwarded-Port: 58272 
2017-10-04 14:27:21: (mod_websocket_handshake.c.545) forward handshake request 
2017-10-04 14:27:21: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x81 
2017-10-04 14:27:21: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
2017-10-04 14:27:21: (mod_websocket_frame.c.510) recv data from client ( fd = 7 ), size = 0x00 
2017-10-04 14:27:21: (mod_websocket.c.688) send data to backend ( fd = 14 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x05e0 
2017-10-04 14:27:22: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
2017-10-04 14:27:22: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x46 
2017-10-04 14:27:22: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
2017-10-04 14:27:22: (mod_websocket_frame.c.510) recv data from client ( fd = 7 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.688) send data to backend ( fd = 14 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x46 
2017-10-04 14:27:22: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
2017-10-04 14:27:22: (mod_websocket_frame.c.510) recv data from client ( fd = 7 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.688) send data to backend ( fd = 14 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x46 
2017-10-04 14:27:22: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
2017-10-04 14:27:22: (mod_websocket_frame.c.510) recv data from client ( fd = 7 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.688) send data to backend ( fd = 14 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x46 
2017-10-04 14:27:22: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
2017-10-04 14:27:22: (mod_websocket_frame.c.510) recv data from client ( fd = 7 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.688) send data to backend ( fd = 14 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x46 
2017-10-04 14:27:22: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
2017-10-04 14:27:22: (mod_websocket_frame.c.510) recv data from client ( fd = 7 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.688) send data to backend ( fd = 14 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x46 
2017-10-04 14:27:22: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
2017-10-04 14:27:22: (mod_websocket_frame.c.510) recv data from client ( fd = 7 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.688) send data to backend ( fd = 14 ), size = 0x00 
2017-10-04 14:27:22: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x011a 
2017-10-04 14:27:22: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
[... 127 lines cut ...]
2017-10-04 14:27:29: (mod_websocket_handshake.c.506) append X-Forwarded-Port: 58298 
2017-10-04 14:27:29: (mod_websocket_handshake.c.545) forward handshake request 
2017-10-04 14:27:29: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x81 
2017-10-04 14:27:29: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
2017-10-04 14:27:29: (mod_websocket_frame.c.510) recv data from client ( fd = 7 ), size = 0x00 
2017-10-04 14:27:29: (mod_websocket.c.688) send data to backend ( fd = 14 ), size = 0x00 
2017-10-04 14:27:32: (mod_websocket_frame.c.510) recv data from client ( fd = 7 ), size = 0x00 
2017-10-04 14:27:32: (mod_websocket.c.688) send data to backend ( fd = 14 ), size = 0xb6f8f2c0 
2017-10-04 14:27:32: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x04 
2017-10-04 14:27:32: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
2017-10-04 14:27:32: (mod_websocket.c.854) disconnected from client ( fd = 7 ) 
2017-10-04 14:27:32: (mod_websocket.c.192) disconnect backend ( fd = 14 ) 
2017-10-04 14:27:32: (mod_websocket.c.340) /websocket/ is match WebSocket extension: /websocket/ 
2017-10-04 14:27:32: (mod_websocket_handshake.c.98) allowed origins are not specified 
2017-10-04 14:27:32: (mod_websocket.c.531) works as WebSocket Proxy 
2017-10-04 14:27:32: (mod_websocket.c.152) try to connect backend -> 127.0.0.1 : 5303 
2017-10-04 14:27:32: (mod_websocket.c.180) connected 192.168.1.101 : 58318 ( fd = 7 ) -> 127.0.0.1 : 5303 ( fd = 14 ) 
2017-10-04 14:27:32: (mod_websocket_handshake.c.477) append X-Forwarded-Proto: http 
2017-10-04 14:27:32: (mod_websocket_handshake.c.493) append X-Forwarded-For: 192.168.1.101:58318 
2017-10-04 14:27:32: (mod_websocket_handshake.c.506) append X-Forwarded-Port: 58318 
2017-10-04 14:27:32: (mod_websocket_handshake.c.545) forward handshake request 
2017-10-04 14:27:33: (mod_websocket.c.234) recv data from backend ( fd = 14 ), size = 0x81 
2017-10-04 14:27:33: (mod_websocket.c.737) send data to client ( fd = 7 ), size = 0xb6cedcac 
2017-10-04 14:27:33: (mod_websocket_frame.c.510) recv data from client ( fd = 7 ), size = 0x00 
2017-10-04 14:27:33: (mod_websocket.c.688) send data to backend ( fd = 14 ), size = 0xb6f8f2c0

lfdominguez, 2017-10-04:
This in the console inside chromium

01-juci.js:1 Section mini_snmpd already defined. Will extend existing section! If this is not your intention, fix your code!

UCIConfig.$registerSectionType @ 01-juci.js:1
(anonymous) @ 50-juci-snmpd.js:1

01-juci.js:1 Section system already defined. Will extend existing section! If this is not your intention, fix your code!

lfdominguez, 2017-10-04:
Console screenshots

[image]
[image]

mkschreder, 2017-10-04:
I would try to figure out why requests are timing out. There is a shell
script in the juci feed that configures a fresh lede repo. You did not
outline the same steps so it could be that you have different versions of
the essential packages.

lfdominguez, 2017-10-04:
Could be.....
When i add the juci i need force that lighttpd use the juci repo and not the base, because the lede lighttpd dont have websockets supports... then what can i do? i will test that script thanks

lfdominguez, 2017-10-04:
that is the openwrt-bootstrap.sh file?

lfdominguez, 2017-10-04:
make from beginind, an error in compilation...

Package liborange is missing dependencies for the following libraries:
 For /STORAGE/GIT_CLONES/lede/build_dir/target-x86_64_musl/orange-rpcd-3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7/ipkg-x86_64/liborange From /STORAGE/GIT_CLONES/lede/staging_dir/target-x86_64_musl/pkginfo/liborange.missing
libwebsockets.so.10

lfdominguez, 2017-10-04:
added the libwebsockets in `make menuconfig`, but the same error

luiskahn718, 2017-10-07:
Hello, Martin.
I am new in JUCI install at Openwrt. Platform is Ralink RTxx.
I got same build error.

...
make[3]: Entering directory '/home/kcy/openwrt/trunk/feeds/juci/orange-rpcd'
mkdir -p /home/kcy/openwrt/trunk/bin/ramips/packages /home/kcy/openwrt/trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.16/orange-rpcd-3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7/ipkg-ramips/liborange/CONTROL /home/kcy/openwrt/trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.16/pkginfo
install -d -m0755 /home/kcy/openwrt/trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.16/orange-rpcd-3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7/ipkg-ramips/liborange/usr/lib/
cp -fpR /home/kcy/openwrt/trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.16/orange-rpcd-3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7/src/.libs/*.so* /home/kcy/openwrt/trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.16/orange-rpcd-3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7/ipkg-ramips/liborange/usr/lib/
find /home/kcy/openwrt/trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.16/orange-rpcd-3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7/ipkg-ramips/liborange -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package liborange is missing dependencies for the following libraries:
libwebsockets.so.10
Makefile:78: recipe for target '/home/kcy/openwrt/trunk/bin/ramips/packages/juci/liborange_3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7_ramips.ipk' failed
....

I did follow as your guide, but what's wrong?
Please help.

techspace01, 2017-12-21:
@lfdominguez

I have issue with LEDE and JUCI missing dependencies "lighttpd-mod-websocket". I'm curious where did you select package "lighttpd-websockets" in make menuconfig?

I have look at Network > Web Servers/Proxies > lighttpd but did not see lighttpd-mod-websocket

mkschreder, 2017-12-22:
It will only appear if your lighttpd package has been correctly overridden
by the version from the juci feed. So install juci feed first then install
all other openwrt packages. That way you will get a different version of
lighttpd.

closed 2021-04-20 by mkschreder