mkschreder/juci #81: LEDE Compile Deprecated error & missing library

dbrody2004, 2017-12-12:
Compiling liborange on LEDE 17.01 gives the following error. The build is x86 using glibc.

OpenWrt-libtool: compile: x86_64-openwrt-linux-gnu-gcc -DPACKAGE_NAME="orangerpcd" -DPACKAGE_TARNAME="orangerpcd" -DPACKAGE_VERSION="0.1.0" "-DPACKAGE_STRING="orangerpcd 0.1.0"" -DPACKAGE_BUGREPORT="mkschreder.uk@gmail.com" -DPACKAGE_URL="" -DPACKAGE="orangerpcd" -DVERSION="0.1.0" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=".libs/" -DHAVE_LUA_H=1 -DHAVE_UCI_H=1 -I. -I/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/target-x86_64_glibc/usr/include -I/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/target-x86_64_glibc/include -I/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/toolchain-x86_64_gcc-5.4.0_glibc/usr/include -I/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/toolchain-x86_64_gcc-5.4.0_glibc/include -DCONFIG_THREADS -D_POSIX_C_SOURCE=201609L -D_BSD_SOURCE -D_XOPEN_SOURCE -D_XOPEN_SOUCE_EXTENDED -D_GNU_SOURCE -Wall -Werror -Wno-format-y2k -W -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wwrite-strings -Wswitch -Wno-cast-align -Wchar-subscripts -Winline -Wtype-limits -Wnested-externs -Wredundant-decls -Wmissing-field-initializers -Wextra -Wformat=2 -Wno-format-nonliteral -Wpointer-arith -Wno-missing-braces -Wno-unused-parameter -Wno-unused-variable -Wno-inline -std=gnu99 -Wall -Werror -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/build_dir/target-x86_64_glibc/orange-rpcd-3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7:orange-rpcd-3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ggdb3 -MT liborange_la-base64.lo -MD -MP -MF .deps/liborange_la-base64.Tpo -c base64.c -fPIC -DPIC -o .libs/liborange_la-base64.o
In file included from /home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/toolchain-x86_64_gcc-5.4.0_glibc/include/bits/libc-header-start.h:33:0,
from /home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/toolchain-x86_64_gcc-5.4.0_glibc/include/string.h:26,
from base64.c:9:
/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/toolchain-x86_64_gcc-5.4.0_glibc/include/features.h:180:3: error: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Werror=cpp]

warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"

^
cc1: all warnings being treated as errors
Makefile:559: recipe for target 'liborange_la-base64.lo' failed

--------

After removing Werror from the makefile.am, it has link error -

OpenWrt-libtool: link: x86_64-openwrt-linux-gnu-gcc -DCONFIG_THREADS -D_POSIX_C_SOURCE=201609L -D_BSD_SOURCE -D_XOPEN_SOURCE -D_XOPEN_SOUCE_EXTENDED -D_GNU_SOURCE -Wall -Wno-format-y2k -W -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wwrite-strings -Wswitch -Wno-cast-align -Wchar-subscripts -Winline -Wtype-limits -Wnested-externs -Wredundant-decls -Wmissing-field-initializers -Wextra -Wformat=2 -Wno-format-nonliteral -Wpointer-arith -Wno-missing-braces -Wno-unused-parameter -Wno-unused-variable -Wno-inline -std=gnu99 -Wall -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/build_dir/target-x86_64_glibc/orange-rpcd-3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7:orange-rpcd-3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -ggdb3 -znow -zrelro -o .libs/orangerpcd orangerpcd-main.o -L/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/target-x86_64_glibc/usr/lib -L/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/target-x86_64_glibc/lib -L/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/toolchain-x86_64_gcc-5.4.0_glibc/usr/lib -L/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/toolchain-x86_64_gcc-5.4.0_glibc/lib /home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/build_dir/target-x86_64_glibc/orange-rpcd-3.16.01-2e02cb0594b35e534c1e155d94d95ecf062d26c7/src/.libs/liborange.so -lblobpack -lutype -lpthread -lwebsockets -lcrypt -llua -luci -lrt
/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/target-x86_64_glibc/usr/lib/liblua.so: undefined reference to dlopen' /home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/target-x86_64_glibc/usr/lib/liblua.so: undefined reference todlclose'
/home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/target-x86_64_glibc/usr/lib/liblua.so: undefined reference to dlerror' /home/dbrody/bamboo-data/xml-data/build-dir/CON-LEDE-JOB1/staging_dir/target-x86_64_glibc/usr/lib/liblua.so: undefined reference todlsym'

mkschreder, 2017-12-12:
Seems you need to add -ldl to linker flags to get it to link on your
system.

dbrody2004, 2017-12-13:
I removed -Werror from the Makefile.am and then added -ldl and it works.  Not sure if it is platform dependent.

mkschreder, 2017-12-14:
It is probably musl dependent :)

dbrody2004, 2017-12-15:
Removed the -Werror and added -ldl to the Makefile and compiles fine.

closed 2017-12-15 by dbrody2004