mkschreder/juci #82: UCI failed to initialize

dbrody2004, 2017-12-13:
Compiled and getting UCI failed to initialize on the login screen console. ubus list -v is below and doesn't show juci.  Normally this is due to lighttpd, but this is compiled with juci lighttpd.  The build is LEDE 17.01 on x86.

[image]

root@marco:~# ubus list -v
'dhcp' @75cbac6c
	"ipv4leases":{}
	"ipv6leases":{}
'dnsmasq' @db64f88b
'log' @c819d418
	"read":{"lines":"Integer","stream":"Boolean"}
	"write":{"event":"String"}
'network' @90f0cf51
	"restart":{}
	"reload":{}
	"add_host_route":{"target":"String","v6":"Boolean","interface":"String"}
	"get_proto_handlers":{}
	"add_dynamic":{"name":"String"}
'network.device' @5cf76ef1
	"status":{"name":"String"}
	"set_alias":{"alias":"Array","device":"String"}
	"set_state":{"name":"String","defer":"Boolean"}
'network.interface' @488feabe
	"up":{}
	"down":{}
	"renew":{}
	"status":{}
	"prepare":{}
	"dump":{}
	"add_device":{"name":"String","link-ext":"Boolean"}
	"remove_device":{"name":"String","link-ext":"Boolean"}
	"notify_proto":{}
	"remove":{}
	"set_data":{}
'network.interface.lan' @2255358a
	"up":{}
	"down":{}
	"renew":{}
	"status":{}
	"prepare":{}
	"dump":{}
	"add_device":{"name":"String","link-ext":"Boolean"}
	"remove_device":{"name":"String","link-ext":"Boolean"}
	"notify_proto":{}
	"remove":{}
	"set_data":{}
'network.interface.loopback' @9941f121
	"up":{}
	"down":{}
	"renew":{}
	"status":{}
	"prepare":{}
	"dump":{}
	"add_device":{"name":"String","link-ext":"Boolean"}
	"remove_device":{"name":"String","link-ext":"Boolean"}
	"notify_proto":{}
	"remove":{}
	"set_data":{}
'network.interface.wan' @55861157
	"up":{}
	"down":{}
	"renew":{}
	"status":{}
	"prepare":{}
	"dump":{}
	"add_device":{"name":"String","link-ext":"Boolean"}
	"remove_device":{"name":"String","link-ext":"Boolean"}
	"notify_proto":{}
	"remove":{}
	"set_data":{}
'network.wireless' @7efe73c2
	"up":{}
	"down":{}
	"status":{}
	"notify":{}
	"get_validate":{}
'service' @d7437899
	"set":{"name":"String","script":"String","instances":"Table","triggers":"Array","validate":"Array","autostart":"Boolean"}
	"add":{"name":"String","script":"String","instances":"Table","triggers":"Array","validate":"Array","autostart":"Boolean"}
	"list":{"name":"String","verbose":"Boolean"}
	"delete":{"name":"String","instance":"String"}
	"signal":{"name":"String","instance":"String","signal":"Integer"}
	"update_start":{"name":"String"}
	"update_complete":{"name":"String"}
	"event":{"type":"String","data":"Table"}
	"validate":{"package":"String","type":"String","service":"String"}
	"get_data":{"name":"String","instance":"String","type":"String"}
	"state":{"spawn":"Boolean","name":"String"}
'session' @686fcc88
	"create":{"timeout":"Integer"}
	"list":{"ubus_rpc_session":"String"}
	"grant":{"ubus_rpc_session":"String","scope":"String","objects":"Array"}
	"revoke":{"ubus_rpc_session":"String","scope":"String","objects":"Array"}
	"access":{"ubus_rpc_session":"String","scope":"String","object":"String","function":"String"}
	"set":{"ubus_rpc_session":"String","values":"Table"}
	"get":{"ubus_rpc_session":"String","keys":"Array"}
	"unset":{"ubus_rpc_session":"String","keys":"Array"}
	"destroy":{"ubus_rpc_session":"String"}
	"login":{"username":"String","password":"String","timeout":"Integer"}
'system' @373a5bcb
	"board":{}
	"info":{}
	"reboot":{}
	"watchdog":{"frequency":"Integer","timeout":"Integer","magicclose":"Boolean","stop":"Boolean"}
	"signal":{"pid":"Integer","signum":"Integer"}
	"sysupgrade":{"path":"String","prefix":"String","command":"String"}
'uci' @a9e47e5d
	"configs":{}
	"get":{"config":"String","section":"String","option":"String","type":"String","match":"Table","ubus_rpc_session":"String"}
	"state":{"config":"String","section":"String","option":"String","type":"String","match":"Table","ubus_rpc_session":"String"}
	"add":{"config":"String","type":"String","name":"String","values":"Table","ubus_rpc_session":"String"}
	"set":{"config":"String","section":"String","type":"String","match":"Table","values":"Table","ubus_rpc_session":"String"}
	"delete":{"config":"String","section":"String","type":"String","match":"Table","option":"String","options":"Array","ubus_rpc_session":"String"}
	"rename":{"config":"String","section":"String","option":"String","name":"String","ubus_rpc_session":"String"}
	"order":{"config":"String","sections":"Array","ubus_rpc_session":"String"}
	"changes":{"config":"String","ubus_rpc_session":"String"}
	"revert":{"config":"String","ubus_rpc_session":"String"}
	"commit":{"config":"String","ubus_rpc_session":"String"}
	"apply":{"rollback":"Boolean","timeout":"Integer","ubus_rpc_session":"String"}
	"confirm":{"ubus_rpc_session":"String"}
	"rollback":{"ubus_rpc_session":"String"}
	"reload_config":{}

[image]

mkschreder, 2017-12-15:
It uses orangerpcd. In the orangerpcd git repo there is an orangectl script
that can be used to login over rpc and show rpc objects. RPC objects and
ubus objects are not the same for the simple fact that RPC is facing
outwards and ubus is an internal system bus. However, through orangerpcd
scripts, some ubus objects can be exposed over RPC as well. When you get
these errors it is quite likely that either orangerpcd is not working
correctly or it is not installed at all.

dbrody2004, 2017-12-15:
Turns out a lighttpd issue.  I used the script in the juci-openwrt repo to compile.  The sequence is
./scripts/feeds update -a
./scripts/feeds install -f -a -p juci
./scripts/feeds install -a  

The install -a command overwrites the installed juci lighttpd which I thought wasn't going to happen.  The solution is to add rm -f -r ./feeds/packages/net/lighttpd right after the install -a to remove the lighttpd package.  Works now.

closed 2017-12-15 by dbrody2004