mkschreder/juci #83: Network connections page protocols error.

dbrody2004, 2017-12-15:
Getting the following error on this page.  I don't see protocols listed under ubus.

[image]

[image]

mkschreder, 2017-12-15:
./orange --host <your router> --username <juci user> --password <juci
login> list

This will list rpc objects exported by orangerpcd to juci.

you can find the client here:
https://github.com/mkschreder/orangerpcd/blob/master/orange

mkschreder, 2017-12-15:
> mkschreder/orangerpcd/blob/master/orange

Check if $rpc.juci.network object exists. You can run orangerpcd with -vvvv
flag manually (instead of through init.d) to dump details of all requests
going back and forth and possibly figure out what the issue is.

mkschreder, 2017-12-15:
A request arrives at lighttpd websocket proxy, gets forwarded to locally
running orangerpcd websocket server, then gets dispatched to appropriate
lua script within orangerpcd, then either through ubus command to ubus
object or directly back to the client with structured data from lua which
gets translated into json along the way.

closed 2017-12-18 by dbrody2004