mkschreder/juci #84: New plugin

dbrody2004, 2017-12-18:
I have a new plugin that works well except for the Apply button.  There are a number of settings on this page.  The Apply button takes a long time to apply changes and the more changes that are there the longer it takes.  What do you think the issue could be?

mkschreder, 2017-12-19:
Difficult to say but apply button should work automatically based on the
settings that have changed. It will appear when some setting changes in the
local uci cache and then submit write requests for all changed values when
you hit apply. You could debug the communication and see how write requests
are being submitted and if there is some kind of delay associated with each
request. That would explain why it takes longer the more settings you need
to save.

dbrody2004, 2017-12-19:
What's the best way to debug the communications?  I replaced the /sbin/uci executable with a script that logs to a file and then executes the original /sbin/uci.  That didn't show me much except that I found out that the service is started after everytime a value is changed.  So for instance if there are three values to be applied, it sets one value, restarts the service, sets the next one, restarts, the service, etc.  These restarts are about 5 seconds in between.

Shouldn't all the changes be applied and the service restarted?

mkschreder, 2017-12-20:
Best way to debug communications is to do "/etc/init.d/orange stop" and
then to start it like this: orangerpcd -vvvv -p /usr/lib/orange/api

This will dump all communication to stdout.

closed 2021-04-20 by mkschreder