mkschreder/juci #85: Compiling Error with LEDE

techspace01, 2017-12-20:
I got issue compiling with JUCI with LEDE and got the following errors. The procedure I did is run the openwrt-bootstrap.sh script and run make. I run same steps on the OpenWRT without issue. Do you have any idea on the error "Headers not exportable for the um architecture"?

-fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result" INSTALL_HDR_PATH="/home/techspace/lede-juci/source/build_dir/toolchain-x86_64_gcc-5.5.0_musl/linux-dev/" headers_install
make[4]: Entering directory '/home/techspace/lede-juci/source/build_dir/toolchain-x86_64_gcc-5.5.0_musl/linux-4.9.70'
  CHK     include/generated/uapi/linux/version.h
**Makefile:1151: *** Headers not exportable for the um architecture.  Stop.**
make[4]: Leaving directory '/home/techspace/lede-juci/source/build_dir/toolchain-x86_64_gcc-5.5.0_musl/linux-4.9.70'
Makefile:98: recipe for target '/home/techspace/lede-juci/source/build_dir/toolchain-x86_64_gcc-5.5.0_musl/linux-4.9.70/.configured' failed
make[3]: *** [/home/techspace/lede-juci/source/build_dir/toolchain-x86_64_gcc-5.5.0_musl/linux-4.9.70/.configured] Error 2

mkschreder, 2017-12-20:
This seems to be an issue with building toolchain so not related to juci.
However it could be that it is not possible to build your toolchain for
usermode linux. I would guess that um means usermode linux. The bootstrap
script configures the build for um linux. Just try building for some
embedded board instead.

techspace01, 2017-12-21:
I tried to compile with for embedded board and got dependencies missing for lighttpd-mod-websocket. 
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for juci:
 *      lighttpd-mod-websocket *

Here are my steps:
make distclean
1. - echo "src-git-full juci https://github.com/mkschreder/juci-openwrt-feed.git" >> feeds.conf.default
2. - ./scripts/feeds update -a 
3. - ./scripts/feeds install -f -a -p juci
4. - ./scripts/feeds install -a
5. cat >> .config
CONFIG_PACKAGE_juci-full-openwrt=y
CONFIG_PACKAGE_orange-rpcd=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUSYBOX_CONFIG_SHA1SUM=y
^D

6. make menuconfig
7. Target System 'Qualcomm Atheros IPQ806X'
8. Target Profile 'TP-Link Archer C2600'
9. make defconfig
10. make -j1 V=s

mkschreder, 2017-12-22:
>       juci-openwrt-feed.git" >> feeds.conf.default

You have likely not installed the juci feed correctly. There is a
replacement lighttpd with corresponding mod websocket in the repo. It
should override the default lighttpd that comes with LEDE.

techspace01, 2017-12-22:
I removed the step #3 above and it's compiled successfully now.

closed 2017-12-22 by techspace01