mkschreder/juci #87: Unable to build

davin4i88, 2019-02-15:
Hello,

from two days i try to build with OpenWRT 15.05, Lede 17 and 18, no success with any of them, looks like there is some problem with the websockets, is there any suggestion how i can build it? I am planning to start writing plugins and extensions for this Interface, i like it a lot also i love that its nodejs bases. I think is is the only one exist for openWRT, not every one can write on Lua. Please Help me with some points. On Lede and 18 versions the Lib sockets not exist in the webserver menu, i am doing the steps exac like in the tutorial and installing the Juci feeds. I also want to ask does Juci will continue with the support or the project is dead?

Thank you in advance!

mkschreder, 2019-02-15:
You are missing juci feed. Look up juci-openwrt-feed and there is an
example script of how to bootstrap your openwrt correctly.

davin4i88, 2019-02-15:
Hello,

thanks for the fast answer, i already try with the script, i am reading the past issues. Are you sure the build working with the lastest libsebsockets repo?

davin4i88, 2019-02-15:
If yes i will try on different platform, i am doing it now with Ubuntu Server 16. Can you suggest me compilation environment?

mkschreder, 2019-02-15:
Since it is an open source project, it is up to you to make it work for
yourself. Try the example bootstrap script first to get all feed overrides
working and then you should be able to build a user mode linux image. From
there you can adopt it to your hardware.

davin4i88, 2019-02-15:
I am tring now with the release version 2.16.02, looks like its ok for now.

Thank you for the answers, i will share the results.

mkschreder, 2019-02-15:
Great!

davin4i88, 2019-02-15:
Hey,

i was success building OpenWRT 15.05.1 with Juci v2.16.02 directly for firmware router mt7620n, there is a lot errors, but i will start debug now, looks like this version working ubus not websockets, i will try now with lede and latest Juci version.

mkschreder, 2019-02-15:
You may get errors on the pages due to some cli utilities being changed or
missing. The juci orangerpcd jsonrpc server has a -vvv flag which you can
use to get a detailed trace of backend calls. It is very useful when
writing and debugging backend scripts.

Btw lua is the only high level language I know that has a grammar that fits
onto a single A4 page. The language is so simple that anyone can learn it
in ablut a day.

davin4i88, 2019-02-15:
Maybe you are right, i will give a try, but i really like your work and the nodejs idea! I am compiling now with ubus and Lede, lets see how will this come. Websocket version was unable to build again, i see there is changes in libwebsockets repo from few hours, maybe this is the problem. Anyway lets see, if ubus and lede 17 working great i will give a try with OpenWrt 18.

closed 2021-04-20 by mkschreder