mkschreder/juci #93: juci project no more update?

RadioOperator, 2021-11-27:
Hi, any open source makes this JUCI works on latest openwrt version 21.02, please let me know, thanks.

dohg, 2022-07-06:
It doesn't seem like the developer is active anymore. I would consider JUCI as legacy software. I wouldn't expect an update anytime soon, however, the latest version seems to work fine.

RadioOperator, 2022-07-06:
[gargoile](https://www.gargoyle-router.com/download.php) maybe an alternative.

mkschreder, 2022-08-07:
> It doesn't seem like the developer is active anymore. I would consider JUCI as legacy software (Last release was in 2016, probably built and compiled with outdated software). I wouldn't expect an update anytime soon, however, the latest version seems to work fine. I've read up on JUCI a bit, and it seems that the dev used it for custom software for clients (AFAIK). I assume the dev doesn't need the custom software for clients anymore, meaning the dev has no reason to continue updating.

I simply stopped working on Web UI part of firmware development and had no reason to maintain this project.

closed 2024-02-25 by mkschreder